eModbus/eModbus #39: flush and rtsPin

DierkJ, 2020-12-28:
**RS.485 rx/tx flow control does not work**
I am using ModbusClientRTU() to read Modbus energy meters.  My RS.485 hardware is using an ESP32 pin to control the receive / transmit.  The implementation in RTUtils.cpp RTUutils::send() does not work, because the RTS pin is switched only on for less than a millisecond.  I uncommented serial.flush()...

**Code in RTUtils.cpp, line 110**
```
// serial.flush();
// Toggle rtsPin, if necessary
if (rtsPin >= 0) digitalWrite(rtsPin, LOW);
```
**exchange with:**
```
// 
// Toggle rtsPin, if necessary
if (rtsPin >= 0) {
   serial.flush();
   digitalWrite(rtsPin, LOW);
}
```

Miq1, 2020-12-28:
I will have a look at this tomorrow.

Miq1, 2020-12-29:
You are right:
[image]

I must confess I had abandoned the non-auto halfduplex modules a while ago, and due to another issue I removed as many ``flush()`` calls as were possible (the issue being that input FIFOs are flushed as well on the ESP32). That should not matter with RTU, so your fix seems to be okay:
[image]

Thank you, will be added!

closed 2020-12-29 by Miq1

DierkJ, 2020-12-30:
Thanks, perfect.