eModbus/eModbus #40: Invalid CRC (RTU Modbus) - RTU06Example

thehaxxa, 2020-12-30:
**Describe the bug**

When using the RTU06Example with an ESP32-WROOM (NodeMCU) Development Board and Modscan64 as a server. The CRC validation fails when reading registers. It is however able to write to registers without issue. 

The connection is made via UART (CP2102) to a Windows host. 

**Expected behavior**

It is expected that CRC validation would succeed in both contexts. 

**To Reproduce**

Install Modscan64 (demo or equivalent) 
Run Example

**Example code**

RTU06Example _(accessed 29/12/2020)_
https://github.com/eModbus/eModbus/blob/d0e5329e93290076d271f374ed13c654fbf00ef7/examples/RTU06example/main.cpp

I have also attempted to add CRC to a Modbus Message manually with no difference in the outcome: 
```
        ModbusMessage msg = ModbusMessage(1, READ_HOLD_REGISTER, 10, 1);
        RTUutils::addCRC(msg);
        err = MB.addRequest(msg, Token++);
        if (err != SUCCESS)
        {
            ModbusError e(err);
            Serial.printf("Error creating request: %02X - %s\n", (int)e, (const char *)e);
        }
```

**Additional context**

I have tried using alternative serial configurations (SERIAL8E1, SERIAL8N2, etc.) to no success. 

_Modbus Messages recorded via Modscan64 (over UART):_

[image]

_CRC Counters:_

[image]

_Register written to successfully:_ 

[image]

_MCU Output:_

```
Response: serverID=1, FC=6, Token=00000457, length=6:
01 06 00 0A 12 34 
Error response: E0 - Timeout
Response: serverID=1, FC=6, Token=00000459, length=6:
01 06 00 0A BE EF 
Error response: E0 - Timeout
```

Miq1, 2020-12-30:
What does tell you the CRC check failing? The error message clearly is stating a TIMEOUT event, so my first idea would be to look for the Modscan64 response and your timeout settings.

Miq1, 2020-12-30:
Another thing: is Modscan64 supposed to work as a server as well? The references I found on the Net were for a client use only.

bertmelis, 2020-12-30:
For reference: the CRC implementation in this lib is the one given as an example in the modbus specification: https://modbus.org/docs/Modbus_over_serial_line_V1_02.pdf

thehaxxa, 2020-12-30:
> Correction: Modscan64 does not know about a dedicated CRC error response, so it will simply not respond at all. This is the reason for ModbusClientRTU to report a TIMEOUT.
> Another thing: is Modscan64 supposed to work as a server as well? The references I found on the Net were for a client use only.

My apologies, I am using ModSim64, I apologize for the confusion.  

I will test the Modbus implementation with some PLCs and see if it functions as expected. I'm not quite sure why it wouldn't be working, as we use Modsim32/64 (& Modscan32/64) quite extensively to test PLCs and Modbus devices. I was hoping it might have been an obvious mistake on my part. 

I tested with diagslave http://www.modbusdriver.com/diagslave.html and it seems to be working, I am still hesitant to write this off as an issue with Modsim32 until I have a clear understanding of why its not working.

Miq1, 2020-12-30:
Please keep us posted about your findings.

thehaxxa, 2021-01-01:
Ok so I spent a lot of time testing over NYE, and here is my findings:

I uploaded the following code to the ESP32 in the `void_setup` section:

_[Notice how the same operations are repeated twice]._ 

```
// Now write 0xBEEF to it
err = MB.addRequest(Token++, 3, WRITE_HOLD_REGISTER, 2, 0xBEEF);
if (err != SUCCESS)
{
    ModbusError e(err);
    Serial.printf("Error creating request: %02X - %s\n", (int)e, (const char *)e);
}

// Read it again to verify
err = MB.addRequest(Token++, 3, READ_HOLD_REGISTER, 2, 1);
if (err != SUCCESS)
{
    ModbusError e(err);
    Serial.printf("Error creating request: %02X - %s\n", (int)e, (const char *)e);
}

// Now write 0xBEEF to it
err = MB.addRequest(Token++, 3, WRITE_HOLD_REGISTER, 2, 0xBEEF);
if (err != SUCCESS)
{
    ModbusError e(err);
    Serial.printf("Error creating request: %02X - %s\n", (int)e, (const char *)e);
}

// Read it again to verify
err = MB.addRequest(Token++, 3, READ_HOLD_REGISTER, 2, 1);
if (err != SUCCESS)
{
    ModbusError e(err);
    Serial.printf("Error creating request: %02X - %s\n", (int)e, (const char *)e);
}
```

I used several PLCs to test with and all had the same error output as follows:

```
Error response: E5 - Packet length error
Error response: E4 - Server ID mismatch
Error response: E0 - Timeout
Response: serverID=3, FC=3, Token=000007BA, length=5:
03 03 02 0B B8 
```

Interestingly if I just have the following:

```
// Now write 0xBEEF to it
err = MB.addRequest(Token++, 3, WRITE_HOLD_REGISTER, 2, 0xBEEF);
if (err != SUCCESS)
{
    ModbusError e(err);
    Serial.printf("Error creating request: %02X - %s\n", (int)e, (const char *)e);
}

// Read it again to verify
err = MB.addRequest(Token++, 3, READ_HOLD_REGISTER, 2, 1);
if (err != SUCCESS)
{
    ModbusError e(err);
    Serial.printf("Error creating request: %02X - %s\n", (int)e, (const char *)e);
}
 ```
 Communication works as expected.
 
 ----------------------------
 
I realised there must have been an issue with the output, so I connected a logic analyser and confirmed the output was not as anticipated.  I have attached the logic analyser capture from running this code. The checksums are invalid and the more tests I ran, the more quirky behaviour I found. 
 
Specifically, if I implemented two of the same read/write holding register operations the communication does not work. If I write, read and write multiple registers it works as expected.

Hopefully the logic analyser capture gives context to this timing issue, but it seems there is a bug in the Modbus output. Although, I might be overlooking something? I have also attached a copy of the code so the issue can be replicated. 

[image]

**Data Captures:**
----------------------
RAW: 
[BIT_CAPTURE.zip](https://github.com/eModbus/eModbus/files/5758787/BIT_CAPTURE.zip)

MODBUS: 
[CAPTURE.zip](https://github.com/eModbus/eModbus/files/5758788/CAPTURE.zip)

CODE:
[CODE.zip](https://github.com/eModbus/eModbus/files/5758791/CODE.zip)

----------------------

It doesn't matter what Modbus server I have, the issue is replicated with different PLCs & MCU's via direct UART using a variety of Serial settings. 

Let me know if you need any more information, I have setup two microcontrollers _(host & client)_ with a logic analyser for testing.

Miq1, 2021-01-01:
Thanks a lot for the materials. I will have a look at this today.

Without having done that yet, I would bet on out-of-sync timing on the RTU Modbus. Did you see the lengthy comment in RTUutils.cpp on the Serial handling issue the arduino-esp32 core has? You will need to either modify the esp-hal_uart.cc as described or accept the much longer interval between requests.

thehaxxa, 2021-01-01:
I tested modifying `esp-hal_uart.c` with `    uart->dev->conf1.rxfifo_full_thrhd = 1;` the same issues persist.

Miq1, 2021-01-01:
Dang. Could you perhaps put a ``#define LOCAL_LOG_LEVEL LOG_LEVEL_VERBOSE`` at the start of both RTUutils.cpp and ModbusServerRTU.cpp, and ``MBUlogLvl = LOG_LEVEL_VERBOSE;`` in your ``setup() `` and then run the stuff again? The extended log may give additional information.

Miq1, 2021-01-01:
In your code I found
```
// Set up Serial2 connected to Modbus RTU
// (Fill in your data here!)
  Serial2.begin(19200, SERIAL_8E1, GPIO_NUM_16, GPIO_NUM_17);
```
The ``SERIAL_8E1`` is at least uncommon - are you sure your server does even parity? I never saw anything else but ``SERIAL_8N1`` before.

I can not take much information from the two capture CSVs, I am afraid. Is your LA able to do a byte printout on the Serial Tx and Rx lines?

I reckon the verbose log may be helpful indeed.

Miq1, 2021-01-04:
@thehaxxa did you get any further in the meantime?

Miq1, 2021-01-09:
Seems to be irrelevant in the meantime.

thehaxxa, 2021-01-11:
@Miq1 I spent additional time testing the eModbus Library in several additional scenarios.

I would like to confirm the comment in the esp-hal_uart.cc did end up resolving my timing issues, more specifically setting `uart->dev->conf1.rxfifo_full_thrhd = 1;` resolved my issues. PlatformIO was compiling using Windows Subsystem for Linux (WSL)  instead of the Windows environment. Thus my initial change was not compiled and I incorrectly concluded the change did not fix the issue. 

Thank you for all your time and support diagnosing the connection.

------------------------

Overall the library is very well documented and is nice to work with in contrast to other Modbus libraries on Embedded platforms. 

I would however propose the comment made in `esp-hal_uart.c` is added to the documentation (provided on Github). User's unfamiliar with this library may assume, the library is not working. While searching through code to find comments on the subject is tedious and elusive. As the library targets the ESP32 platform, I think it would be a good addition to the documentation. Thoughts?

In any case, thanks once again for your help and for producing a well designed and documented library.

Miq1, 2021-01-11:
Thank you for your report and kind words. 👍

Initiated by your issue I already did what you are proposing, the advice to change the core code now is part of the Requirements section of the docs.

Ebrahim1604, 2021-07-06:
@Miq1 can you point me out at which section does the info about changing core code currently resides?

Can't seem to find it in the requirements section of the docs (or am i missing something?)

Also it seems the lengthy comment in RTUutils.cpp has been perhaps removed, right?

Just to give you'll some context, I'm getting error E5 for a just single read operation utilizing the RTU06 example.

Miq1, 2021-07-06:
@Ebrahim1604 we changed the complete timeout handling with RTU in the last but one release, and the core patch is no longer required, too, as the necessary change is done temporarily for the running program in UARTinit().

The E5 packet length error is indicating some other reason. Can yo please post the relevant parts of your code to let us have a look?

Ebrahim1604, 2021-07-06:
Sure, here are the details.

Trying to setup a Rtu cleint in LOLIN D32 PRO

The rtu server is a pressure sensor as following: https://download.keller-druck.com/api/download/muQjLg2dGMJAbgLYSAHreT/en/latest.pdf (If you'd like to have a look section 4 is where you can)

The rs485 to uart convertor being employed here is as follows: https://www.robotshop.com/uk/sfe-uart-to-rs-485-converter.html (Not auto half duplex so requires a rts pin)

Here is the modified .ino file where I try to read the value at register 0x0200, which is for setting the baud rate of the server, default being 9600 the bit0:bit2 should be 0:
```
// =================================================================================================
// eModbus: Copyright 2020 by Michael Harwerth, Bert Melis and the contributors to ModbusClient
//               MIT license - see license.md for details
// =================================================================================================

// Example code to show the usage of the eModbus library. 
// Please refer to root/Readme.md for a full description.

// Includes: <Arduino.h> for Serial etc.

/*
 Pin connections:
      LOLIN d32 pro            Uart to rs485 convertor
    1)   GPIO 12--------------------------TX
    2)   GPIO 11--------------------------RX
    3)   GPIO 15--------------------------RTS
    4)    3.3V----------------------------VCC
    5)    GND-----------------------------GND    
 */

// Include the header for the ModbusClient RTU style
#include "ModbusClientRTU.h"

// For demonstration, use the LOG statements for output
#include "Logging.h"

const byte rxd = 12;
const byte txd = 13;
const byte rts = 15;

#define demo_addr 0x0200 //bit0, bit1 & bit2 are for baud rate, default 9600 

// Create a ModbusRTU client instance
ModbusClientRTU MB(Serial2, rts);

// Define an onData handler function to receive the regular responses
// Arguments are Modbus server ID, the function code requested, the message data and length of it, 
// plus a user-supplied token to identify the causing request
void handleData(ModbusMessage response, uint32_t token) 
{
  // Only print out result of the "real" example - not the request preparing the field
  if (token > 1111) {
    LOG_N("Response: serverID=%d, FC=%d, Token=%08X, length=%d:\n", response.getServerID(), response.getFunctionCode(), token, response.size());
    HEXDUMP_N("Data", response.data(), response.size());
  }
}

// Define an onError handler function to receive error responses
// Arguments are the error code returned and a user-supplied token to identify the causing request
void handleError(Error error, uint32_t token) 
{
  // ModbusError wraps the error code and provides a readable error message for it
  ModbusError me(error);
  LOG_E("Error response: %02X - %s\n", (int)me, (const char *)me);
}

// Setup() - initialization happens here
void setup() {
// Init Serial monitor
  Serial.begin(9600);
  while (!Serial) {}
  Serial.println("__ Kellar33x Modbus demo code __");

// Set up Serial2 connected to Modbus RTU
// (Fill in your data here!)
  Serial2.begin(9600, SERIAL_8N1, rxd, txd);

// Set up ModbusRTU client.
// - provide onData handler function
  MB.onDataHandler(&handleData);
// - provide onError handler function
  MB.onErrorHandler(&handleError);
// Set message timeout to 2000ms
  MB.setTimeout(2000);
// Start ModbusRTU background task
  MB.begin();

  uint32_t Token = 1111;

// Create request by
// - token to match the response with the request.
// - server ID = 1
// - function code = 0x03 (read holding register)
// - address to read = demo_addr
// - data words to read = 1
//
// If something is missing or wrong with the call parameters, we will immediately get an error code 
// and the request will not be issued

// Read register
 Error err = MB.addRequest(Token++, 1, READ_HOLD_REGISTER, demo_addr, 1);
  if (err!=SUCCESS) {
    ModbusError e(err);
    LOG_E("Error creating request: %02X - %s\n", (int)e, (const char *)e);
  }
 
// loop() - nothing done here today!
void loop() 
{
}
```

Here is the Serial monitor output, where along with those garbage values(not sure how that appeared) the E5 packet length error is seen:
[image]

I hope that's enough. Thanks for your help!

Miq1, 2021-07-06:
I see. What makes me frown in the first row ist section 4.2 of the Keller documentation where they state that their CRC has a different byte order than Modbus. WTF?
Can you recompile your sketch with ``-DLOG_LEVEL=6`` and post the log output here? That should give us hints on what is going back and forth.

Ebrahim1604, 2021-07-06:
@Miq1 sure ill do that, but i think what you're referring to is kellar protocol.

It is their own custom protocol which is slightly different from Modbus.

But I prefer going with Modbus as the api they provided for Kellar has tons of issues and modbus is quite popular & has tons of resources as opposed to kellar.

Miq1, 2021-07-09:
@Ebrahim1604 any progress on this?

Ebrahim1604, 2021-07-12:
Hello, Apologies for the delay, have been sick past week and just got back to work.

I just wanted to clarify, I'm using arduino ide and not paltform.io so I'm not entirely sure how can i recompile the code with  the flag -DLOG_LEVEL=6

Have enabled the the verbose output in the File->Preferences section of the Ardunio IDE.

Got the following output:

> >>

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5856
entry 0x400806a8
_ Kellar33x Modbus demo code _
Setup() function running on core: 1
handle error() function running on core: 0
[1;33m[E] 2138| Kellar_ESP32.ino     [  48] handleError: Error response: E0 - Timeout
[0m

<<<

Ebrahim1604, 2021-07-12:
Ok as an update, I added the line : #define LOG_LEVEL LOG_LEVEL_VERBOSE above the #include "Logging.h", so this is the code:

```
/*
 Pin connections:

      LOLIN d32 pro            Uart to rs485 convertor
    1)   GPIO 13--------------------------TX
    2)   GPIO 12--------------------------RX
    3)   GPIO 15--------------------------RTS
    4)    3.3V----------------------------VCC
    5)    GND-----------------------------GND    
 */

// Include the header for the ModbusClient RTU style
#include "ModbusClientRTU.h"

#define LOG_LEVEL LOG_LEVEL_VERBOSE
// For demonstration, use the LOG statements for output
#include "Logging.h"

const byte rxd = 12;
const byte txd = 13;
const byte rts = 15;

#define demo_addr 0x0200

// Create a ModbusRTU client instance
ModbusClientRTU MB(Serial2, rts);

// Define an onData handler function to receive the regular responses
// Arguments are Modbus server ID, the function code requested, the message data and length of it, 
// plus a user-supplied token to identify the causing request
void handleData(ModbusMessage response, uint32_t token) 
{
  Serial.print("handledata() function running on core: ");
  Serial.println(xPortGetCoreID());
  // Only print out result of the "real" example - not the request preparing the field
  if (token > 1111) {
    LOG_N("Response: serverID=%d, FC=%d, Token=%08X, length=%d:\n", response.getServerID(), response.getFunctionCode(), token, response.size());
    HEXDUMP_N("Data", response.data(), response.size());
  }
}

// Define an onError handler function to receive error responses
// Arguments are the error code returned and a user-supplied token to identify the causing request
void handleError(Error error, uint32_t token) 
{
  Serial.print("handle error() function running on core: ");
  Serial.println(xPortGetCoreID());
  // ModbusError wraps the error code and provides a readable error message for it
  ModbusError me(error);
  LOG_E("Error response: %02X - %s\n", (int)me, (const char *)me);
}

// Setup() - initialization happens here
void setup() {
// Init Serial monitor
  Serial.begin(115200);
  while (!Serial) {}
  Serial.println("__ Kellar33x Modbus demo code __");

  Serial.print("Setup() function running on core: ");
  Serial.println(xPortGetCoreID());
// Set up Serial2 connected to Modbus RTU
// (Fill in your data here!)
  Serial2.begin(9600, SERIAL_8N1, rxd, txd);

// Set up ModbusRTU client.
// - provide onData handler function
  MB.onDataHandler(&handleData);
// - provide onError handler function
  MB.onErrorHandler(&handleError);
// Set message timeout to 2000ms
  MB.setTimeout(2000);
// Start ModbusRTU background task
  MB.begin();

  uint32_t Token = 1111;

// Create request by
// - token to match the response with the request.
// - server ID = 1
// - function code = 0x03 (read holding register)
// - address to read = word 10
// - data words to read = 1
//
// If something is missing or wrong with the call parameters, we will immediately get an error code 
// and the request will not be issued

// Read register
 Error err = MB.addRequest(Token++, 1, READ_HOLD_REGISTER, demo_addr, 1);
  if (err!=SUCCESS) {
    ModbusError e(err);
    LOG_E("Error creating request: %02X - %s\n", (int)e, (const char *)e);
  }
  
/*
// Now write 0xBEEF to it
  err = MB.addRequest(Token++, 1, WRITE_HOLD_REGISTER, 10, 0xBEEF);
  if (err!=SUCCESS) {
    ModbusError e(err);
    LOG_E("Error creating request: %02X - %s\n", (int)e, (const char *)e);
  }

// Read it again to verify
  err = MB.addRequest(Token++, 1, READ_HOLD_REGISTER, 10, 1);
  if (err!=SUCCESS) {
    ModbusError e(err);
    LOG_E("Error creating request: %02X - %s\n", (int)e, (const char *)e);
  }

// Provoke an error just to show it
  err = MB.addRequest(Token++, 1, USER_DEFINED_44, 10, 1);
  if (err!=SUCCESS) {
    ModbusError e(err);
    LOG_E("Error creating request: %02X - %s\n", (int)e, (const char *)e);
  }
*/
}
// loop() - nothing done here today!
void loop() 
{
}

```
For some reason, the error message went from E0 to E4 i.e. server id mismatch which is extremely weird as the server id of the sensor is set to 1 and verified. Here is the output:

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5856
entry 0x400806a8
_ Kellar33x Modbus demo code _
Setup() function running on core: 1
handle error() function running on core: 0
[1;33m[E] 153| Kellar_ESP32.ino     [  51] handleError: Error response: E4 - Server ID mismatch
[0m

Miq1, 2021-07-12:
In the IDE unfortunately it is not possible to set compiler flags on the fly, so you need to change the default log level in the ``Logging.h`` file. There is a line close to the beginning stating ``#define LOG_LEVEL LOG_LEVEL_ERROR``. Please change to ``LOG_LEVEL_VERBOSE``.

You will have to change that back at a time where you will want to get rid of the additional output.

Ebrahim1604, 2021-07-12:
ok so after changing the logging.h file, here is the output:

________________________________________________________________________________
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5856
entry 0x400806a8
_ Kellar33x Modbus demo code _
Setup() function running on core: 1
[D] 28| ModbusClientRTU.cpp  [ 100] setTimeout: Timeout set to 2000
[D] 28| ModbusClientRTU.cpp  [  72] begin: Worker task 1073452664 started. Interval=3645
[V] 36| ModbusMessage.cpp    [ 529] checkData: Check data #3
[D] 41| ModbusClientRTU.cpp  [ 107] addRequestM: request for 01/03
[D] 47| ModbusClientRTU.cpp  [ 156] addToQueue: RC=01
[D] 52| ModbusClientRTU.cpp  [ 118] addRequestM: RC=00
[D] 128| ModbusClientRTU.cpp  [ 174] handleConnection: Pulled request from queue
[D] Sent packet: @3FFB9804/6:
  | 0000: 01 03 02 00 00 01                                 |......          |
[D] 137| ModbusClientRTU.cpp  [ 179] handleConnection: Request sent.
[D] Received packet: @3FFB984C/9:
  | 0000: 00 01 03 02 00 00 B8 44  00                       |.......D.       |
[D] 154| ModbusClientRTU.cpp  [ 185] handleConnection: Data response (9 bytes) received.
[V] Data: @3FFB984C/9:
  | 0000: 00 01 03 02 00 00 B8 44  00                       |.......D.       |
[D] 169| ModbusClientRTU.cpp  [ 214] handleConnection: Response generated.
[V] Response packet: @3FFB9804/3:
  | 0000: 01 83 E4                                          |...             |
handle error() function running on core: 0
[1;33m[E] 189| Kellar_demo_1.ino    [  52] handleError: Error response: E4 - Server ID mismatch
[0m
________________________________________________________________________________

Also tried changing tx and rx, the output was as follows:

________________________________________________________________________________
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5856
entry 0x400806a8
_ Kellar33x Modbus demo code _
Setup() function running on core: 1
[D] 28| ModbusClientRTU.cpp  [ 100] setTimeout: Timeout set to 2000
[D] 28| ModbusClientRTU.cpp  [  72] begin: Worker task 1073452664 started. Interval=3645
[V] 36| ModbusMessage.cpp    [ 529] checkData: Check data #3
[D] 41| ModbusClientRTU.cpp  [ 107] addRequestM: request for 01/03
[D] 47| ModbusClientRTU.cpp  [ 156] addToQueue: RC=01
[D] 52| ModbusClientRTU.cpp  [ 118] addRequestM: RC=00
[D] 128| ModbusClientRTU.cpp  [ 174] handleConnection: Pulled request from queue
[D] Sent packet: @3FFB9804/6:
  | 0000: 01 03 02 00 00 01                                 |......          |
[D] 137| ModbusClientRTU.cpp  [ 179] handleConnection: Request sent.
[D] Received packet: @3FFBDB4C/1:
  | 0000: E0                                                |.               |
[D] 2142| ModbusClientRTU.cpp  [ 185] handleConnection: Error response (1 bytes) received.
[V] Data: @3FFBDB4C/1:
  | 0000: E0                                                |.               |
[D] 2157| ModbusClientRTU.cpp  [ 214] handleConnection: Response generated.
[V] Response packet: @3FFBDB4C/3:
  | 0000: 01 83 E0                                          |...             |
handle error() function running on core: 0
[1;33m[E] 2177| Kellar_demo_1.ino    [  52] handleError: Error response: E0 - Timeout
[0m
________________________________________________________________________________

Miq1, 2021-07-12:
Wonderful, thank you.

The second request is ending in a TIMEOUT and may be a consequence of the first.

But the first has some remarkable oddities: the request correctly is asking for just one register at address 0x0200. 

The response is ``00 01 03 02 00 00 B8 44 00``, led in by a 00 that is not supposed to be there, since the first regular byte is 01, the server ID. Then follows the function code 03 - okay - and the next is supposed the length byte: 02, what is what we should expect,a single register has 2 bytes.The next two bytes are the register content 00 00, the next two bytes supposedly are the CRC. After that again comes a stray 00. 

So it looks like the server is enclosing a correct message in 00 bytes - not standard and driving a standard client nuts... The message to a conforming client looks like "server id 00, function code 01, 2 coils to be expected.  Both are 0, the CRC of 00 B8 is wrong and ooops: another two bytes?!?" 

I would suggest to refer to the sensor's specifications again to see if this behaviour may be corrected by configuration.

Ebrahim1604, 2021-07-12:
Thanks for the inputs @Miq1 , so I'll try to list out what I found:

Since I'm trying to read 0x0200 which is basically the Serial configuration register & for 9600 baud rate and 8N1 serial configuration, the output data should be 2 bytes with value 00 00.

So, that makes our frame as: 

> 01 03 02 00 00 CRC_L  CRC_H

After calculating the crc its value comes out as 0x44B8, so the exact expectd response frame should be:

> 01 03 02 00 00 B8 44

Now comparing this with our actual response:

> 00 01 03 02 00 00 B8 44 00

So as you suggested, these 00 bytes before & after are responsible for this mess. I'm not sure why exactly these bytes are being sent before & after the actual frame by the sensor. Is there anything I can do about it?

Miq1, 2021-07-12:
I simply do not know, I am afraid. These 00 bytes must be generated by the sensor for some purpose, so it would be interesting to learn the rationale behind. Even better would be some means to have these bytes omitted.

I will have a look at the sensor specs you have linked above to find a clue, but would ask you to do the same or even contact the manufacturer regarding that. The response is not a valid Modbus message, so if they claim to support Modbus, they must do better 😁

Miq1, 2021-07-12:
The documentation is okay, there the Modbus messages are described as according to the standard. I found no indication whatsoever that the framing 00 bytes are intentional.
If I was concerned, my next step would be to attach a logic analyzer to the sensor side serial interface, to see if the 00 bytes do come together with the message proper or are artifacts.
I suppose you do have a proper RS485 termination with resistors? Usually on the client end B- is connected by a 5K resistor to GND, A+ with another 5K resistor to Vcc. Between A+ and B- an 120 Ohms resistor is recommended.

Miq1, 2021-07-22:
Closing this again - the user's problemsseems to be solved.

KevinDesbonnets, 2021-08-19:
Hello (yes it's me again), I have the same issue with another modbus device

ie: 
[image]

I'm also on even parity (8E1), but even with none (8N1), the two 00 bytes occurs, i'm looking forward to solve this right now

Miq1, 2021-08-19:
The server seems to send the two 00 bytes like a sort of wrapper. Parity is not involved, as the parity bits are stripped off on lower protocol levels.

KevinDesbonnets, 2021-08-19:
How to get trough this wrapper? I checked configuration of the external mb system and I can do nothing except change baudrate parity and stop bits

Miq1, 2021-08-19:
Logic analyzer time again, I am afraid... Can you try to catch the native message the server is sending, to sort out potential library issues?

moebassist, 2021-08-19:
I think the NULL bytes (00) are misleading.

ModSim works fine and responds perfectly when these are present in the incoming message....however, if a modbus server is inserting NULL bytes in the reply, this library quite rightly fails because the Modbus packet should start with the responding address and end with a checksum.

Perhaps a 'fix' for this is to modify RTUutils::receive to ignore prefixed NULL bytes, and flush after checksum?

KevinDesbonnets, 2021-08-19:
Okay, other issue, the analyzer don't work anymore, I come back a bit later to make it work
[image]

Miq1, 2021-08-19:
> I think the NULL bytes (00) are misleading.
>
> ModSim works fine and responds perfectly when these are present in the incoming message....however, if a modbus server is inserting NULL bytes in the [...]

That would cover a symptom and not solve the root cause, I am afraid. The Modbus standard is the guideline for our library and does not mention any lead-in or lead-out zero bytes at all. I rather would like to learn what is causing these bytes. If it is a non-standard sensor, we may consider adding an option to cope with these; it also may be a fault of the lib I would like to weed out in case.

KevinDesbonnets, 2021-08-19:
I just made a quick patch

```cpp
// Unwrap unnecessary 00 Bytes
uint16_t ModbusMessage::unwrap() { 
  MM_data.erase(MM_data.begin());
  MM_data.pop_back(); 
  return MM_data.size(); 
}
```

I will test a bit later with logic analyzer

Ebrahim1604, 2021-08-19:
Hello @Miq1 @KevinDesbonnets 

I almost forgot about this thread until I received a notification again today, I would like to add an update on how I tackled this issue.

@Miq1 had suggested to check the termination resistors whether if they were in place or not, turns out they were not for the converter I was using. But the issue remained even after adding those resistors. 

I had no choice but to modify the Rtuutils.cpp file to filter out unnecessary stray bits. But then again, the error was not reproducible in all the readings. As opposed to only getting 00 bytes before and after the actual readings, these are the packets i received after trying to read data from other addresses:

> 00 01 03 04 FF FF FF FF  FB A7 00
14 F8 00 01 03 04 FF FF  FF FF FB A7 00
00 00 00 00 00 01 03 02  00 0B F9 83 00
B8 36 01 03 02 1F 1E 30  7C 00
>>>

So @KevinDesbonnets I suggest you verify this too for multiple addresses of different data lengths before putting it in place for the final system.

At last I had to prepare a patch encompassing an algorithm which will filter out data in standard modbus format. I knew the received packet should be in following format always for the data bits at addresses I was requesting for:

> 01 03 0X 
where 0X = no of bytes, which in my case was always 2 or 4. So the rest of the packet would have either 2 or 4 bytes accordingly. 

So here is the modified Rtuutils.cpp file I'm currently using which works:

```
// =================================================================================================
// eModbus: Copyright 2020 by Michael Harwerth, Bert Melis and the contributors to eModbus
//               MIT license - see license.md for details
// =================================================================================================
#include "options.h"
#include "ModbusMessage.h"
#include "RTUutils.h"
#undef LOCAL_LOG_LEVEL
// #define LOCAL_LOG_LEVEL LOG_LEVEL_VERBOSE
#include "Logging.h"

// calcCRC: calculate Modbus CRC16 on a given array of bytes
uint16_t RTUutils::calcCRC(const uint8_t *data, uint16_t len) {
  // CRC16 pre-calculated tables
  const uint8_t crcHiTable[] = {
    0x00, 0xC1, 0x81, 0x40, 0x01, 0xC0, 0x80, 0x41, 0x01, 0xC0, 0x80, 0x41, 0x00, 0xC1, 0x81,
    0x40, 0x01, 0xC0, 0x80, 0x41, 0x00, 0xC1, 0x81, 0x40, 0x00, 0xC1, 0x81, 0x40, 0x01, 0xC0,
    0x80, 0x41, 0x01, 0xC0, 0x80, 0x41, 0x00, 0xC1, 0x81, 0x40, 0x00, 0xC1, 0x81, 0x40, 0x01,
    0xC0, 0x80, 0x41, 0x00, 0xC1, 0x81, 0x40, 0x01, 0xC0, 0x80, 0x41, 0x01, 0xC0, 0x80, 0x41,
    0x00, 0xC1, 0x81, 0x40, 0x01, 0xC0, 0x80, 0x41, 0x00, 0xC1, 0x81, 0x40, 0x00, 0xC1, 0x81,
    0x40, 0x01, 0xC0, 0x80, 0x41, 0x00, 0xC1, 0x81, 0x40, 0x01, 0xC0, 0x80, 0x41, 0x01, 0xC0,
    0x80, 0x41, 0x00, 0xC1, 0x81, 0x40, 0x00, 0xC1, 0x81, 0x40, 0x01, 0xC0, 0x80, 0x41, 0x01,
    0xC0, 0x80, 0x41, 0x00, 0xC1, 0x81, 0x40, 0x01, 0xC0, 0x80, 0x41, 0x00, 0xC1, 0x81, 0x40,
    0x00, 0xC1, 0x81, 0x40, 0x01, 0xC0, 0x80, 0x41, 0x01, 0xC0, 0x80, 0x41, 0x00, 0xC1, 0x81,
    0x40, 0x00, 0xC1, 0x81, 0x40, 0x01, 0xC0, 0x80, 0x41, 0x00, 0xC1, 0x81, 0x40, 0x01, 0xC0,
    0x80, 0x41, 0x01, 0xC0, 0x80, 0x41, 0x00, 0xC1, 0x81, 0x40, 0x00, 0xC1, 0x81, 0x40, 0x01,
    0xC0, 0x80, 0x41, 0x01, 0xC0, 0x80, 0x41, 0x00, 0xC1, 0x81, 0x40, 0x01, 0xC0, 0x80, 0x41,
    0x00, 0xC1, 0x81, 0x40, 0x00, 0xC1, 0x81, 0x40, 0x01, 0xC0, 0x80, 0x41, 0x00, 0xC1, 0x81,
    0x40, 0x01, 0xC0, 0x80, 0x41, 0x01, 0xC0, 0x80, 0x41, 0x00, 0xC1, 0x81, 0x40, 0x01, 0xC0,
    0x80, 0x41, 0x00, 0xC1, 0x81, 0x40, 0x00, 0xC1, 0x81, 0x40, 0x01, 0xC0, 0x80, 0x41, 0x01,
    0xC0, 0x80, 0x41, 0x00, 0xC1, 0x81, 0x40, 0x00, 0xC1, 0x81, 0x40, 0x01, 0xC0, 0x80, 0x41,
    0x00, 0xC1, 0x81, 0x40, 0x01, 0xC0, 0x80, 0x41, 0x01, 0xC0, 0x80, 0x41, 0x00, 0xC1, 0x81,
    0x40
  };

  const uint8_t crcLoTable[] = {
    0x00, 0xC0, 0xC1, 0x01, 0xC3, 0x03, 0x02, 0xC2, 0xC6, 0x06, 0x07, 0xC7, 0x05, 0xC5, 0xC4,
    0x04, 0xCC, 0x0C, 0x0D, 0xCD, 0x0F, 0xCF, 0xCE, 0x0E, 0x0A, 0xCA, 0xCB, 0x0B, 0xC9, 0x09,
    0x08, 0xC8, 0xD8, 0x18, 0x19, 0xD9, 0x1B, 0xDB, 0xDA, 0x1A, 0x1E, 0xDE, 0xDF, 0x1F, 0xDD,
    0x1D, 0x1C, 0xDC, 0x14, 0xD4, 0xD5, 0x15, 0xD7, 0x17, 0x16, 0xD6, 0xD2, 0x12, 0x13, 0xD3,
    0x11, 0xD1, 0xD0, 0x10, 0xF0, 0x30, 0x31, 0xF1, 0x33, 0xF3, 0xF2, 0x32, 0x36, 0xF6, 0xF7,
    0x37, 0xF5, 0x35, 0x34, 0xF4, 0x3C, 0xFC, 0xFD, 0x3D, 0xFF, 0x3F, 0x3E, 0xFE, 0xFA, 0x3A,
    0x3B, 0xFB, 0x39, 0xF9, 0xF8, 0x38, 0x28, 0xE8, 0xE9, 0x29, 0xEB, 0x2B, 0x2A, 0xEA, 0xEE,
    0x2E, 0x2F, 0xEF, 0x2D, 0xED, 0xEC, 0x2C, 0xE4, 0x24, 0x25, 0xE5, 0x27, 0xE7, 0xE6, 0x26,
    0x22, 0xE2, 0xE3, 0x23, 0xE1, 0x21, 0x20, 0xE0, 0xA0, 0x60, 0x61, 0xA1, 0x63, 0xA3, 0xA2,
    0x62, 0x66, 0xA6, 0xA7, 0x67, 0xA5, 0x65, 0x64, 0xA4, 0x6C, 0xAC, 0xAD, 0x6D, 0xAF, 0x6F,
[... 336 lines cut ...]
        }
        state = FINISHED;
      }
      break;
    // FINISHED: we are done, clean up.
    case FINISHED:
      // CLear serial buffer in case something is left trailing
      // May happen with servers too slow!
      while (serial.available()) serial.read();
      break;
    }
  }
  // Deallocate buffer
  delete[] buffer;

  HEXDUMP_D("Received packet", rv.data(), rv.size());

  return rv;
}

```

I'm aware this is going against modbus standards, but the problems is I am remotely working on this one and can't arrange for a logic analyzer and employ it with the person hosting the setup.

@KevinDesbonnets  you can test the file for multiple cases and let me know if there any issue.

KevinDesbonnets, 2021-08-19:
I tested the file with a simple READ_HOLD_REGISTER function on the modbus device and it's work well, maybe make a pull request of it?

I will test further this patch along with my project thanks you

KevinDesbonnets, 2021-08-19:
I checked the code a bit, it's will work with other function than 0x3? I don't think so...

I will this as soon as I can and make some change if necessary

Ebrahim1604, 2021-08-19:
Currently it only works for 03 function code, that too with either 2 or 4 data bytes.

You can modify that in the clean_data function.

Im guessing for making a pull request there needs to support for all the function codes and it might require many more changes pertaining the lenght of the final buffer.

Miq1, 2021-08-19:
Sorry, I am not going to accept a pull request that will catch a single case only and not tackle the root cause.

The stray byte issue must have special preconditions to happen that the both of you unfortunately seem to have met. Else we would have had more reports alike. So I would rather attempt to find what your projects do have in common.

KevinDesbonnets, 2021-08-19:
I read the code a bit further and if you use other function you can have a UB I think

```cpp
for (int i=0; i<bufferPtr; i++)
    {       
        if (buffer1[i]==0x01 && trigger1==0)
        {
            temp1 = buffer1[i];
            trigger1 = 1;
        }
        
        else if (trigger1 == 1 && count2==0)
        {
            if (buffer1[i-1]==0x01 && buffer1[i]==0x03)
            {
                temp2 = buffer1[i];
            }
            
            else if(buffer1[i-2]==0x01 && buffer1[i-1]==0x03)
            {
```

If you use function 0x04 for exemple you will check `buffer1[i-2]==0x01` when i == 1

Ebrahim1604, 2021-08-19:
@Miq1 that's fair enough.

The only reason I'm sharing this here is to have a temporary fix of the problem, merging it with the main branch isn't a feasible solution.

@KevinDesbonnets if you have the time and resources, it would be better if you debug the root cause.

KevinDesbonnets, 2021-08-19:
Right now I will just use a patch for my case but I will continue a bit and if I find it I will come back to share it :)

Miq1, 2021-08-19:
- Can you leave out the RS485 hardware for a while and use the Serial ports only?
- can you connect your respective servers to a PC and Modsim64 or similar to see if the issue is still there? 
- can you intercept the signal with a scope or logic analyzer as close at the server as possible? 

The bytes are shown with the log output immediately upon receipt from Serial, so chances are high that they indeed have been inducted on the Serial line, the RS485 line or are originating from the server itself.

Miq1, 2021-08-21:
Guys, anything new? Do you have a wiring diagram at hand you can share?

Ebrahim1604, 2021-08-22:
@Miq1 I have asked the person hosting the setup to do the above tasks but it might take a while.

Btw @KevinDesbonnets, which company sensor are you using here? More details would definitely help.
Also which rs485 to ttl converter are you using?

I feel we can close this issue for now as it seems to be not a library issue, we will update here as we have any progress.

Miq1, 2021-08-22:
Thanks, @Ebrahim1604!

I will leave this open as a question - maybe someone else is having a similar issue and can add some information.

Miq1, 2021-09-12:
I got another user with nearly the same issue, but the 0x00 bytes coming in separated from the message proper.

I found an interesting [report](https://www.avrfreaks.net/forum/rs485-framing-error-when-toggling-re-pin) at AVRfreaks that sounds quite like it could point to a cause.

The other user but checked the wiring on the RS485MAX adaptor he is using and according the schematic the RO line should be pulled up as advised - but who knows what the Chinese manufacturer did...

I would ask you to add a ``pinMode(RX_PIN, INPUT_PULLUP);`` to your code and see what happens. ``RX_PIN`` must be your GPIO you are using for RX of course.

moebassist, 2021-09-12:
Also….do you have 10k bias resistors on both rx/tx lines? If not, noise on the line might trigger a receiver.

Miq1, 2021-09-12:
> Also….do you have 10k bias resistors on both rx/tx lines? If not, noise on the line might trigger a receiver.

According to the schematics all four RX/TX/DE/RE lines do have the pullups, but you never know... The coincidence of the error scheme is striking.

bertmelis, 2021-09-12:
I always used the schematic as drawn in the old repo:
https://github.com/bertmelis/esp32ModbusRTU

It shows the biasing resistors, terminating resistor and ground loop.

It's only three cheap components more, so I added them.

Miq1, 2021-11-11:
Closing this now in regards to [discussions item #112](https://github.com/eModbus/eModbus/discussions/112).

closed 2021-11-11 by Miq1