eModbus/eModbus #41: Unable to use "ModbusBridgeRTU" (Rx packet not recived)

francescogirotto, 2021-01-03:
I cannot use the "ModbusBridgeRTU", the packets are sent correctly but the reply message that is sent by the TCP server does not reach the RTU client.
Please help me, and congratulations for this very useful library!!

MY CODE:

```
  #define LOCAL_LOG_LEVEL LOG_LEVEL_6
  #include "Logging.h"
  #include "modbusBrigeTasks.h"
  
  char ssid[] = "****";                     // SSID and ...
  char pass[] = "****";                     // password for the WiFi network used

  IPAddress serverIP = {192,168,43,100};
  uint16_t port = 502;                      // port of modbus server
  
  // Create a ModbusTCP client instance
  WiFiClient theClient;
  ModbusClientTCP myClient(theClient,5000);
  // Create server
  ModbusBridgeRTU myBridge(Serial2, RS485_EN);
  
//function in the setup()
  void modbusTasks_begin(){
  
      // Connect to WiFi
      WiFi.begin(ssid, pass);
      delay(200);
      while (WiFi.status() != WL_CONNECTED) {
          Serial.print(". ");
          delay(500);
      }
      IPAddress wIP = WiFi.localIP();
      Serial.printf("WIFi IP address: %u.%u.%u.%u\n", wIP[0], wIP[1], wIP[2], wIP[3]);
  
      // Set TCP Modbus message timeout to 2000ms
        myClient.setTimeout(2000,0);
  
      // Start ModbusTCP background task
        myClient.begin();
  
      // Define and start RTU bridge
        myBridge.attachServer(1, 1, ANY_FUNCTION_CODE, &myClient, serverIP, port);
  
      // Check: print out all combinations served to Serial
        myBridge.listServer();
  
      // Start the bridge. Port 502, 4 simultaneous clients allowed, 600ms inactivity to disconnect client
        myBridge.start();
  
  }
```

There is also a video of the test;

https://user-images.githubusercontent.com/76907794/103480889-89caaa80-4dd7-11eb-83c9-da145e1a77c7.mp4

Miq1, 2021-01-03:
Please have a look at [RTUutils.cpp:191](https://github.com/eModbus/eModbus/blob/1e63c9996b3c8cd513d7a0335ce924f88057c2dd/src/RTUutils.cpp#L192)!

I suppose you have *not* modified the arduino-esp32 core as advised there. Please try it (or uncomment the workaround line  in RTUutils.cpp instead) and report back your results.

Miq1, 2021-01-03:
Oh, by the way... This:
```
#define LOCAL_LOG_LEVEL LOG_LEVEL_6
```
will not work. The correct line would be:
```
#define LOCAL_LOG_LEVEL LOG_LEVEL_VERBOSE
```
😉

bertmelis, 2021-01-03:
> Please have a look at [RTUutils.cpp:191](https://github.com/eModbus/eModbus/blob/1e63c9996b3c8cd513d7a0335ce924f88057c2dd/src/RTUutils.cpp#L192)!
>
> I suppose you have _not_ modified the arduino-esp32 [...]

If this works, we maybe can create an automated way of patching this file using Platformio: https://docs.platformio.org/en/latest/projectconf/advanced_scripting.html#override-package-files

Miq1, 2021-01-03:
@bertmelis : yes, I think that would be the only way. The maintainers of the arduino-esp32 core did not react at all on my request to have it at least configurable in the core.

francescogirotto, 2021-01-03:
Nothing to do, like before, changing "uart-> dev-> conf1.rxfifo_full_thrhd" to 1 still doesn't work.

[image]

What I have to do here? 
If I uncomment "if (micros() - MR_lastMicros >= 16000) {" there isn't the final "}", and "MR_lastMicros" isen't delared 

[image]

Miq1, 2021-01-03:
You will have to comment the ``if (nicros() - lastMicros >= interval) {`` line, if you are going to use the fallback.

But if your change to the core file has no effect, the problem will be elsewhere.

Can you please use the `build_flags = -DLOG_LEVEL=LOG_LEVEL_VERBOSE` line in your `platformio.ini` and show us the output of the failing response message?

francescogirotto, 2021-01-03:
Of course!!

[image]

Miq1, 2021-01-03:
This is a perfect output for what you want to achieve! The response has been sent obviously to the requester - are you able to have a closer look at what the RTU client is doing? It looks like the packet is lost on the receiving side. Are you sure to have both TX and RX wires connected correctly? 

From all what I see, the bridge is working as supposed.

francescogirotto, 2021-01-04:
Today I checked with the oscilloscope the communication in the TX, RX, RTS pins which go to the RS485 transmitter.
I have found that the RTS pin does not switch HIGH when the TX packet is transmitted and therefore the transmitter does not transmit data.
The RTS pin is well declared in "ModbusBridgeRTU myBridge (Serial2, RS485_EN);" (RS485_EN is the pin 4 os my esp32-wroom-32d)
Here I attach a screenshot of the test done with the oscilloscope:
- the blue signal is the TX pin
- the red signal is the RTS pin

[image]

I don't know how to fix it :(

Miq1, 2021-01-04:
That reminds me of something... Would you mind downloading a current copy of our library? I fixed a function dealing with the RTS pin a few days ago. The reason was, that the pin was pulled to HIGH, but only for milliseconds.

francescogirotto, 2021-01-04:
I found the problem!

I changed the default timeout and TOV of the RTU bridge:
 "ModbusBridgeRTU myBridge(Serial2, RS485_EN);" -> "ModbusBridgeRTU myBridge(Serial2, 3000, 3000, RS485_EN);"
and I have calibrated the TCP client timeouts:
 "myClient.setTimeout(2000,0);" -> "myClient.setTimeout(500,100);"

Now it work well!

Miq1, 2021-01-04:
Good to read, but "ouch!" - I missed that bit with the constructor call `ModbusBridgeRTU myBridge(Serial2, RS485_EN);` in your code - that could not work at all! With that call you gave a timeout of 4ms and no rtsPin - this would never have worked.

closed 2021-01-04 by Miq1