eModbus/eModbus #43: Publish eModbus to PlatformIO registry

Miq1, 2021-01-04:
We need to have the eModbus lib listed in PlatformIO - [Registry](https://docs.platformio.org/en/latest/core/userguide/package/cmd_publish.html#cmd-package-publish)

bertmelis, 2021-01-12:
Do you have a pio account?

As you are this orgs admin, maybe you can do `pio package publish`
https://docs.platformio.org/en/latest/core/userguide/package/cmd_publish.html

Miq1, 2021-01-12:
Yeah, will try that. Is ``library.json`` complete, up-to-date and formally okay for that? I never dealt with the pio innards before...

Miq1, 2021-01-12:
Did a publish - we'll see how that processes. I will keep you posted.

bertmelis, 2021-01-12:
> Yeah, will try that. Is `library.json` complete, up-to-date and formally okay for that? I never dealt with the pio innards before...

Shoot, it's not. There's a mismatch between Github's release version and the version in the library.json/library.properties. But if we update, then we have to change the GH release also.

Miq1, 2021-01-12:
I changed the internal version in ``library.json`` to ``1.0.0``, so it may come out okay.

bertmelis, 2021-01-13:
https://platformio.org/lib/show/11682/eModbus

closed 2021-01-13 by bertmelis