eModbus/eModbus #44: Pointer out of the heap

francescogirotto, 2021-01-05:
Do you know why when I have an error in the translation ID in modbus TCP sometimes a pointer goes out of the heap area and the ESP32 reboot?

[image]

Miq1, 2021-01-05:
Not right away...

Can you put this:
```
monitor_filters = esp32_exception_decoder
```
in your ``platformio.ini`` file and try again? This will give us a readable backtrace to see what originally was causing the exception.

Miq1, 2021-01-05:
@francescogirotto : Any news?

Miq1, 2021-01-12:
@francescogirotto : may we assume the bug was caused from another source? Then I would like to close this issue.

Miq1, 2021-01-22:
Closing this for now.

closed 2021-01-22 by Miq1