eModbus/eModbus #48: Create diff/patch file for RTU requirements

bertmelis, 2021-01-12:
The RTU version requires a feature currently not available in the Arduino framework. A workaround is suggested in the documentation but this requires changing Arduino source files. (https://emodbus.github.io/requirements#important-note-for-using-the-rtu-bridge-server-and-client)

Platformio IDE has the possibility to update the file automatically: https://docs.platformio.org/en/latest/projectconf/advanced_scripting.html#override-package-files

Feature: create sample scripts for eModbus.

bertmelis, 2021-01-12:
@Miq1
Correct me if I'm wrong

current code requires the core file to be patched, right?

and if not patched, then you'd have to do this:
```cpp
      // Gap of at least _interval micro seconds passed without data?
      // ***********************************************
      // Important notice!
      // Due to an implementation decision done in the ESP32 Arduino core code,
      // the correct time to detect a gap of _interval µs is not effective, as
      // the core FIFO handling takes much longer than that.
      //
      // Workaround: uncomment the following line to wait for 16ms(!) for the handling to finish:
      if (micros() - lastMicros >= 16000) {
      //
      // Alternate solution: is to modify the uartEnableInterrupt() function in
      // the core implementation file 'esp32-hal-uart.c', to have the line
      //    'uart->dev->conf1.rxfifo_full_thrhd = 1; // 112;'
      // This will change the number of bytes received to trigger the copy interrupt
      // from 112 (as is implemented in the core) to 1, effectively firing the interrupt
      // for any single byte.
      // Then you may uncomment the line below instead:
      //if (micros() - lastMicros >= interval) {
        state = DATA_READ;
      }
```

Am I correct?

I propose to
- create a patch file to update the core file
- add some preprocessor to RTUUtils.cpp to select which workaround is used
- create the python scripts as per platformio's docs
- create an example with sample platformio.ini that includes the scripts and preprocessor

Miq1, 2021-01-12:
You are completely right. Your proposals sound good to me as well - go ahead, I would say.

bertmelis, 2021-01-12:
What should be the default? Patch or the workaround?

Asking for the preprocessor in our source.

Miq1, 2021-01-12:
Workaround, although I do not like it. We must cope with people not willing or able to modify their cores, I am afraid.

bertmelis, 2021-01-12:
Got it in https://github.com/eModbus/eModbus/tree/uart-patch

Don't know if it works.

Miq1, 2021-01-12:
Compilation of RTUutils.cpp fails - something with the if construct braces, I reckon.

bertmelis, 2021-01-12:
ah, not familiar with preprocessor... fixed?

Miq1, 2021-01-12:
Most probably. I was editing it at the same time as you did, but pulled the short straw 😁 

I had enclosed it by ``#if RXFIFO_FULL_THRHD_PATCHED == 0``, then the workaround, then ``#else`` and the patched version, ``#endif``, but that may come out similarly.

bertmelis, 2021-01-12:
yeah, i didn't want to interfere in the comments. depending on the settings, large parts are greyed out making it hard to read on my screen.

Miq1, 2021-01-12:
👍

bertmelis, 2021-01-12:
OK to (squash) merge?

Miq1, 2021-01-12:
Wait a second. I shall build and run the test suite first.

bertmelis, 2021-01-12:
Mind that the github actions build might not work as expected. I don't know if the custom platformio.ini is taken into account. But it does build.

Miq1, 2021-01-12:
I did not test the patch, but the patched library - and it obviously still is running correctly. So go ahead and 🎃 !

Miq1, 2021-01-12:
In case you wondered about the Emoji: [Wikipedia](https://en.wikipedia.org/wiki/Butternut_squash) 🤣

closed 2021-01-12 by bertmelis