eModbus/eModbus #50: CodeQL is not able to scan

Miq1, 2021-01-12:
The default CodeQL scan script is failing due to being unable to build a target. 

I have no experiences with the Github actions, so could you please have a look, @bertmelis ?

Miq1, 2021-01-12:
I was wondering if you could do the same here that you did for the build test script? If the Examples will build, almost all of the library will be included. May be that is sufficient for CodeQL?

bertmelis, 2021-01-13:
Awaiting some progress here, I did run `cppcheck` and no warnings were given except for the magic in the byte order determination functions. `cppcheck` complains about taking the address of a variable at non zero indexes. But there it is on purpose...

bertmelis, 2021-01-29:
CodeQL is not going to work soon I'm afraid

If we want to have something similar, we might have to look into alternatives:
- cpplint
- cppcheck
- Arduino-lint (something new, didn't look into this yet): https://github.com/arduino/arduino-lint
- https://www.codacy.com/
- https://bettercodehub.com/
- ...

Leaving this road is also an option.

Miq1, 2021-01-29:
I think we are both using ``cpplint`` and ``cppcheck`` with PlatformIO - aren't we? So I would think it is not worth further effort (yet). My suggestion is to trash the CodeQL script. 

Do you agree?-

bertmelis, 2021-01-29:
ok, I can integrate cpplint and cppcheck into github actions if needed. I suppose it's not high on the priority list.

Miq1, 2021-01-29:
👍

Miq1, 2021-01-29:
Case closed 😁

closed 2021-01-29 by Miq1