eModbus/eModbus #51: Support For Simplified Examples

parthbhat13, 2021-01-15:
Hi,
I was using and following the older modbus library by mr.bert, but had faced a tons of issues figuring out to read the messages and make it more literate. now that i am using the emodbus, trust me im tired to figure out how to have a basic example where i can read few holding registers from a slave device, which are in 32bit floating format. perhaps the slave is also an arduino device and the code is written by me, it is reading data on the master software on laptop, but i am tired to figure out how on earth am i supposed to be combining the way too literate documentation for decoding the message. yes i am getting the data, but how on earth can i make it just read a simple message "25.786" thats the temperature reading on holding register 40000 and 40001 .. its split into 2 16bit int. 

humble request to add some really basic and literate examples in the code so that it is useful. 

here is my slave code..
[https://pastebin.com/LUs2NhXR](url)

Here is the ESP32 Master Code (Which gives data, but not a readable one)
[https://pastebin.com/bsCUdysP](url)

apologies if i sounded rude, but its just whole 3 days of frustration coming out as i couldn't figure out any way to make this work.

thanks in advance.

Miq1, 2021-01-15:
I know how you are feeling, trust me, but "there ain't no such thing as a free lunch" ... 

Having said that, I would propose to first take one of our examples for function code 0x03, READ_HOLD_REGISTER, make sure that your wiring is functional, adjust the settings (addresses, register counts etc.) to fit your server and give it a go.

The float value will come in two adjacent uint16_t words, so the clue will be to use the ``get()`` function  of the ModbusMessage  to take the float in the right byte order - that is depending on your server's hardware.

bertmelis, 2021-01-15:
I didn't look into your code yet, but you want a basic example of a client? or do you need a server? And you want to read a 32 bit value, which is coded as IEEE 754 floating point?

parthbhat13, 2021-01-15:
> I know how you are feeling, trust me, but "there ain't no such thing as a free lunch" ...
>
> Having said that, I would propose to first take one of our examples for function code 0x03, [...]

well, thanks a lot for the prompt reply, and trust me, my father have always taught me, and so do i follow doing everything as much as possible by searching and working hard over it but this is more of me stuck with no way out so finally i had to bug you guys, defiantly i aint gonna ask for free lunch, but rather would ask for guiding me to make a delicious lunch which you all know how to cook haha. 

about what you suggested, yes, i did check my wiring , and i did get the data, which is something like this

`[N] 923| sketch_jan15a.ino    [  29] handleData: Response: serverID=12, FC=3, Token=00000457, length=15:`
`[N] Data: @3FFB9818/15:`
 `| 0000: 0C 03 0C B8 50 41 E8 70  A0 41 DF 80 00 41 B0     |....PA.p.A...A. |`
 
 but i am stuck how should i decode this value and make it more useful for my usecase. 
 now ill follow the clue to use the `get()` function and see how that helps me. and i will be adding some examples once i am done making it work if you guys dont mind. 
 
 and @bertmelis , i need example of a client (master hope im right when i say that) , and yes i am reading a 32bit floating value. which is indeed coded as IEEE754 i believe (I'm not so sure about this tho)
 
 thanks a lot again for baring with me and for the prompt replies.

Miq1, 2021-01-15:
You are obviously reading three float values here. It looks like the server is doing a SWAP_REGISTER for floats, so the ``get()`` calls could be ``response.get(3, flt1, SWAP_REGISTER);``,  ``response.get(7, flt2, SWAP_REGISTER);`` and ``response.get(11, flt3, SWAP_REGISTER);``.

parthbhat13, 2021-01-15:
> You are obviously reading three float values here. It looks like the server is doing a SWAP_REGISTER for floats, so the `get()` calls could be `response.get(3, flt1, SWAP_REGISTER);`, `response.get(7, flt2, SWAP_REGISTER);` and `response.get(11, flt3, SWAP_REGISTER);`.

WOOHOOOO YES IT WORKS, I LOVE YOU FOR THAT ONE.. 

now, honestly, thanks a ton for that, but, how did you understand that its a 3 float values? how you decided to use the swap_register. i would really like to understand and build my knowledge on this one please. 
seriously thanks a ton again.

Miq1, 2021-01-15:
If you have seen some IEEE754 floats, you will know that these always will have the same pattern (have a look at the really enlightening Wikipedia article on IEEE754) in front. The '41' byte looks like it, and it always is on the third byte of the four of a float. So the first assumption was that the two-byte parts of the float were swapped. Luckily this first guess seems to be right 😄

Miq1, 2021-01-15:
Looks like being solved. 😉

closed 2021-01-15 by Miq1

parthbhat13, 2021-02-02:
Hey,
I really feel ashamed of myself to again ask for help, but i'm really facing issues with conversions and my math is really weak. i want to understand, im getting data and im even getting the numbers, for example ..

`Data: @3FFB97E8/5:
  | 0000: 03 03 02 00 34 `
  
  the last 0034 is my hexadecimal format , and once i convert it to decimal (Online converter) the values are right. 
  now i used 
`byte word1;`
`response.get(4,word1)`

now im actually confused how can i really use this, as usually converting the whole hex to decimal is a different thing,  the arduino codes and everything uses string or char so on. but here im using byte and i doubt it will be a right thing to do as later when the code changes to some other value like 4A,4B so byte wont be able to handle it i believe? 
im confused where i am going wrong here?

I am really so sorrry for bothering you guys again, but i had no other way out. 
thanks in advance

bertmelis, 2021-02-02:
You might as well stop using byte, int, long etc and use uint16_t (unsigned integer of 16bits) or int16_t (signed 16 bit integer). It makes it easier to understand.

In your example you transform a 2 byte number to a 1 byte number (even though you call it "word1"). Something doesn't fit, right?

See 
https://en.cppreference.com/w/cpp/types/integer

Miq1, 2021-02-02:
I might add that Modbus generally is mostly using 16-bit values (a "register" in the Modbus language is an unsigned 16-bit variable).

To get such a register from a message you will have to keep in mind that 16 bits is 2 bytes (2 times 8 bits), so you will have to adjust your offset as well:
```
uint16_t word1;
response.get(3, word1);
```

parthbhat13, 2021-02-02:
thanks alot for the prompt response from both of you. 
really thank you very much. 

the thing is, this is a pure Chinese made sensor, and they arent following that much of standards of the value. i will still check what you asked me to do. thanks a ton again.

Miq1, 2021-02-02:
Would you mind providing us a link to the sensor in question? If there is some specification, we may see if it differs from the Modbus standards.

parthbhat13, 2021-02-02:
well,  i need to maintain some client protocols. so i will delete after you guys read it. and ill share the pdf over here. 
as well, i did both ways, byte one and by using uint16_t , i am getting correct values when i use `Serial.print` . but not when im using the `ESP_LOGI` or `serial.printf` i can still see the hex values. 
now im wondering am i supposed to be converting them, or is it just the print functions as i have to finally send them over mqtt. so now im writing the code for that. 

[Soil NPK Sensor.pdf](https://github.com/eModbus/eModbus/files/5912436/Soil.NPK.Sensor.pdf)

Miq1, 2021-02-02:
I got the PDF, thanks. Looks pretty straightforward to me.

As there are effectively three registers only with continuous addresses, I would propose to read those in one call
``` 
uint32_t token = millis();   // or whatever you like to have as a token
client.addRequest(token, 0x01, READ_HOLD_REGISTER, 0x001E, 3); // Three registers, starting at 0x001E
```
The response will have all three values. You may retrieve these like
```
uint16_t nitrogen, phosphorous, potassium;
response.get(3, nitrogen);
response.get(5, phosphorous);
response.get(7, potassium);
```
The three ``uint16_t`` variables will now hold the mg/kg values. You may print it like
```
Serial.printf("N=%d, P=%d, K=%d\n", nitrogen, phosphorous, potassium);
```
Remember that the format (like ``%d``) in printf() will control how you will see the contents of the variables.

parthbhat13, 2021-02-02:
Alright I will follow this, thank you again. 
I really learn a lot from you thank you. 

I wonder what's the use of token here ? 
And I'm running Rtos , so do I need to force the cpu core in the 'mb.begin()' 

Thanks a lot again.

Miq1, 2021-02-02:
The token is meant for you to be able to identify the request in the onData and onError handler function. This does not really matter for your application, but will be necessary as soon as you will use multiple clients in an asynchronous fashion.
You can use another core (I suppose you are using an ESP32? Then it should be core ``1``) if you get into communication timing issues. I reckon as long as your baud rates are below 115200, this will not easily happen.

parthbhat13, 2021-02-02:
Thanks for the reply, 
Well now I'm a bit worried as yes I'll be having around 15 such sensors so I hope things go according to the plan. 

And yeah baudrate is 9600 not more than that.

parthbhat13, 2021-04-02:
Hey, 
hope you all are doing great. 
i was trying to read coils with the library and there are no examples i can see. 
i am getting the right data when i wanna read around 5 quantity. 
but in the hex format i cannot see the individual bytes like i get for holding or input. 

all i get is the following `01 01 01 04` , when i change one of the coil it will go from 04 to 05 to 07 and so on. but how can i read individual coils? and put that in the `response.get` ?
thanks in advance

Miq1, 2021-04-02:
Coils are single bits basically, so you will have to mask the bytes you got on a read. 0x04 f.i. means a single bit (coil) is set, because 0x04 means 00000100 binary. If you will set coil (bit) 0, you will have 00000101 binary, which is 0x05 in hexadecimal.

parthbhat13, 2021-04-02:
> Coils are single bits basically, so you will have to mask the bytes you got on a read. 0x04 f.i. means a single bit (coil) is set, because 0x04 means 00000100 binary. If you will set coil (bit) 0, you will have 00000101 binary, which is 0x05 in hexadecimal.

okay i get that, but how am i supposed to fetch that in the response.get? or i cannot use that for this situation?

Miq1, 2021-04-02:
You will need to get() uint18_t byte values - one by one if there are more than 8 coils. C++ has no standalone "bit" data type, so there is no way around separating bits yourself. You might have a look at the binary C++ operators ``|`` and ``&`` to learn how to separate and change single bits 😉

parthbhat13, 2021-04-02:
im still confused with what am i supposed to do with the `response.get` but let me try what you just said and get back to you. thank you so very much for always being so helpful.