eModbus/eModbus #52: ModbusServerTCPasync does not react on 8-byte request

Miq1, 2021-01-17:
I am using a USER_DEFINED_42 function code on my smart plug's ModbusServerTCPasync to trigger writing data to EEPROM.
The TCP request packet is like 
```
00 01 00 00 00 02 01 42
```
This is just the FC, no more data, and a valid Modbus request AFAIK.

The ModbusServerTCPasync does not react on it at all, though. If I will add another (illegal) byte to the request:
```
00 01 00 00 00 02 01 42 01
```
the request will be processed, ignoring the obsolete byte.

I must confess I did not find where this may be caused, so I hand it over to you, @bertmelis 😁

bertmelis, 2021-01-17:
Found the problem I think.

When entering `onData` with a 8 bytes, the while-loop will not be executed anymore after the 8th iteration because `i` is 8 and `len` is also 8. so `i < 8` is not valid anymore.

Could be solved by not using the outer while loop to fetch the first 8 bytes but do it like this (line 46):
```
    //  1. get minimal 8 bytes to move on
    while (message->size() < 8 && i < len) {
      message->push_back(data[i++]);
    }
```
Then you do not need to enter the outer while loop anymore.

I can't test myself now, tomorrow I'm able to.

bertmelis, 2021-01-18:
Ok, proposed solution works:
[image]
[image]

bertmelis, 2021-01-18:
I've also updated the version to 1.0.1. I didn't create a new release though.

closed 2021-01-18 by bertmelis

Miq1, 2021-01-18:
👍

Is the version automatically updated at the PIO registry, or will I have to publish it again?

bertmelis, 2021-01-18:
I actually don't know. I think it is updated automatically. I used to think it referred to the release-tag of the git repository. But that's a mere textual tag. So I don't know for sure.