eModbus/eModbus #63: CRC check error.

fahadalisarwar1, 2021-02-10:
I am using modbus RTU. When I read 2 or 4 input registers my program works ok. But For my application I need to access data which is located in reg 13 and 81. SO I tried to make a request with 100 as number of registers to read. However, I am getting CRC check error. 
[image]

i only have one function "onData" to read. Can I make saperate calls for different register numbers ?

Miq1, 2021-02-10:
Can you please compile your program again with LOG_LEVEL_VERBOSE and show us the output?

As a rule of thumb, it is much better to do two single register reads instead of this huge number of registers - you are wasting resources...
The ``onData`` routine will be called the same for both requests, it is your task to tell which request the response is belonging to. Hint: the ``token`` value you are providing with the ``addRequest()`` call will be returned to ``onData`` unmodified, so you may mark your requests with it 😉

fahadalisarwar1, 2021-02-10:
Yes. It was not a good idea to read 100 regs and use only 2. Found a solution with sending different tokens for different regs and then using a simple check for tokens in OnData routine. Thanks :)

fahadalisarwar1, 2021-02-10:
Will do it tomorrow ☺️

Miq1, 2021-02-10:
Thanks! 👍 

I am looking forward to it!

Miq1, 2021-03-12:
Looks like an issue on the user's side, therefore closing here

closed 2021-03-12 by Miq1