eModbus/eModbus #64: Handle data response dinamically

arangu97, 2021-03-01:
Hello, I am starting to use this library and I am wondering if there is a way to get the response after an addRequest().

I mean, the response is a ModbusMessage that you can print in the handleData() function, but could I get that ModbusMessage after the request in order to process it in a different way depending on the register that I have read?

Thank you.

bertmelis, 2021-03-01:
There is a way, but not in the way you describe.

This library works asynchronously. So at the time of making the request (`addRequest()`), the response is not yet available. Only when the response is received, the `handleData()` function is called.

If you need to handle data based on the request you made, you can use the `token`. There are a million possible ways to use this. One option could be:

```cpp
// Set the token value to something sensible:
MB.addRequest(/*token id 1*/ 1, 20, READ_HOLD_REGISTER, 10, 4);
MB.addRequest(/*token id 2*/ 2, 20, READ_HOLD_REGISTER, 14, 4);
MB.addRequest(/*token id 3*/ 3, 20, READ_HOLD_REGISTER, 18, 4);
MB.addRequest(/*token id 3*/ 3, 20, READ_HOLD_REGISTER, 22, 4);
MB.addRequest(/*token id 0*/ 0, 20, READ_HOLD_REGISTER, 22, 4);

void handle1(ModbusMessage response) {
  // process data from reg 10
}

void handle2(ModbusMessage response) {
  // process data from reg 14
}

void defaultHandle(ModbusMessage response) {
  // default
}

void handleData(ModbusMessage response, uint32_t token) 
{
  switch (token) {
    case 1:
      handle1(ModbusMessage response);
      break;
  }
    case 1:
      handle2(ModbusMessage response);
      break;
  }
    case 1:
      defaultHandle(ModbusMessage response);
      break;
  }
}

See https://emodbus.github.io/modbusclient-common-api#the-token-concept

arangu97, 2021-03-01:
Ok, thank you so much. Really useful.

Another thing:

When receiving a message, as seen in the image, I try to get the payload : 00 01 00 00
I have done this : 

uint32_t value;
response.get(2,value)

However value instead of 1 is 67109120. 
Maybe there is a need to swap the bytes received in order to get the correct value?

Thank you

[image]

Miq1, 2021-03-01:
Uint32_t is 4 bytes. You are reading from byte 2 of the response, so your data is 0x04000100 - try reading from byte 3 instead, behind the length byte 0x04 😉

Miq1, 2021-03-01:
Oh,by the way: the response bytes are 0x00010000, so I reckon the server is using 16 bit values (as are all Modbus registers per se). If you really are expecting a 32-bit value,it looks like the server in fact has a word swapped.

closed 2021-03-01 by Miq1