eModbus/eModbus #69: library doesn't compile in arduino IDE

TLS1000, 2021-03-17:
when attempting to compile the RTU06 example it returns error:

```
In file included from U:\NewPrimary\My Documents\Arduino\libraries\eModbus-master\src\ModbusClientTCPasync.cpp:5:0:

U:\NewPrimary\My Documents\Arduino\libraries\eModbus-master\src\ModbusClientTCPasync.h:9:22: fatal error: AsyncTCP.h: No such file or directory

compilation terminated.
```

Miq1, 2021-03-17:
You are supposed to have the AsyncTCP library installed - please have a look at the [lib as provided by @me-no-dev](https://github.com/me-no-dev/AsyncTCP)!

closed 2021-03-17 by Miq1

bertmelis, 2021-03-17:
Unfortunately there is no clean way of exclusing source files from compiling so you need to have all the dependencies installed even though they won't be included in the final firmware.

TLS1000, 2021-03-17:
now it works, thank you

Miq1, 2021-03-17:
👍