eModbus/eModbus #81: ModbusServerRTU not working reliable for function code 0x10

EmbeddedDevver, 2021-05-13:
My client (windows PC) ask from my server (ESP32) to write holding registers (0x10). When writing less than 5 words, it _usually_ goes fine but still sometimes (maybe 1 out of 40 packets) gets an error thrown (PACKET_LENGTH_ERROR). 

Increasing the amount of word writes to holding registers, you will receive more frequent errors. Going to 100 word writes (I know, absurd value) you will always get an error. I use 100 word writes to holding registers as a test case in the code. And I got it fixed that never an error will be thrown.

The problem is because of the following;

The ModbusServerRTU function serve() is pinned to core 0 and assigned the highest priority (number 8 priority).
See:
`    xTaskCreatePinnedToCore((TaskFunction_t)&serve, taskName, 4096, this, 8, &serverTask, coreID >= 0 ? coreID : NULL);`

Now going to the file RTUuntils.cpp.
The function ModbusMessage RTUutils::receive waits for new data to arrive on the serial port in the state "WAIT_DATA". When a single byte is received, the state is changed from "WAIT_DATA" to "IN_PACKET".

And here is the problem. "IN_PACKET" is an blocking function. It blocks the entire RTOS. Thus also Queues will not work anymore from FreeRtos. The serial port espHAL uses the FreeRtos Queue to transfer data from hardware uart to the FreeRtosQueue.

Meaning, when in the state "IN_PACKET" no new serial port data can arrive in the freertos queue. You will only receive the current queue data until it is empty. New serial port data during this time will not be added to the queue causing that modbus frame's will not be received completely if they are a longer than 1mS.

1) Quick solution that I have tested to work FINE with 100 words. (zero faults!).

In the "IN_PACKET" routine add a delay "delay(1);" thus giving the scheduler some space to allow the queues to be filled. Also increase the interval for this state to e.g. 10mS. Why such a high interval? Because on the core0 also WiFI functions are operating. When using WiFi like I do, it is possible that there will be some context switches available that can cause context switches delay of several mS.

The question also is, why is such tight "interval" value is needed anyways. Because currently at 56k6 BAUD rate, the interval will be only 650uS. For a server, it is absolutely no problem to react some time later. Even the most robust PLC's that I work with, do not reply that soon. They take 50mS or more for a response. Do note that the interval specified in the ModbusRTU is _mininum_ interval. Not maximum. RTU servers may not respond earlier than this interval, but later is no problem.

2) Solid solution: It could be improved the following way:

Add to the "IN_PACKET" state an routine that checks the incoming data. With the first let say 6 bytes from the modbus frame, you can already determine what frame length of data is to be expected. Using such a peek function, after just receiving 6 bytes of data from a client, you can calculate how many bytes are left to be received thus at arrival of the latest char you can directly start processing without any interval delay.

Miq1, 2021-05-13:
@EmbeddedDevver Thank you for your analysis!

You are right with the missing ``delay(1)``. I will add that to the code asap. 

I am not really convinced by your argument on relaxed interval times, especially with RTU servers. If you have multiple servers on a RTU Modbus, the servers need to react on every message to see if it is a request targeted to them. If the server would not be able to maintain the interval, it may miss a request, resulting in timeouts at the client.

Please note that you have the option to do a ``start(1)`` to have the server run on core 1.

EmbeddedDevver, 2021-05-13:
@Miq1 

May I first start to say that your code is awesome! Thank you for that.

**Interval**

When having multiple servers, let say we have 50 ESP32 on a RS485 bus, then adding the relaxed interval delay of 10mS would cause that every server replies after 10mS if the request is targeted to them. No server will miss a request, it will only process it later.

I never ever have seen any client at a PLC having a option to select a timeout response of lower than 100mS. (Coming from Phoenix contact, Omron and WAGO PLC's).

**Core 0 or Core 1**
Running the modbus library at core 1 does not work here, I tried that before, the complete library freezed then. I did not check what the cause of it was, maybe I do that later

EmbeddedDevver, 2021-05-13:
> [Edited upon some more thoughts]
> The `delay(1)` you are proposing does not make sense with baud rates where the calculated interval is lower than 1ms, the very first delay will make the gap detection pop up.

A short calculation follows:

When writing 100 words of holding registers this comes to:

1 byte slave address
1 byte function code
2 bytes data address
2 bytes number of data
1 byte number of byte to follow
200 bytes of data (equals 100 words)
2 bytes of CRC
---------------+
209 bytes = * 10 = 2090 bits

At a BAUD rate of 115k2, this comes to

2090 / 115200 = 18mS for a frame.

Note that the interval for 115k2 is set to 325uS only.

Miq1, 2021-05-13:
Hmm, never heard about that looong intervals before. The client in fact is dictating the interval, so if a server timing-wise would be below the standards of the client, it could not work reliably. In the ``ModbusClientRTU`` code the ``begin()`` function has an optional factor parameter to scale the interval up to cope with slow responding servers.

But I always wanted to have the server code maintain the interval as good as possible. I will experiment a bit with different ideas I already got. PLS SBY! 😉

Miq1, 2021-05-13:
For the time being I created the ``RTU-intervals`` branch, where both the client ``start()`` and server ``begin()`` functions have an optional second parameter that lets you overwrite the calculated interval (given value needs to be larger than the calculated).

I also added the ``delay(1);`` to the receive loop, but only if the interval is >1ms.

EmbeddedDevver, 2021-05-13:
Nice one!

I will see if I have the time to make such a peek function tomorrow as described earlier. That should also resolve your wish to not have delays when not needed. :-)

EmbeddedDevver, 2021-05-13:
Yessss!! my 'peek'  function is implemented and working well. Tested for 100 holding writes and 100 holding reads. No problem anymore.

Keep in mind:  While I was thinking. With the current implementation, even for small requests from clients, there is a possibility that the frame is not received completely and thus the server will never respond. You can see this problem as a race condition within the OS running at the ESP32.

I will try to explain:

1. You have the ESP32 hardware UART with 128 bytes dedicated RX FIFO buffer.
2. The hardware UART is managed by a ESP HAL library with a task, say **task0** on core0. This task0 reads any data from the hardware FIFO UART buffer and puts them in a FreeRTOS queue so that any other task can read the data safely.
3. The Modbus task, say **task1** waits for data to come from the FreeRTOS Queue and is also running on core0
4. Now task0, the ESP HAL lib, detects 2 chars in the UART FIFO buffer, it copies them to the FreeRTOS queue.  This task0 then gets a context switch.
5. During (or after) this context switch, 4 more bytes were received by the hardware UART and is ready to read by task0. But as task0 is suspended, the hardware is not serviced yet.
6. now task1, the Modbus task, starts reading the queue. It detects 2 chars and reads them out, then it waits for more chars but they never arrive because task0 cannot (yet) run. Even if your Modbus task1 is using delay(1) **once**  you cannot be sure that task0 is run as the next task. It could be also WiFi, ADC, watchdog, or other peripheral from the ESP32. After the expired delay(1), the Modbus task1 has the first priority to run (as it is the highest priority).
7. Now the modbus task never receives any more data and thus the task1 interval timer expires and generates an Modbus error because of an incomplete frame.
8. The Modbus client that requested data never receives data and continues after his timeout has expired (causing far more delay than the initial 10mS).

I hope the above explains the current error/race condition. Because of the OS the Modbus can never 100% work reliable with the current implementation. Also if you switch core, you still have this race condition because core0 is shared with ADC and WiFi and many more. You can never be sure if and when the hardware FIFO is serviced by the ESP32 HAL.

**THE solution**
The solution below fixes all. No need for looong timer as you prefer, but 100% guaranteed no half received frames. 

I did some testing @ 115k2 BAUD:
- 100 word writes (FC 16) = 7163uS
- 100 word reads (FC 3) = 1094uS
I think some excellent timings.

Also its interesting to see, if you enable the debug option with my new code below. 

This debug output is for the 100 word writes. Note the remaining quantity. Every quantity is updated after delay(1); was run once. As you can see the data is not coming at the same rate from the ESP32 HAL libs. 
For example, the first delay(1); only received 8 more bytes (from 168 to 160 remaining).

```
[D] 10165| ModbusServerRTU.cpp  [ 208] serve: Response sent.
[D] 24971| RTUutils.cpp         [ 241] receive: IN_PACKET 
[D] 24973| RTUutils.cpp         [ 310] receive: RTU Expecting bytes 206, remaining 168
[D] 24974| RTUutils.cpp         [ 310] receive: RTU Expecting bytes 206, remaining 160 
[D] 24982| RTUutils.cpp         [ 310] receive: RTU Expecting bytes 206, remaining 64
[D] 24989| RTUutils.cpp         [ 310] receive: RTU Expecting bytes 206, remaining 0 
[D] 24996| RTUutils.cpp         [ 317] receive: RTU Interval end, expired time: 7163
[D] 25004| RTUutils.cpp         [ 335] receive: DATA_READ OK
```
Below the code

```
  case IN_PACKET:
      // Data waiting and space left in buffer?
      while (serial.available())
      {
        // Yes. Catch the byte
        // Due to a latent bug in esp32-hal-uart.c of the arduino-esp32 core,
        // available() may return a 1 although there is no byte in the buffer yet.
        // So we have to attempt a read() and discard the result, if it is < 0.
        int b = serial.read();
        if (b >= 0)
        {
          buffer[bufferPtr++] = b;
          // Buffer full?
          if (bufferPtr >= bufferBlocks * BUFBLOCKSIZE)
          {
            // Yes. Extend it by another block
            bufferBlocks++;
            uint8_t *temp = new uint8_t[bufferBlocks * BUFBLOCKSIZE];
            memcpy(temp, buffer, (bufferBlocks - 1) * BUFBLOCKSIZE);
            delete[] buffer;
            buffer = temp;
          }
          // Rewind timer
          lastMicros = micros();
        }
        else
        {
[... 24 lines cut ...]
          numBytes = 0;
          break;
        }
        // add frame first 6 bytes plus the last 2 crc. This is the total to be expected frame sz.
        numBytes += 5 + 2;
        LOG_D("RTU Expecting bytes %d, remaining %d \n", numBytes, (numBytes - (bufferPtr - 1)));

        // only exit the receive if: exact right amount of bytes is received AND interval has timeout so that
        // we cannot reply too soon (interval) and also not exit too soon (still bytes left in uart)
        if (micros() - lastMicros >= interval && numBytes == (bufferPtr - 1))
        {
          state = DATA_READ;
          LOG_D("RTU Interval end, expired time: %lu \n", ((micros() - lastMicros)));
        }
      }

      // Timeout handling: The incoming message must be received completely within 50mS
      // else we may conclude that the client request has an error or was interrupted.
      if (micros() - lastMicros >= 50000)
      {
        state = FINISHED;
        rv.push_back(PACKET_LENGTH_ERROR);
        LOG_D("Err: Timeout during receiving frame, received %d bytes\n", (bufferPtr - 1));
      }
```

Miq1, 2021-05-13:
Well, yes, I get the idea. Unfortunately it is not completely fitting, I am afraid. 

There are only some function codes where we have a chance to peek ahead to learn the number of bytes to expect, the user defined codes are free-form and will be missed. And real out-of-gauge packets will confuse the mechanism as well.

EmbeddedDevver, 2021-05-13:
@Miq1

That I do understand. For that, I do not know any other solution than having a large enough delay or you should make your own hardware implementation for the esp32 :) 

People using this library are more helped with something that works right with some extra delay than something that sometimes does not work. 😉

Miq1, 2021-05-13:
> @Miq1
>
> That I do understand. For that, I do not know any other solution than having a large enough delay or you should make your own hardware implementation for the esp32 :)
>
> People using this [...]

Right. But give me some more time to think about it - maybe I get another idea...

Miq1, 2021-05-14:
I referred to the Modbus standard once more and found this:
[image]
So it is mandatory to maintain the gap and character timing for baud rates of 19200 and below to be conforming to the standard, with a little relaxation for baud rates above that. 
Having read that I would reckon we never will be able to achieve that on the ESP32 without a much deeper control of the serial communications. I was not aware of the 1.5t condition for in-frame gaps that would require packet discarding; as the gap times between packets are already hard to match, the in-frame gap checking is completely out of reach. 
I will try to collect the real constellations we may encounter with clients and servers maintaining the proper timing or not and then decide how to tackle the issue.

Miq1, 2021-05-14:
This is going to be lengthy and meandering, so please bear with me 😉 

1) we should concentrate on ``receive`` rather than also considering ``send``. Only in receiving messages we might get into timing issues.
2) we will ignore the intra-message timing the standard obviously is requiring as it seems to be too challenging to meet that
3) given the internal UART buffer size of 128 bytes the Arduino core mandates we can be sure we will have to do repeated reads for longer messages anyway
4) we will disregard pathological messages or fragments

### Client considerations
- a client is controlling the bus. Timing is completely depending on the client.
- a client will receive only messages following a previous request sent by itself.
- the time gap between the request and the response is irrelevant, only the basic timeout to detect non-responding servers is to be maintained.
- The gap after the response is irrelevant as well, as the next to be active on the bus again will be the client.
Concluding we may state that the client can wait as long as it likes for all data to be received.

### Server considerations
- a server is just one of several connected to the same bus.
- a server has to watch all the time if a request is on the bus addressed to it
- if another server will answer immediately after the bus interval has passed, missing it will be harmless for the server
- if the client issues a request immediately after the last response has been received, the server needs to be able to catch it in case it is directed to it ⚡ 
- the server may take several read cycles to read a message. If the message is addressed to it, this will only affect the general timeout. If it is not, it is regardless anyway, as it will have to ignore it. 
Conclusion: there can be an issue if the client has a tight timing. All other cases may be handled more relaxed.

### Implications
- in general, we will not care about the interval while receiving, but do read cycles with some ``delay`` in between until no more bytes are found 
- the client still needs to maintain at least the interval before sending another request.
- a ModbusServerRTU thus does not guarantee to react immediately after the interval has passed any more.

Does this all make sense to you?

Miq1, 2021-05-14:
I committed a modified RTU receive handling according to the above considerations to the ``RTU-intervals`` branch.

I am looking forward to your results.

EmbeddedDevver, 2021-05-14:
A quick reply here, as I am on my mobile. I understand your points and I think that is correct. Some points however:

> internal UART buffer size of 128 bytes

This can be easily be enlarged to e.g. 1k bytes buffer. Make a call to Serial.setRxBufferSize(SERIAL_SIZE_RX); where SERIAL_SIZE_RX can be defined e.g. 1000bytes of buffer.

I think there is no problem currently with the lib when sending characters, frame's can be transmitted to the serial buffer at once. The serial port has its own DMA so there will never be a silence between two characters that might lead to a timeout. So that is fixed 💯 

The only problem is indeed at reception, but only because the preference to have stricter timings. If you look for example to MS Windows operated HMI systems, they reply every 50 to 200 mS to a client request.

From the view of MS Windows, there is no other option than just wait for 50mS or longer and then check / read the entire serial buffer and see if it's a valid and complete frame. This works fine and very stable, at the costs of relaxed timings.

As long as this lib is running at an OS, I see no other option than 
A) have more relaxed timings
B) use software serial
C) built your own HAL
D) predict frame lengths (peeking) for known frames. For unknown frame you can switch automatically to a relaxed timing.

In my opinion it should work for users without having knowledge how its works. And if it works 100% stable at a e.g. 50mS reply rate to a client, than that is a specification that can be supplied with this library.

So my preferred solution would be relaxed timings combined with peek function to speed up the known functions. Also you can peek for frames that is addressed to the received server, if it is not, you can trash it directly. 

And clients will have to meet the requirements of the (slowest) servers. 

Again, awesome work that you make here!!!

Miq1, 2021-05-15:
I tested the new code up to 921600 baud without hickup (with the Test sketch, looping through Serial1<->Serial2, one task as client, another as server). 

Would be interesting to know if your code does pass as well, without any other means.

EmbeddedDevver, 2021-05-15:
awesome to read that it now passes.
wednesday iam back home, then I will redo the test with the new code.
i can only go till max of 112k6 bc of the tranceivers im using in my setup.

ill keep you posted

Miq1, 2021-05-19:
@EmbeddedDevver Do you have results already?

EmbeddedDevver, 2021-05-19:
Not yet, I have a problem to buy certain IC chips and I had to place this project on hold until I can find the needed chips somewhere to buy. I keep you informed.

Miq1, 2021-05-26:
@EmbeddedDevver Howzit t'day?

EmbeddedDevver, 2021-05-26:
Almosttt there :-).

Since my last message I am daily trying to get chips but everything is sold out :-(
But I will come back :-))

EmbeddedDevver, 2021-05-27:
I thoroughly checked your branch  RTU-intervals. 

The "case IN_PACKET:" routine is still blocking, am I right? There is only delayMicroseconds() available, but this routine is not giving the scheduler room to handle other tasks as it is an blocking delay. For as long as the modbus is running on core 0, note that core 0 also has the watchdog running. Any long delay at core0 causes a force reboot.
I recommend a delay(1) here to give the OS room.
I think you do this because you want to keep strict timings here, thus having a fast reply. I can get that view, however I am more on the safe side than strict timing. 
Note when you're using the wifi with some code (webserver, sockets, or stuff like that), core 0 cannot be blocking for more than several mS before you get errors with the wifi handling. 
Better would be to run the code anyway at core1. Wifi needs some dedicated fast acting core. I think that is the reason they used wifi at core 0 and user code at core 1.

Also a caution about this routine below. When bogus information is coming from serial port (e.g. a wire come lose), then this routine will never end the while loop. And because this loop is running at highest priority, the entire system will freeze up. I recommend a timeout, e.g. after 1 second throw an exception.
''
      while (b >= 0)
      {
        buffer[bufferPtr++] = b;
        // Buffer full?
        if (bufferPtr >= bufferBlocks * BUFBLOCKSIZE)
        {
          // Yes. Extend it by another block
          bufferBlocks++;
          uint8_t *temp = new uint8_t[bufferBlocks * BUFBLOCKSIZE];
          memcpy(temp, buffer, (bufferBlocks - 1) * BUFBLOCKSIZE);
          delete[] buffer;
          buffer = temp;
        }
        hadBytes = true;
        b = serial.read();
      }
''

I still need to test the code.... all be back :-)

EmbeddedDevver, 2021-05-27:
I double checked that the `delayMicroseconds()` routine for you.
Here you can see what the actual routine is like, it is indeed an blocking code as I said previously.

```
void IRAM_ATTR delayMicroseconds(uint32_t us)
{
    uint32_t m = micros();
    if(us){
        uint32_t e = (m + us);
        if(m > e){ //overflow
            while(micros() > e){
                NOP();
            }
        }
        while(micros() < e){
            NOP();
        }
    }
}
```

I was just thinking, but besides `delay(1)`, you could also consider `taskYIELD()`. This way, you give the scheduler the option to do some housekeeping. FreeRTOS description for `taskYIELD()`:

``
taskYIELD() is used to request a context switch to another task. However, if there are no other tasks at a higher or equal priority to the task that calls taskYIELD() then the RTOS scheduler will simply select the task that called taskYIELD() to run again.
``

If you go this route with `taskYIELD()`, you will probably have a delay far less than 1mS.

Miq1, 2021-05-27:
Thanks for confirming. The ``taskYIELD`` is sounding nice, although I would have to combine that with a ``micros()`` check on the interval passing. That was why I originally had the ``delayMicroseconds(interval)`` in it. If the interval had passed without further data, I considered that as the packet's end.

I committed another change along these lines now. Runs okay at least with my test sketch at 115200 baud.

EmbeddedDevver, 2021-05-27:
I think this looks better now. 
One more note, I think the `taskYIELD` can better get a place here:

```

      // Did we read some?
      if (hadBytes) {
        // Yes, take another turn
        intervalEnd = micros();
        **taskYIELD();**
        // delay(1);
      } else {
        // No. Has a complete interval passed without data?
        lastMicros = micros();
        if (lastMicros - intervalEnd >= interval) {
          // Yes, go processing data
          state = DATA_READ;
        }

//here is a better place as in both if and else clause this taskYield will be executed. 
        **taskYIELD();** 
      

```

Miq1, 2021-05-27:
Well, the high priority was chosen to protect the tight RTU timing. The more we are discussing here, the more this concept seems to get brittle. I am not sure if we are going into the right direction here, but I have no personal experience to really judge that.

EmbeddedDevver, 2021-05-27:
Before I created many libraries for Microchip 16/32 bit devices and one of them was Modbus. 
The power with those devices was that there is no OS. You could pre-calculate timings to the microsecond.

However, this is not a time-critical bus (Modbus). If someone need something time critical, probably they choose CANbus. For now, I will use this library with the `delay(1);` inside, works rock solid and I think it is the best you can do when A) using an OS and B) not sacrificing that other libraries wont work anymore and C) to have an universal library. 

Next step for me is to change the priority to zero and maybe increase the hardware buffer to 512 bytes. That way, the OS at ESP32 just gets an trigger when there is data in the buffer and thus can be read out at once.

Now I am also using an auto direction RS485 chip, so no more delay toggling the receive/drive enable pin. If you have something new to test, let me know :-)

Miq1, 2021-05-27:
I will revert to the ``delay(1)`` as well and take out the ``taskYIELD() ``.

Despite using RTU and WIFI in parallel on the bridge devices I do not remember having any freezes or reboots, so I will leave the RTU task priority at 6. Anyone is free to choose core 1 for it anyway - that is in fact what we suggest in the docs. 

Thank you for your insights and suggestions!

The current "hot thing" I am working on is a ``CoilData`` class to make handling of this data type easier. This will be released next together with the RTU changes. Feel free to try it out!

Miq1, 2021-06-01:
Closing this now as the issues seems to be resolved.

closed 2021-06-01 by Miq1