eModbus/eModbus #83: In examples RTU server i'm only able to read nine registers not ten as expected

Bach-Danielsson, 2021-05-16:
**Describe the bug**
During my test with your code exaple RTU server on a ESP32 i can´t read register 00
Staring from register 01 works fine.

I can max read nine registers but i was expecting to be able reading ten registers.

please see my data log as bellow:

[RTU]>Tx > 16:45:48:762 - 01  03  00  01  00  09  D4  0C  
[RTU]>Rx > 16:45:48:800 - 01  03  12  00  01  00  02  00  03  00  04  00  05  00  06  00  07  00  08  00  09  9C  B4
  
[RTU]>Tx > 16:46:55:237 - 01  03  00  00  00  09  85  CC  
[RTU]>Rx > 16:46:55:268 - 01  83  02  C0  F1  
Sys > 16:46:55:268 - Error : Illegal data address

RTU]>Tx > 17:13:29:646 - 01  03  00  00  00  0A  C5  CD  
[RTU]>Rx > 17:13:29:666 - 01  83  02  C0  F1  
Sys > 17:13:29:666 - Error : Illegal data address

Miq1, 2021-05-16:
This is according to the Modbus standard - address 1 is supposed to be the lowest possible address. This is guarded by the following code in the ``FC03`` worker function in the example:
```
// Address and words valid? We assume 10 registers here for demo
  if (address && words && (address + words) <= 10) {
    // Looks okay. Set up message with serverID, FC and length of data
```
``address`` needs to have a value != 0 for the ``if`` clause to be true.

In removing the very first condition you can change the behaviour.

Bach-Danielsson, 2021-05-16:
Thank you for fast responce and making this powerful tool eMODBUS.

closed 2021-05-16 by Miq1