eModbus/eModbus #87: arduino-esp32 version

jandegr, 2021-05-25:
Hi,

I have a little project on a Lolin32 board that uses https://github.com/bertmelis/esp32ModbusRTU for 2 years approx.
I wanted to make a little modification to it and I noticed using ardiono-esp32 10.5 or higher breaks it. It still works with version 10.4
https://github.com/espressif/arduino-esp32/tags

So I just decided to migrate to eModbus and noticed the same, a few lines I wrote work on 10.4 but give error E5 - Packet length error when I use 10.6 

I have tested with 10.4 (ok) and 10.6(fail) but not with 10.5 yet.

It is a setup with a max485 and a lolin32 board and uses ModbusClientRTU

I may have overlooked something but since I get the same problem with eModbus I wanted to ask what the youngest version of arduino-esp32 is this has been tested with ?

THX

Gamelauncher, 2021-05-25:
I am afraid the problem lies deeper than that. I've noticed similar performance issues starting with 1.0.5 and above. Allot of things that depends on AsyncTCP are affected by it so my only solution for now is to stay at 1.0.4 which runs everything I needed. There is for sure a bug detected in the analogue section which causes slow downs and random problems.

For reference:
https://github.com/espressif/arduino-esp32/issues/5205
https://github.com/espressif/arduino-esp32/issues/4973

Miq1, 2021-05-25:
I see. I myself am using the AsyncTCP variety on the ESP8266 only, so that may explain my ignorance 😁

I do not see any impact on the RTU code, though - the processing code is completely separate from either TCP version.

Could you elaborate a bit more on your configuration, which components of eModbus you are using etc.?

Miq1, 2021-05-26:
Any update available?

jandegr, 2021-05-26:
To keep it running I stay at 1.0.4 untill it can have some downtime again to do more testing,

thx

Miq1, 2021-05-26:
Okay, understood.

I may release the RTU changes in the meantime, so as soon as you will be able to try, you might have to use the master branch again.

closed 2021-06-01 by Miq1