eModbus/eModbus #94: ModbusServerTCP (with WifiServer) does not handle stop/start correctly

domonoky, 2021-07-21:
The ModbusServerTCP (with WifiServer) code does not handle a stop/start cycle correctly.

If i call stop on the server and then start again the call to listen will fail because the internal WifiServer was never stopped correctly.
This is because the stop() call in ModbusServerTCPtemp just does a vTaskDelete on the server task (serve function).
When the task gets deleted, the memory is freed, but the destructor of the WifiServer is never called.

Miq1, 2021-07-21:
Thank you for the information. I will look into it as soon as possible, but that may take some time as I am pretty busy currently.

Miq1, 2021-07-30:
It does not look easy to solve this...

I now aded a local context in ``serve()`` to have the server's destructor called before doing a ``vTaskDelete()``, but nevertheless a subsequent ``start()`` seems not to be successful, all further requests are failing.

I am refraining from stopping WiFi itself and restarting it, due to other services may run at the time, but I think there is something left in the WiFi stack that prevents a fresh server from being started completely.

Any ideas are welcome!

domonoky, 2021-08-02:
According to my experiments the destructor is also not called if you just add a local contect around the serve function. It seems like vTaskDelete(..) from outside the task is evil :-) 
What seems to work is to add a signaling method to the serve task (semaphore or queue) and signaling the task to clean up and end itself with vTaskDelete(Null).

Miq1, 2021-08-02:
I was not exhaustive enough telling what I did: exactly what you are proposing. The flag gets set by the main task and the ``serve() `` function will leave the local context, wait 500ms and then do a ``vTaskDelete(NULL);`` to kill itself.
Nevertheless the server is not starting correctly, so it seems.

domonoky, 2021-08-02:
This is what i did to fix this, and it is working in my system. I can now stop, reconfigure and start the modbus server without problems:

template <typename ST, typename CT>
void ModbusServerTCP<ST, CT>::serve(ModbusServerTCP<ST, CT> *myself) {
  
  {
      // Set up server with given port - no backlog allowed
      ST server(myself->serverPort,1);

      // Start it
      server.begin();

      // Loop until being killed
      while (!myself->serverEndTask) 
      {
        // Do we have clients left to use?
        if (myself->clientAvailable()) {
          // Yes. accept one, when it connects
          CT ec = server.accept();
          // Did we get a connection?
          if (ec) {
            // Yes. Forward it to the Modbus server
            myself->accept(ec, myself->serverTimeout, 0);
            LOG_D("Accepted connection - %d clients running\n", myself->activeClients());
          }
        }
        // Give scheduler room to breathe
        delay(10);
      }
  }
  //delete yourself
  vTaskDelete(NULL);
}

The boolean myself->serverEndTask should probably be replaced with a semaphore or similar to make it task/multicore safe.

domonoky, 2021-08-02:
In my debugging session i found out the original problem is, that if you do not call the destructor of the wifiserver, then the port ist not given back to the system. 
So when you start the wifiserver again, the call to listen() fails and so the wifiserver does not work correctly. Unfortunately the return codes of the listen and bind calls are not checked in the wifiserver.
So if you want to test if your changes fix the original problem, add some ouptuts to void WiFiServer::begin(uint16_t port, int enable) to see if the bind and listen calls do not return errors after stop/start.

Miq1, 2021-08-04:
I am meeting my old Nemesis here - ``connected() ``...
The client code is my problem. It is checking if the target host is the same as before. If it is, no disconnect is made. Next ``connected()`` is used to check if the connection is still open. In the case of a stop/start of the server this call returns ``true``, hence the request is sent into oblivion.
I have no idea as of how to fix that, as the problem lies within the arduino-esp32 core code. I encountered that before and raised an issue with the core, but that was never worked on.

Miq1, 2021-08-06:
Tried some more on that, but to no avail. I cannot get the client recognize the server has closed the connection. Rats...

Miq1, 2021-08-06:
If I do an explicit ``stop()`` before each request on the client side, all is working fine. So again: for whatever reason, the client does not detect the server's ``stop()``.

Miq1, 2021-08-08:
I might be chasing ghosts here again - I found indications that the update of ``connected()`` is not done in WiFi loopback mode. My test sketch is using the loopback extensively and this may be the reasomn I am not getting any connection information on the stopped server.

closed 2021-08-08 by Miq1