eModbus/eModbus #97: RTU: Server don't receive answer from client

KevinDesbonnets, 2021-07-29:
**Describe the bug**
When using the RTU06 example on a olimex-esp32-poe + mod RS-485 with RTUserver example on a olimex-esp32-poe + mod RS-485 the server was unable to get the answer from the client.
I did slight change in library to make 2 rts pin (RE and DE are not wired together), and running test file show no errors so I don't thinks it come from this changes.

**Expected output**

```
15:16:58.867 -> [D] 5011| ModbusClientRTU.cpp  [  76] begin: Worker task 1073452664 started. Interval=1822
15:16:58.915 -> [V] 5013| ModbusMessage.cpp    [ 529] checkData: Check data #3
15:16:58.915 -> [D] 5019| ModbusClientRTU.cpp  [ 111] addRequestM: request for 01/03
15:16:58.915 -> [D] 5025| ModbusClientRTU.cpp  [ 160] addToQueue: RC=01
15:16:58.915 -> [D] 5030| ModbusClientRTU.cpp  [ 122] addRequestM: RC=00
15:16:58.961 -> [D] 5111| ModbusClientRTU.cpp  [ 178] handleConnection: Pulled request from queue
15:16:59.007 -> [D] Sent packet: @3FFB9804/6:
15:16:59.007 ->   | 0000: 01 03 00 01 00 01                                 |......          |
15:16:59.007 -> [D] 5117| ModbusClientRTU.cpp  [ 183] handleConnection: Request sent.
15:16:59.007 -> [D] Received packet: @3FFB98C0/9:
15:16:59.007 ->   | 0000: 01 03 00 01 00 01 D5 CA  00                       |.........       |
```

**Current output**

Server :
```
15:16:58.867 -> [D] 5011| ModbusClientRTU.cpp  [  76] begin: Worker task 1073452664 started. Interval=1822
15:16:58.915 -> [V] 5013| ModbusMessage.cpp    [ 529] checkData: Check data #3
15:16:58.915 -> [D] 5019| ModbusClientRTU.cpp  [ 111] addRequestM: request for 01/03
15:16:58.915 -> [D] 5025| ModbusClientRTU.cpp  [ 160] addToQueue: RC=01
15:16:58.915 -> [D] 5030| ModbusClientRTU.cpp  [ 122] addRequestM: RC=00
15:16:58.961 -> [D] 5111| ModbusClientRTU.cpp  [ 178] handleConnection: Pulled request from queue
15:16:59.007 -> [D] Sent packet: @3FFB9804/6:
15:16:59.007 ->   | 0000: 01 03 00 01 00 01                                 |......          |
15:16:59.007 -> [D] 5117| ModbusClientRTU.cpp  [ 183] handleConnection: Request sent.
15:16:59.007 -> [D] Received packet: @3FFBDB4C/1:
15:16:59.007 ->   | 0000: E5                                                |.               |
15:16:59.007 -> [D] 5132| ModbusClientRTU.cpp  [ 189] handleConnection: Error response (1 bytes) received.
15:16:59.007 -> [V] Data: @3FFBDB4C/1:
15:16:59.007 ->   | 0000: E5                                                |.               |
15:16:59.007 -> [D] 5149| ModbusClientRTU.cpp  [ 218] handleConnection: Response generated.
15:16:59.054 -> [V] Response packet: @3FFBDB4C/3:
15:16:59.054 ->   | 0000: 01 83 E5                                          |...             |
15:16:59.054 -> [1;33m[E] 5166| eModBusMaster.ino    [  43] handleError: Error response: E5 - Packet length error
```
Client 
```
15:16:59.007 -> [D] Received packet: @3FFB98C0/9:
15:16:59.007 ->   | 0000: 01 03 00 01 00 01 D5 CA  00                       |.........       |
15:16:59.007 -> [D] 1545015| ModbusServerRTU.cpp  [ 127] serve: Request received.
15:16:59.007 -> [D] 1545019| ModbusServer.cpp     [  30] getWorker: Worker found for 01/03
15:16:59.007 -> [D] 1545025| ModbusServerRTU.cpp  [ 132] serve: Callback found.
15:16:59.007 -> [D] 1545031| ModbusServerRTU.cpp  [ 135] serve: CRC okay.
15:16:59.007 -> [D] 1545036| ModbusServerRTU.cpp  [ 145] serve: Callback called.
15:16:59.007 -> [V] Callback response: @3FFB98A8/5:
15:16:59.007 ->   | 0000: 01 03 02 00 01                                    |.....           |
15:16:59.054 -> [D] Sent packet: @3FFB9874/5:
15:16:59.054 ->   | 0000: 01 03 02 00 01                                    |.....           |
15:16:59.054 -> [D] 1545061| ModbusServerRTU.cpp  [ 185] serve: Response sent.
15:16:59.054 -> [D] Received packet: @3FFBDB4C/1:
15:16:59.054 ->   | 0000: E5                                                |.               |
15:16:59.054 -> [1;33m[E] 1545076| ModbusServerRTU.cpp  [ 193] serve: RTU receive: E5 - Packet length error
```

**To Reproduce**
I used two olimex-esp32-poe + mod RS-485 both wired together with A <-> A and B <-> B

The TX pin is GPIO4
The RX pin is GPIO36
The RTS pin is GPIO5 (for DE)
The second RTS pin is GPIO14 (for RE)

**Example code**
Acces on 19 july 2021
Exemple RTU06 :
https://github.com/eModbus/eModbus/blob/master/examples/RTU06example/main.cpp
Exemple RTUserver:
https://github.com/eModbus/eModbus/blob/master/examples/RTUserver/main.cpp

Only read part is use with slight changes
Server :
```
#define TXEN 5
#define RXEN 14

ModbusClientRTU MB(Serial2,TXEN, RXEN);

Serial2.begin(19200, SERIAL_8N1, GPIO_NUM_36, GPIO_NUM_4);
```

Client
```
#define TXEN 14
#define RXEN 5

ModbusServerRTU MBserver(Serial2, 5000, TXEN, RXEN);

Serial2.begin(19200, SERIAL_8N1, GPIO_NUM_36, GPIO_NUM_4);
```

**Additional context**
When running test file on the original lib and the modified lib I have no error with log on Error (level 2) but one test fail with verbose, see below :

with Error
```
14:28:12.016 -> __ OK __
14:28:22.152 -> -----> Some timeout tests may take a while. Wait patiently for 'All finished.'
14:28:22.152 -> ----->    Generate messages tests:   68, passed:   68
14:28:49.155 -> ----->    TCP loop stub tests:   14, passed:   14
14:28:58.261 -> ----->    RTU loop tests:   12, passed:   12
14:29:02.273 -> ----->    TCP WiFi loopback tests:    9, passed:    9
14:29:04.658 -> ----->    Synchronous request tests:    7, passed:    7
14:29:06.900 -> ----->    Bridge tests:    6, passed:    6
14:29:06.900 -> ----->    CoilData tests:   24, passed:   24
14:29:06.900 -> 
14:29:06.900 -> 
14:29:06.900 ->  *** ----> All finished.
```

with verbose
```
14:34:49.613 -> ----->    RTU loop tests:   12, passed:   11
````

Miq1, 2021-07-29:
I am afraid I can't make too much sense from what you have posted. You modified the library to maintain DE and RE separately - fine, but that may be a cause for the problem you encountered. It looks like the client's request is received correctly by the server, but the response gets damaged.

It would be good to see your modifications to the library and also the error reported by the test sketch immediately before the line of RTU tests.

KevinDesbonnets, 2021-07-30:
Here the repo I make to show the change :
https://github.com/KevinDesbonnets/eModBus-REDE/commit/4c10aaa150dee6f133dc227c68b311267dce9542

I don't really understand what to show in the test, here a pastbin with all the log:
https://pastebin.com/9aPxi4AF

Thanks for your help

Miq1, 2021-07-30:
The log has at line 604
```
14:34:47.602 -> RTS toggle counts do not match. Expected=6, HIGH=7, LOW=8
```
I am using the RTS toggle callback feature here that misses to be called the right number of times. So I am suspecting your RTS routine may be interfering.

I will have a look at your repository asap, but would you mind explaining why you would need two separate toggles for DE and RE at all?

Miq1, 2021-07-30:
You added
```
   digitalWrite(MSRrtsPin2, level);
```
in your modification of the RTS toggle lambda function. Please note the DE and RE pins of the common RS485MAX modules need to be set to opposite levels, whereas your code will set the same. This may explain why the RTU handshake is failing.

KevinDesbonnets, 2021-07-30:
If I put
```
   digitalWrite(MSRrtsPin2, !level);
```

The server just hear itself :
```
[D] 5010| ModbusClientRTU.cpp  [ 104] setTimeout: Timeout set to 2000
11:33:33.809 -> [D] 5011| ModbusClientRTU.cpp  [  76] begin: Worker task 1073452664 started. Interval=1822
11:33:33.809 -> [V] 5013| ModbusMessage.cpp    [ 529] checkData: Check data #3
11:33:33.809 -> [D] 5019| ModbusClientRTU.cpp  [ 111] addRequestM: request for 01/03
11:33:33.809 -> [D] 5025| ModbusClientRTU.cpp  [ 160] addToQueue: RC=01
11:33:33.809 -> [D] 5030| ModbusClientRTU.cpp  [ 122] addRequestM: RC=00
11:33:33.903 -> [D] 5110| ModbusClientRTU.cpp  [ 178] handleConnection: Pulled request from queue
11:33:33.903 -> [D] Sent packet: @3FFB9804/6:
11:33:33.903 ->   | 0000: 01 03 00 01 00 01                                 |......          |
11:33:33.903 -> [D] 5116| ModbusClientRTU.cpp  [ 183] handleConnection: Request sent.
11:33:33.903 -> [D] Received packet: @3FFB9804/8:
11:33:33.903 ->   | 0000: 01 03 00 01 00 01 D5 CA                           |........        |
11:33:33.903 -> [D] 5131| ModbusClientRTU.cpp  [ 189] handleConnection: Data response (8 bytes) received.
11:33:33.950 -> [V] Data: @3FFB9804/8:
11:33:33.950 ->   | 0000: 01 03 00 01 00 01 D5 CA                           |........        |
11:33:33.950 -> [D] 5148| ModbusClientRTU.cpp  [ 218] handleConnection: Response generated.
11:33:33.950 -> [V] Response packet: @3FFB9804/6:
11:33:33.950 ->   | 0000: 01 03 00 01 00 01                                 |......          |
```

DE/RE pin need to be HIGH together and LOW together
Because when DE is HIGH sending is enable and when RE is LOW receiving is enable, RE/DE are commonly wired together but in MOD-RS485 it's not the case and I cannot solder the cables together. It's why I change it at logical level.

The log:
```
14:34:47.602 -> RTS toggle counts do not match. Expected=6, HIGH=7, LOW=8
```
come from the original code (i.e I just git clone and test)

Miq1, 2021-07-30:
You are right with the DE/RE levels, sorry for that.

The original test code does run here without any flaws, so something else must be different. I suppose you did the cross wiring between the Serials as necessary?

KevinDesbonnets, 2021-07-30:
Yes, I changed the GPIO use because PIN 17 don't exist on my board and I have correctly cross wired
The main difference may be the arduino IDE
The test fail occured only when I am at log level 4 or above (directly set in logging.h file)

Miq1, 2021-07-30:
That (error occurring only with higher log levels) would suggest timing issues - potentially possible with RTU, but normally only with RS485 hardware in between and higher baud rates. There are three Serial interfaces involved in that case, but AFAIK all are driven by their own hardware interrupts. Strange...

The RTS callback seems to be called one time more than expected, so the server is sending one excess response. Whyever.

KevinDesbonnets, 2021-07-30:
I reproduced this bug 10 times in a row with 100% success

Miq1, 2021-07-30:
I tried it myself with ``LOG_LEVEL=6`` and can confirm the error is real.

Miq1, 2021-07-30:
OK, fixed that one - was an error in the test sketch itself. The sketch would not wait long enough for the server task to finish all requests and thus was missing some RTS toggles, when additional log output slowed down the background task.

So this seems to be caught -  we now may focus on your E5 error mentioned in the first post.

KevinDesbonnets, 2021-08-03:
I didn't find anything right now. I post a picture of the hardware to make sure I didn't make an obvious error:

[image]

[image]

Miq1, 2021-08-03:
Looks okay, if a bit incomplete. But for the shortness of your wires it should work without termination.
The E5 error you are getting mostly is caused by signal issues (noise on the line, other servers talking out of bounds, low levels, servers having gaps between bytes wider than the bus quiet time etc.) 
Can we have a verbose log again of your client and server doing just one request and response? May be I missed something from the last.

KevinDesbonnets, 2021-08-03:
the server message:
```c
// Read register
  Error err = MB.addRequest(Token++, 1, READ_HOLD_REGISTER, 1, 1);
  if (err!=SUCCESS) {
    ModbusError e(err);
    LOG_E("Error creating request: %02X - %s\n", (int)e, (const char *)e);
  }
```

The client handle :
```c
ModbusMessage FC03(ModbusMessage request) {
  uint16_t address;           // requested register address
  uint16_t words;             // requested number of registers
  ModbusMessage response;     // response message to be sent back

  // get request values
  request.get(2, address);
  request.get(4, words);

  // Address and words valid? We assume 10 registers here for demo
  if (address && words && (address + words) <= 10) {
    // Looks okay. Set up message with serverID, FC and length of data
    response.add(request.getServerID(), request.getFunctionCode(), (uint8_t)(words * 2));
    // Fill response with requested data
    for (uint16_t i = address; i < address + words; ++i) {
      response.add(42 + i);
    }
  } else {
    // No, either address or words are outside the limits. Set up error response.
    response.setError(request.getServerID(), request.getFunctionCode(), ILLEGAL_DATA_ADDRESS);
  }
  return response;
}
```

Server verbose:
```
13:29:03.394 -> [D] 5011| ModbusClientRTU.cpp  [  76] begin: Worker task 1073452664 started. Interval=1822
13:29:03.394 -> [V] 5013| ModbusMessage.cpp    [ 529] checkData: Check data #3
13:29:03.441 -> [D] 5019| ModbusClientRTU.cpp  [ 111] addRequestM: request for 01/03
13:29:03.441 -> [D] 5025| ModbusClientRTU.cpp  [ 160] addToQueue: RC=01
13:29:03.441 -> [D] 5030| ModbusClientRTU.cpp  [ 122] addRequestM: RC=00
13:29:03.488 -> [D] 5111| ModbusClientRTU.cpp  [ 178] handleConnection: Pulled request from queue
13:29:03.488 -> [D] Sent packet: @3FFB9804/6:
13:29:03.488 ->   | 0000: 01 03 00 01 00 01                                 |......          |
13:29:03.534 -> [D] 5117| ModbusClientRTU.cpp  [ 183] handleConnection: Request sent.
13:29:03.534 -> [D] Received packet: @3FFBDB4C/1:
13:29:03.534 ->   | 0000: E5                                                |.               |
13:29:03.534 -> [D] 5132| ModbusClientRTU.cpp  [ 189] handleConnection: Error response (1 bytes) received.
13:29:03.534 -> [V] Data: @3FFBDB4C/1:
13:29:03.534 ->   | 0000: E5                                                |.               |
13:29:03.534 -> [D] 5149| ModbusClientRTU.cpp  [ 218] handleConnection: Response generated.
13:29:03.534 -> [V] Response packet: @3FFBDB4C/3:
13:29:03.534 ->   | 0000: 01 83 E5                                          |...             |
13:29:03.581 -> [1;33m[E] 5166| eModBusMaster.ino    [  43] handleError: Error response: E5 - Packet length error
```

Client verbose:
```
13:29:00.950 -> [D] 5029| ModbusServer.cpp     [  16] registerWorker: Registered worker for 01/03
13:29:00.950 -> [D] 5030| ModbusServerRTU.cpp  [  88] start: Server task 1073452776 started. Interval=1822
13:29:00.950 -> [D] Received packet: @3FFBDB4C/1:
13:29:00.997 ->   | 0000: E5                                                |.               |
13:29:00.997 -> [1;33m[E] 5043| ModbusServerRTU.cpp  [ 193] serve: RTU receive: E5 - Packet length error
13:29:00.997 -> [0m[D] Received packet: @3FFB9888/9:
13:29:03.518 ->   | 0000: 01 03 00 01 00 01 D5 CA  00                       |.........       |
13:29:03.518 -> [D] 7578| ModbusServerRTU.cpp  [ 127] serve: Request received.
13:29:03.518 -> [D] 7582| ModbusServer.cpp     [  30] getWorker: Worker found for 01/03
13:29:03.518 -> [D] 7588| ModbusServerRTU.cpp  [ 132] serve: Callback found.
13:29:03.518 -> [D] 7593| ModbusServerRTU.cpp  [ 135] serve: CRC okay.
13:29:03.518 -> [D] 7598| ModbusServerRTU.cpp  [ 145] serve: Callback called.
13:29:03.566 -> [V] Callback response: @3FFB98A8/7:
13:29:03.566 ->   | 0000: 01 03 02 00 00 00 2B                              |......+         |
13:29:03.566 -> [D] Sent packet: @3FFB98C0/7:
13:29:03.566 ->   | 0000: 01 03 02 00 00 00 2B                              |......+         |
13:29:03.566 -> [D] 7623| ModbusServerRTU.cpp  [ 185] serve: Response sent.
13:29:03.566 -> [D] Received packet: @3FFBDB4C/1:
13:29:03.566 ->   | 0000: E5                                                |.               |
13:29:03.566 -> [1;33m[E] 7638| ModbusServerRTU.cpp  [ 193] serve: RTU receive: E5 - Packet length error
```

KevinDesbonnets, 2021-08-03:
If client is shutdown the sever verbose is the same:

```
13:33:04.302 -> [D] 5011| ModbusClientRTU.cpp  [  76] begin: Worker task 1073452664 started. Interval=1822
13:33:04.302 -> [V] 5013| ModbusMessage.cpp    [ 529] checkData: Check data #3
13:33:04.302 -> [D] 5019| ModbusClientRTU.cpp  [ 111] addRequestM: request for 01/03
13:33:04.302 -> [D] 5025| ModbusClientRTU.cpp  [ 160] addToQueue: RC=01
13:33:04.302 -> [D] 5030| ModbusClientRTU.cpp  [ 122] addRequestM: RC=00
13:33:04.395 -> [D] 5111| ModbusClientRTU.cpp  [ 178] handleConnection: Pulled request from queue
13:33:04.395 -> [D] Sent packet: @3FFB9804/6:
13:33:04.395 ->   | 0000: 01 03 00 01 00 01                                 |......          |
13:33:04.395 -> [D] 5117| ModbusClientRTU.cpp  [ 183] handleConnection: Request sent.
13:33:04.395 -> [D] Received packet: @3FFBDB4C/1:
13:33:04.395 ->   | 0000: E5                                                |.               |
13:33:04.395 -> [D] 5132| ModbusClientRTU.cpp  [ 189] handleConnection: Error response (1 bytes) received.
13:33:04.395 -> [V] Data: @3FFBDB4C/1:
13:33:04.395 ->   | 0000: E5                                                |.               |
13:33:04.442 -> [D] 5149| ModbusClientRTU.cpp  [ 218] handleConnection: Response generated.
13:33:04.442 -> [V] Response packet: @3FFBDB4C/3:
13:33:04.442 ->   | 0000: 01 83 E5                                          |...             |
13:33:04.442 -> [1;33m[E] 5166| eModBusMaster.ino    [  43] handleError: Error response: E5 - Packet length error
```

Miq1, 2021-08-03:
You are switching the server/client names (ModbusClient is the client, ModbusServer is th server), but anyway. 

If you shut off the server and the client still is receiving data (the E5 byte) in the very millisecond (13:33:04.395), that must have been lingering in the Serial buffer from a previous response or is put there by another call. Since one byte is too short for a proper Modbus response, the client is generating an error response instead that incidentally has the E5 error code - packet length error. 

The server initially gets a stray E5 byte as well, and the request that is received next has an obsolete 00 byte at the end. 

The response looks good on the server side, but seemingly never is received at the client. 

Is it possible you are using the same Serial interface on server and/or client for other purposes than the Modbus communications? The stray bytes must be sent somewhere either in the client or in the server.

Miq1, 2021-08-03:
In your initial post you showed the initialization of both server and client.I suppose both code snippets are run on different MCUs of course?
second, the Serial.begin() calls better be before defining the ModbusServer and ModbusClient using them - just in case..

KevinDesbonnets, 2021-08-03:
Sorry for the confusion

`Serial2` begin is call before Server/Client Start/Begin

Yes I use two different Olimex-ESP32-poe mcu

The only other Serial I use is `Serial` to see debug message

`Serial2` is only use for modbus purpose

Miq1, 2021-08-03:
Erm. I have no clue yet what might be causing this. 😟
Could you post both your sketches in full? I must have a look at all detail I am afraid.

KevinDesbonnets, 2021-08-03:
Thanks you for helping me :)

Master:
```c
// =================================================================================================
// eModbus: Copyright 2020 by Michael Harwerth, Bert Melis and the contributors to ModbusClient
//               MIT license - see license.md for details
// =================================================================================================

// Example code to show the usage of the eModbus library. 
// Please refer to root/Readme.md for a full description.

// Includes: <Arduino.h> for Serial etc.
#include <Arduino.h>

// Include the header for the ModbusClient RTU style
#include "ModbusClientRTU.h"
#include "HardwareSerial.h"

// For demonstration, use the LOG statements for output
#include "Logging.h"

#define TXEN 14
#define RXEN 5

// Create a ModbusRTU client instance
// In my case the RS485 module had auto halfduplex, so no second parameter with the DE/RE pin is required!
ModbusClientRTU MB(Serial2,TXEN, RXEN);

// Define an onData handler function to receive the regular responses
// Arguments are Modbus server ID, the function code requested, the message data and length of it, 
// plus a user-supplied token to identify the causing request
void handleData(ModbusMessage response, uint32_t token) 
{
  // Only print out result of the "real" example - not the request preparing the field
  if (token > 1111) {
    LOG_N("Response: serverID=%d, FC=%d, Token=%08X, length=%d:\n", response.getServerID(), response.getFunctionCode(), token, response.size());
    HEXDUMP_N("Data", response.data(), response.size());
  }
}

// Define an onError handler function to receive error responses
// Arguments are the error code returned and a user-supplied token to identify the causing request
void handleError(Error error, uint32_t token) 
{
  // ModbusError wraps the error code and provides a readable error message for it
  ModbusError me(error);
  LOG_E("Error response: %02X - %s\n", (int)me, (const char *)me);
}

// Setup() - initialization happens here
void setup() {
// Init Serial monitor
  Serial.begin(115200);
  while (!Serial) {}
  Serial.println("__ OK __");

// Set up Serial2 connected to Modbus RTU
// (Fill in your data here!)
  Serial2.begin(19200, SERIAL_8N1, GPIO_NUM_36, GPIO_NUM_4);
  while (!Serial2) {}
  delay(5000);
// Set up ModbusRTU client.
// - provide onData handler function
  MB.onDataHandler(&handleData);
// - provide onError handler function
  MB.onErrorHandler(&handleError);
// Set message timeout to 2000ms
  MB.setTimeout(2000);
// Start ModbusRTU background task
[... 136 lines cut ...]
// Setup() - initialization happens here
void setup() {
// Init Serial monitor
  Serial.begin(115200);
  while (!Serial) {}
  Serial.println("__ OK __");

// Init Serial2 connected to the RTU Modbus
// (Fill in your data here!)
  Serial2.begin(19200, SERIAL_8N1, GPIO_NUM_36, GPIO_NUM_4);
  while (!Serial2) {}
  delay(5000);

// Register served function code worker for server 1, FC 0x03
  MBserver.registerWorker(0x01, READ_HOLD_REGISTER, &FC03);

// Start ModbusRTU background task
  MBserver.start();
}

// loop() - nothing done here today!
void loop() {
  delay(10000);
}
```

Miq1, 2021-08-03:
Let me streamline the code bit to see better.

Client:
```
#include <Arduino.h>

// Include the header for the ModbusClient RTU style
#include "ModbusClientRTU.h"
#include "HardwareSerial.h"
#include "Logging.h"

#define TXEN 14
#define RXEN 5

ModbusClientRTU MB(Serial2,TXEN, RXEN);

void handleData(ModbusMessage response, uint32_t token) 
{
  LOG_N("Response: serverID=%d, FC=%d, Token=%08X, length=%d:\n", response.getServerID(), response.getFunctionCode(), token, response.size());
  HEXDUMP_N("Data", response.data(), response.size());
}

void handleError(Error error, uint32_t token) 
{
  // ModbusError wraps the error code and provides a readable error message for it
  ModbusError me(error);
  LOG_E("Error response: %02X - %s\n", (int)me, (const char *)me);
}

// Setup() - initialization happens here
void setup() {
// Init Serial monitor
  Serial.begin(115200);
  while (!Serial) {}
  Serial.println("__ OK __");

// Set up Serial2 connected to Modbus RTU
  Serial2.begin(19200, SERIAL_8N1, GPIO_NUM_36, GPIO_NUM_4);
  while (!Serial2) {}
  delay(5000);
// Set up ModbusRTU client.
  MB.onDataHandler(&handleData);
  MB.onErrorHandler(&handleError);
  MB.setTimeout(2000);

// Start ModbusRTU background task
  MB.begin();

  uint32_t Token = 1111;

// Read register
  Error err = MB.addRequest(Token++, 1, READ_HOLD_REGISTER, 1, 1);
  if (err!=SUCCESS) {
    ModbusError e(err);
    LOG_E("Error creating request: %02X - %s\n", (int)e, (const char *)e);
  }
}

// loop() - nothing done here today!
void loop() {
}
```

Server:
```
#include <Arduino.h>
#include "HardwareSerial.h"

// Modbus server include
#include "ModbusServerRTU.h"

#define TXEN 14
#define RXEN 5

ModbusServerRTU MBserver(Serial2, 5000, TXEN, RXEN);

// FC03: worker do serve Modbus function code 0x03 (READ_HOLD_REGISTER)
ModbusMessage FC03(ModbusMessage request) {
  uint16_t address;           // requested register address
  uint16_t words;             // requested number of registers
  ModbusMessage response;     // response message to be sent back

  // get request values
  request.get(2, address);
  request.get(4, words);

  // Address and words valid? We assume 10 registers here for demo
  if (address && words && (address + words) <= 10) {
    // Looks okay. Set up message with serverID, FC and length of data
    response.add(request.getServerID(), request.getFunctionCode(), (uint8_t)(words * 2));
    // Fill response with requested data
    for (uint16_t i = address; i < address + words; ++i) {
      response.add(42 + i);
    }
  } else {
    // No, either address or words are outside the limits. Set up error response.
    response.setError(request.getServerID(), request.getFunctionCode(), ILLEGAL_DATA_ADDRESS);
  }
  return response;
}

// Setup() - initialization happens here
void setup() {
// Init Serial monitor
  Serial.begin(115200);
  while (!Serial) {}
  Serial.println("__ OK __");

// Init Serial2 connected to the RTU Modbus
  Serial2.begin(19200, SERIAL_8N1, GPIO_NUM_36, GPIO_NUM_4);
  while (!Serial2) {}
  delay(5000);

// Register served function code worker for server 1, FC 0x03
  MBserver.registerWorker(0x01, READ_HOLD_REGISTER, &FC03);

// Start ModbusRTU background task
  MBserver.start();
}

// loop() - nothing done here today!
void loop() {
  delay(10000);
}
```

Nothing obvious...

The only think making me frown a bit is 
```
#define TXEN 14
#define RXEN 5
```
The GPIO pins are not necessarily numbered the same as they are labelled. I rather would write
```
#define TXEN GPIO_NUM_14
#define RXEN GPIO_NUM_5
```
If you are unlucky, you may hit one of the Serial pins with plain pin numbers.

KevinDesbonnets, 2021-08-04:
I just made the change and nothing append

Miq1, 2021-08-04:
Dang. The E5 byte must be sent somewhere, and it would be a major coincidence if it was not an E5 error code. The problem is, a single byte normally never will be sent over the line alone, so it must be injected erroneously somewhere in the code.
If I would not have seen your code I would have bet for a ``Serial2.print(err);`` statement somewhere.

KevinDesbonnets, 2021-08-04:
It's been 2 week i'm on it, it's not my first time on IoT but it's the first time I code on Arduino, I think I missed something due to the lack of knowledge...

Miq1, 2021-08-04:
No, your code looks fine! What is intriguing me is that the E5 bytes seem to pop up out of the blue, but repeatable so. So there must be some code generating it, that closely is coupled to the regular program function. I just can not see it yet.

KevinDesbonnets, 2021-08-05:
There is no way to just flush data before trying to receive an answer?

Miq1, 2021-08-05:
The flush is in fact done. The E5 byte is coming from somewhere else I am sure. Either your adapters are misbehaving or there is another task interfering or whatever. I am sorry I cannot give you a more specific hint at this point. 

Do you have a logic analyzer? Would be interesting to watch the Serial2 interfaces on both ESPs to see where and when the byte pops up.

KevinDesbonnets, 2021-08-06:
Unfortunaly no

I just find 2 max485, I will try with it  this week end to see if it's come from the module or not

Miq1, 2021-08-06:
I can recommend these [just an example, there ar enumerous](https://www.amazon.com/-/en/dp/B07K6HXDH1) - all around 10 bucks...

bertmelis, 2021-08-06:
Chiming in here. Could you just test with the two ESPs connected to each other, without the MAX485?

The code doesn't care about the medium over which the data is transmitted.

Miq1, 2021-08-09:
@KevinDesbonnets Anything new on this?

KevinDesbonnets, 2021-08-09:
The logic analyzer come tomorow, I will try @bertmelis idea tomorow too

KevinDesbonnets, 2021-08-11:
I'm a bit late, the delivery was delayed

If I just pin two controller directly with tx and rx pin i have a timeout (E0 error), maybe i did something wrong

I connected the uext connector of the master directly on the analyzer and something strange happen : Why RE/DE pin is always HIGH? 

[image]

Miq1, 2021-08-11:
So we know that the Modbus request is sent correctly. The server gets it, but obviously can not respond - possibly due to the always high DE.
Do me a favor: please build your sketch with the unmodified eModbus library once and see if the DE/RE is switched correctly. I know that your hardware does require your modification, but on the LA you should see if it is behaving differently.

KevinDesbonnets, 2021-08-16:
The project build with the unmodified library have exactly the same behavior

[image]

Sorry for slow response, other project was ongoing, I will now be much faster

Miq1, 2021-08-16:
No worries, take your time.

If the unmodified library also has the DE/RE pins constantly pulled HIGH I would suggest checking again your wiring. Normally if you give a pin number!=-1 in the constructor, that pin will be pulled LOW upon initialization. So something there must be wrong.

KevinDesbonnets, 2021-08-16:
I'm sorry I did something wrong (logic analyzer don't record input on 2 pin (reserved for gnd)

If I wire diffently I see this, the normal behavior

[image]

Miq1, 2021-08-16:
Does this hold true with your modified library as well?

If so, we may sort out the DE/RE handling as being the culprit and will have to look further. 

Did you try the direct Serial-to-Serial connection without the RS485 modules, as @bertmelis suggested?

KevinDesbonnets, 2021-08-16:
The previous screen is with the modified library (work with the original on too with only D6 going HIGH and LOW for RE/DE)

I tried but it's seem to don't work with direct connection (I  wired 14 to 14 and 5 to 5 maybe I missed something)

Miq1, 2021-08-16:
You will want to cross the RX and TX lines, so probably 14->5 and 5->14 should be correct.

KevinDesbonnets, 2021-08-17:
I think the issue is solved,

With the logic analyser I saw something strange, data that was not receive was sent correctly.

I stole two different wire (more thick and with some tin at extremity) and now it's work, the wire had bad contact despite the multiple re-wire.

Really sorry for the inconvenience, and thanks to you and your advice I didn't give up so thanks you 😄 

[image]

KevinDesbonnets, 2021-08-17:
I always have error for some reason
[image]

But now the communication look fine
[image]

KevinDesbonnets, 2021-08-17:
Okay now it's work (just needed gnd to be wired)

Miq1, 2021-08-17:
Glad you worked that one out!

The E5 error again is an indication for a failing communication. It mostly is raised if the packet has gaps too wide between bytes. 

In the LA picture there are two transmissions. I suppose both are requests? Would be interesting to see the responses as well to determine the timing on the bus.

KevinDesbonnets, 2021-08-17:
The second is the answer, it's only 1 request here

I have E5 on slave due to the pin going high to low when I wakeup the esp32
[image]

Miq1, 2021-08-17:
Ah, that explains it. You must have set both Serial2 lines identical, but you need to cross them - RX to TX and TX to RX! Request and response have to be sent on different wires.

KevinDesbonnets, 2021-08-17:
No it's normal, I use modbus rs-485 halfduplex, it's why I needed the trick with RE/DE

A and B Send the oposite data (when A high B is low) to remove noise on the communication, RE/DE switch between listen and talk (like talkie-walkie) to make sending and receiving possible on the same wires

Miq1, 2021-08-17:
So you have the LA attached to the RS485 wire? I never did that due to the much higher voltages possible there.

KevinDesbonnets, 2021-08-17:
What is LA?

Miq1, 2021-08-17:
LA = Logic Analyzer. Sorry for being lazy... 

I know about the halfduplex on RS485, but still the Serial lines need to be crossed between RS485 adaptors and the ESPs on both sides.

KevinDesbonnets, 2021-08-17:
The module do this by his own, always send the receice data (ie when RE enable) to rx pin

Miq1, 2021-08-17:
Ok, sorry for being insistent, but I myself mixed that up at times. 

How do you wake up the ESP? Do you have a pulldown resistor on the DE/RE line?

KevinDesbonnets, 2021-08-17:
Right now I juste compile again and again, I needed example to work before try it on more complexe project, so it's quick and dirty.

[Here](https://www.olimex.com/Products/Modules/Interface/MOD-RS485/resources/MOD-RS485.pdf) the documention of the module, I didn't change anything

Miq1, 2021-08-17:
Strange idea to use the I2C pins on the UEXT connector for DE and RE... I am wondering what will happen if someone really will connect these to I2C?

You could have solved your DE/RE problem by a piece of wire connecting SDA and SCL, by the way.

RO is connected to RX and DI to TX on the module, so your Serial2 RX needs to be connected to the module's TX and vice versa, so it seems.

KevinDesbonnets, 2021-08-17:
Don't know, maybe I can try it someday

Yes I know but I can't, for some reason, wire think together, it's why I made it by logical way and not physical

Now I will try to really get datas from a real slave and show them on a website (it's the big project)

If I can help you on any of your project feel free to ask, I'm very grateful of your help, i will be glad if i can return the favor

Miq1, 2021-08-17:
Don't sweat it, I have fun helping! 😄

Miq1, 2021-08-18:
Closing this as issues seem to be resolved.

closed 2021-08-18 by Miq1