eModbus/eModbus #101: Odd data being recieved

moebassist, 2021-08-19:
I'm using ModSim as a Modbus RTU server. For some reason, extra bytes are received:

[image]

Black background = received by Modsim
White background = transmitted by Modsim

After a valid Modbus message is received ([000][001][003][000][000][000][051][005][223][000]

there is sometimes junk data that causes timeout:

[064][032[064]064]....etc. 

Any thoughts?

Thanks!

Miq1, 2021-08-19:
Please see the parallel issue thread, where we currently are trying tosolve a very similar problem. There are 00 bytes observed before and after the message proper, exactly as you do. The original poster had Modsim employed as well, whereas @KevinDesbonnets has a real server connected.

moebassist, 2021-08-19:
Fixed this issue - wiring was inverted and the serial constructor was called with 'invert=true'. Obviously something weird in the serial port implementation for inversion.

closed 2021-08-19 by moebassist

Miq1, 2021-08-19:
Wonderful! I love issues disappearing! 🤣