eModbus/eModbus #104: Packet length error

fanzypantz, 2021-09-20:
#### Do not use to discuss topics! 

I'm getting a packet length error with a SN65HVD10DR chip, and was wondering what could cause a packet length error. It's no longer timing out so there must be some sort of communication going on. 

The result I was looking for was receiving wattage data from a meter.

My code:
```c++
#include <Arduino.h>
#include <ModbusClientRTU.h>

#define DERE 27  
// Create a ModbusRTU client instance
ModbusClientRTU MB(Serial2, DERE);
int modbusRead = 3;

unsigned long previousMillis = 0;
const long interval = 5 * 1000;

// Define an onData handler function to receive the regular responses
// Arguments are Modbus server ID, the function code requested, the message data and length of it, 
// plus a user-supplied token to identify the causing request
void handleData(ModbusMessage response, uint32_t token) {
    // Serial.print("\n");
    // Serial.printf("Response: serverID=%d, FC=%d, Token=%08X, length=%d:\n", response.getServerID(), response.getFunctionCode(), token, response.size());
    // for (auto& byte : response) {
    //     Serial.printf("%02X ", byte);
    // }
    // Serial.println("");
    uint16_t wattage;
    
    response.get(modbusRead, wattage);
    Serial.print(F("[EVSE] Modbus reading: "));
    Serial.println(wattage); 
}

// Define an onError handler function to receive error responses
// Arguments are the error code returned and a user-supplied token to identify the causing request
void handleError(Error error, uint32_t token) {
    // ModbusError wraps the error code and provides a readable error message for it
    ModbusError me(error);
    Serial.printf("Error response: %02X - %s\n", (int)me, (const char *)me);
}

void setup() {
    while (!Serial);
    Serial.begin(115200);
    delay(100);  
    Serial.println(F("Starting"));
      
    // Set up Serial2 connected to Modbus RTU
    Serial2.begin(9600, SERIAL_8N1, GPIO_NUM_16, GPIO_NUM_17);  
    // Set up ModbusRTU client.
    // - provide onData handler function
    MB.onDataHandler(&handleData);
    // - provide onError handler function
    MB.onErrorHandler(&handleError);
    // Set message timeout to 2000ms
    MB.setTimeout(2000);
    // Start ModbusRTU background task
    MB.begin();
}

void checkModbus() {
    uint32_t Token = 1111;

    Error err = MB.addRequest(Token++, 1, READ_HOLD_REGISTER, 0x0141, 0001);
        if (err!=SUCCESS) {
        ModbusError e(err);
        Serial.printf("Error creating request: %02X - %s\n", (int)e, (const char *)e);
    }
}

void loop() {
    // For non blocking delay
    unsigned long currentMillis = millis();

    if (currentMillis - previousMillis >= interval) {
        previousMillis = currentMillis;

        checkModbus();
    }

}

```

This is on a custom board, and here is my current schematic:
[image]
 
If it is relevant, the modbus reader is the [Circutor CEM-C6-MID](http://docs.circutor.com/docs/M187B01-03.pdf)

Miq1, 2021-09-20:
Can you compile your sketch with the ``-DLOG_LEVEL=6`` compiler flag and show us the Serial output for it? Then we can see the bytes going back and forth in detail.

I was wondering why you put the 10 Ohms resistors in the A and B lines?

There is no termination on the RS485 connector - intentionalky?

fanzypantz, 2021-09-21:
Here is the output with the compiler flag set:

```
[D] 114| ModbusClientRTU.cpp  [ 100] setTimeout: Timeout set to 2000
[D] 114| ModbusClientRTU.cpp  [  72] begin: Worker task 1073452664 started. Interval=3645
[V] 5000| ModbusMessage.cpp    [ 529] checkData: Check data #3
[D] 5000| ModbusClientRTU.cpp  [ 107] addRequestM: request for 01/03
[D] 5000| ModbusClientRTU.cpp  [ 156] addToQueue: RC=01
[D] 5001| ModbusClientRTU.cpp  [ 174] handleConnection: Pulled request from queue
[D] 5005| ModbusClientRTU.cpp  [ 118] addRequestM: RC=00
[D] Sent packet: @3FFB981C/6:
  | 0000: 01 03 01 41 00 01                                 |...A..          |
[D] 5026| ModbusClientRTU.cpp  [ 179] handleConnection: Request sent.
[D] Received packet: @3FFBDB4C/1:
  | 0000: E5                                                |.               |
[D] 5042| ModbusClientRTU.cpp  [ 185] handleConnection: Error response (1 bytes) received.
[V] Data: @3FFBDB4C/1:
  | 0000: E5                                                |.               |
[D] 5059| ModbusClientRTU.cpp  [ 214] handleConnection: Response generated.
[V] Response packet: @3FFBDB4C/3:
  | 0000: 01 83 E5                                          |...             |
Error response: E5 - Packet length error
```

As for the resistors, I've understood them to be there to limit residual clamping current to prevent latching up. But I'm not the electrical engineer.

Miq1, 2021-09-21:
You seem to have used an older version of the lib. Would you mind downloading a fresh copy and recompile your program with that? Then we will have a "Raw buffer" dump in the log that will show us what has been received originally. My bets are it will be something with additional 00 bytes in it, as we have encountered that in other cases recently.

fanzypantz, 2021-09-21:
`lib_deps = 
  ModbusClient=https://github.com/eModbus/eModbus.git` Should do that fine right? Seems like it has slightly different outputs

```
[D] 114| ModbusClientRTU.cpp  [ 100] setTimeout: Timeout set to 2000
[D] 114| ModbusClientRTU.cpp  [  72] begin: Worker task 1073452664 started. Interval=3645
[V] 5000| ModbusMessage.cpp    [ 529] checkData: Check data #3      
[D] 5000| ModbusClientRTU.cpp  [ 107] addRequestM: request for 01/03
[D] 5000| ModbusClientRTU.cpp  [ 156] addToQueue: RC=01
[D] 5001| ModbusClientRTU.cpp  [ 174] handleConnection: Pulled request from queue
[D] 5005| ModbusClientRTU.cpp  [ 118] addRequestM: RC=00
[D] Sent packet: @3FFB981C/6:
  | 0000: 01 03 01 41 00 01                                 |...A..          |
[D] 5026| ModbusClientRTU.cpp  [ 179] handleConnection: Request sent.
[D] Raw buffer received: @3FFB2068/1:
  | 0000: 00                                                |.               |
[D] Received packet: @3FFBDB4C/1:
  | 0000: E5                                                |.               |
[D] 5052| ModbusClientRTU.cpp  [ 185] handleConnection: Error response (1 bytes) received.
[V] Data: @3FFBDB4C/1:
  | 0000: E5                                                |.               |
[D] 5069| ModbusClientRTU.cpp  [ 214] handleConnection: Response generated.   
[V] Response packet: @3FFBDB4C/3:
  | 0000: 01 83 E5                                          |...             |
Error response: E5 - Packet length error
```

Miq1, 2021-09-21:
Yes, that is what I expected to see, though I am not really amused...

You can see the solitairy 00 byte being received? This is an artefact generated somewhere between the RS485 A/B connectors on the adapter and the RX FIFO on the ESP32. There are others experiencing that as well (see the parallel open issue).

Since it does happen only with RS485 adapters with RE/DE lines, it must have to do with these. Adapters with auto halfduplex handling are not affected.

We do not know yet the root cause, unfortunately. It could be the timing of pulling and dropping the RE/DE lines is on the edge - there are some reports on the Net regarding that.