eModbus/eModbus #115: Compilation Error on PlatformIO

LeoCMade, 2021-11-12:
#### Do not use to discuss topics! 

**Describe the bug**
eModbus does not compile

**Expected behavior**
.pio/libdeps/az-delivery-devkit-v4/eModbus/src/ModbusClientTCP.cpp:75:26: error: 'TCP' directive output may be truncated writing 3 bytes into a region of size between 2 and 4 [-Werror=format-truncation=] 

**To Reproduce**
Compile eModbus in a PlatformIO ESP32 project

**Example code**
```
any code
```

**Additional context**

Miq1, 2021-11-12:
Did you try that with a more recent version of the eModbus lib? To verify - can you list a couple of lines around the line 75 in ``ModbusClientTCP.cpp``, that the error message mentions?

We had a different length of the ``taskName`` variable (12) there before, but that has been changed to 18 a while ago already.

LeoCMade, 2021-11-12:
Hello, 
i installed the library v1.2.0 (the latest release available) from PlatformIO, it is different from the last commit as you describe.

Miq1, 2021-11-12:
I see. For the time being, please use ``eModbus = https://github.com/eModbus/eModbus`` in the ``lib_deps`` section of your ``platformio.ini``. Looks like PlatformIO needs a formal release to incorporate the recent changes.

Miq1, 2021-11-12:
Just released 1.3 incorporating the changes since on the PlatformIO registry.

closed 2021-11-12 by Miq1