eModbus/eModbus #129: Modbus Server over ethernet not compiling

Miq1, 2021-12-23:
### Discussed in https://github.com/eModbus/eModbus/discussions/128

<div type='discussions-op-text'>

<sup>Originally posted by **codePavan** December 23, 2021</sup>
Hello I am trying to make a code for the Modbus server over ethernet

IDE: platform io
```
[env:esp32doit-devkit-v1]
platform = espressif32
board = esp32doit-devkit-v1
framework = arduino
lib_deps = 
	me-no-dev/AsyncTCP@^1.1.1
	arduino-libraries/Ethernet@^2.0.0
	miq19/eModbus@^1.3.0
```

Code 

```
// =================================================================================================
// eModbus: Copyright 2020 by Michael Harwerth, Bert Melis and the contributors to ModbusClient
//               MIT license - see license.md for details
// =================================================================================================
// Includes: <Arduino.h> for Serial etc., WiFi.h for WiFi support
#include <Arduino.h>
#define HAS_ETHERNET 1
#include "ModbusServerEthernet.h"

ModbusServerEthernet MBserver;

// Create server
uint16_t memo[32]; // Test server memory: 32 words

// Server function to handle FC 0x03 and 0x04
ModbusMessage FC03(ModbusMessage request)
{
  Serial.println(request);
  ModbusMessage response; // The Modbus message we are going to give back
  uint16_t addr = 0;      // Start address
  uint16_t words = 0;     // # of words requested
  request.get(2, addr);   // read address from request
  request.get(4, words);  // read # of words from request

  // Address overflow?
  if ((addr + words) > 20)
  {
    // Yes - send respective error response
    response.setError(request.getServerID(), request.getFunctionCode(), ILLEGAL_DATA_ADDRESS);
  }
  // Set up response
  response.add(request.getServerID(), request.getFunctionCode(), (uint8_t)(words * 2));
  // Request for FC 0x03?
  if (request.getFunctionCode() == READ_HOLD_REGISTER)
  {
    // Yes. Complete response
    for (uint8_t i = 0; i < words; ++i)
    {
      // send increasing data values
      response.add((uint16_t)(memo[addr + i]));
    }
  }
  else
  {
    // No, this is for FC 0x04. Response is random
    for (uint8_t i = 0; i < words; ++i)
    {
      // send increasing data values
      response.add((uint16_t)random(1, 65535));
    }
  }
  // Send response back
  return response;
}

// Setup() - initialization happens here
void setup()
{
  // Init Serial monitor
  Serial.begin(115200);
  Serial.println("__ OK __");
  Ethernet.init(5);
  IPAddress ip(192,168,0,50);
  byte mac[6]={0xAA,0xAB,0xAC,0xAD,0xAE,0xAF};
  Ethernet.begin(mac,ip);
  // Set up test memory
  for (uint16_t i = 0; i < 32; ++i)
  {
    memo[i] = 100 + i;
  }
  MBserver.registerWorker(1, READ_HOLD_REGISTER, &FC03);
  MBserver.start(502, 2, 20000);
}

// loop() - nothing done here today!
void loop()
{
  static uint32_t lastMillis = 0;
  if (millis() - lastMillis > 1000)
  {
    lastMillis = millis();
    Serial.printf("free heap: %d\n", ESP.getFreeHeap());
  }
}
```

Compilation error:
```
Compiling .pio\build\esp32doit-devkit-v1\src\main.cpp.o
In file included from .pio\libdeps\esp32doit-devkit-v1\eModbus\src/ModbusServerEthernet.h:11:0,
                 from src\main.cpp:9:
.pio\libdeps\esp32doit-devkit-v1\eModbus\src/ModbusServerTCPtemp.h: In instantiation of 'static void ModbusServerTCP<ST, CT>::serve(ModbusServerTCP<ST, CT>*) 
[with ST = EthernetServer; CT = EthernetClient]':
.pio\libdeps\esp32doit-devkit-v1\eModbus\src/ModbusServerTCPtemp.h:162:45:   required from 'bool ModbusServerTCP<ST, CT>::start(uint16_t, uint8_t, uint32_t, int) [with ST = EthernetServer; CT = EthernetClient; uint16_t = short unsigned int; uint8_t = unsigned char; uint32_t = unsigned int]'
src\main.cpp:73:28:   required from here
.pio\libdeps\esp32doit-devkit-v1\eModbus\src/ModbusServerTCPtemp.h:221:8: error: cannot declare variable 'server' to be of abstract type 'EthernetServer'     
     ST server(myself->serverPort);
        ^
In file included from .pio\libdeps\esp32doit-devkit-v1\eModbus\src/ModbusServerEthernet.h:9:0,
                 from src\main.cpp:9:
.pio\libdeps\esp32doit-devkit-v1\Ethernet\src/Ethernet.h:253:7: note:   because the following virtual functions are pure within 'EthernetServer':
 class EthernetServer : public Server {
       ^
In file included from C:\users\pavan\.platformio\packages\framework-arduinoespressif32\cores\esp32/Arduino.h:152:0,
                 from src\main.cpp:7:
C:\users\pavan\.platformio\packages\framework-arduinoespressif32\cores\esp32/Server.h:28:18: note:     virtual void Server::begin(uint16_t)
     virtual void begin(uint16_t port=0) =0;
                  ^
In file included from .pio\libdeps\esp32doit-devkit-v1\eModbus\src/ModbusServerEthernet.h:11:0,
                 from src\main.cpp:9:
.pio\libdeps\esp32doit-devkit-v1\eModbus\src/ModbusServerTCPtemp.h:244:5: error: 'class EthernetServer' has no member named 'end'
     server.end();
     ^
*** [.pio\build\esp32doit-devkit-v1\src\main.cpp.o] Error 1
========================= [FAILED] Took 2.42 seconds =========================
The terminal process "C:\Users\pavan\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.
```
</div>

Miq1, 2021-12-23:
This definitely is a bug - class ``EthernetServer`` has no ``end()`` method that ``WifiServer`` has and needs.

Miq1, 2021-12-24:
This is a tricky one - the W5100/W5200/etc. chips used on the Ethernet boards seem to have no way to stop a running session other than to wait for it to timeout. This is the reason the ``Ethernet.h`` in the standard library does not provide an ``end()`` call for the ``EthernetServer``.

I think this can be solved easily only by using a preprocessor ``#define SERVER_END server.end()``, which I consider ugly. The C++ SFINAE tricks will work only with some additional overhead I do not like as well.

I created a ``ServerEnd`` branch that has the required preprocessor additions. 

@codePavan: can you give it a try?

Miq1, 2021-12-24:
Addendum: forgot to modify the Bridge sources - these will need the defines as well.

Miq1, 2021-12-25:
Done.

codePavan, 2021-12-25:
Thankyou. I will try it.

Miq1, 2021-12-28:
@codePavan Any news on this?

codePavan, 2021-12-28:
In my application, I had to use both the "ModbusServerEthernet.h", "ModbusServerWiFi.h" where the end user decides what they need to use based on the available infrastructure. So I added a empty server.end() in ethernet library

Kindly find the changes below in custom-Ethernet
Ethernet.h
[image]

ethernetServer.cpp
[image]

This is working well for me.

Miq1, 2021-12-29:
Thanks, I take it you are ahead of this issue now.

The modified lib should work with including both the WiFi and Ethernet varieties without the need to have your Ethernet.h modified, so if you find some time to try that nevertheless I would be happy to get your results.

Miq1, 2022-01-07:
Merged into master.

closed 2022-01-07 by Miq1