eModbus/eModbus #136: RTU sniffer

Miq1, 2022-01-10:
Given the majority of issues the users seem to have are concerning RTU - would it be helpful to have something like ``ModbusRTUsniffer`` class? It could be made listening to the RTU bus it is connected to and collect and filter all traffic on the bus in a queue with time stamps. A main task then could start/stop/reset the sniffer and set filtering rules for it. 
Reading the queue the captured messages can be logged, displayed or whatnot.

Worth investing time into that?

Miq1, 2022-01-11:
> Would a sniffer be able to do that? Read data as an "invisible" device in the Modbus network?

Yes, that could be an application. You would of course need to separate the SmartEVSE requests from the meters' responses in your sketch.

brainstorm, 2022-03-15:
Yes please, @Miq1, that a sniffer that works on ESP32 modules out of the box would be quite handy since I'm now toying with a ModBus-RTU power meter (see below on the diff output for a non-out-of-the-box minor issue I just had and circumvented):

https://github.com/brainstorm/yigedianqi-modbus

Where I've run some python tests with the computer instead of using microcontrollers:

<details><summary>Python minimalmodbus code</summary>

```python
#!/usr/bin/env python
import minimalmodbus
from time import sleep

minimalmodbus.BAUDRATE = 9600
instrument = minimalmodbus.Instrument('/dev/ttyUSB0', 1) # port name, slave address (in decimal)
#instrument = minimalmodbus.Instrument('/dev/ttyUSB1', 1) # port name, slave address (in decimal)

# First three registers are integers
print("Int register {} has value: {}".format(2, instrument.read_register(2, functioncode=3, numberOfDecimals=1)))
print("Int register {} has value: {}".format(3, instrument.read_register(3, functioncode=3, numberOfDecimals=1)))
print("Int register {} has value: {}".format(4, instrument.read_register(4, functioncode=3, numberOfDecimals=1)))

while True:
    try:

        # Followed by registers 5-21 which are "reserved", so we ignore them
        #for reg in range(5,21):
        #    val = instrument.read_register(reg, functioncode=3, numberOfDecimals=3)
        #    print("Int register {} has value: {}".format(reg, val))

        #  Register 22 is labeled as "Remote message" in chinese, whatever that means...
        #  print("Int register {} has value: {}".format(22, instrument.read_register(22, functioncode=3, numberOfDecimals=2)))

        # The rest (from 23 to 80) are pairs of floats
        for reg in range(23,80,2):
            val = instrument.read_float(reg, functioncode=3, numberOfRegisters=2)
            print("Float register {}, {} has value: {}".format(reg, reg+1, val))
    except IOError:
        continue
    except ValueError:
        continue
    break
```
</details>
<details><summary>Python minimalmodbus output</summary>

```
$ ./minimodbus.py 
Int register 2 has value: 0.0
Int register 3 has value: 0.0
Int register 4 has value: 0.0
Float register 23 has value: 0.0
Float register 25 has value: 0.0
Float register 27 has value: 0.0
Float register 29 has value: 99.9990005493
Float register 31 has value: 0.0
Float register 33 has value: 0.0
Float register 35 has value: 0.0
Float register 37 has value: 0.0
Float register 39 has value: 0.0
Float register 41 has value: 0.0
Float register 43 has value: 0.0
Float register 45 has value: 0.0
Float register 47 has value: 0.0
Float register 49 has value: 0.0
Float register 51 has value: 0.0
Float register 53 has value: 0.0
Float register 55 has value: 0.0
Float register 57 has value: 0.0
Float register 59 has value: 0.0
Float register 61 has value: 0.0
Float register 63 has value: 0.0
Float register 65 has value: 3547.2199707
Float register 67 has value: 3547.2199707
Float register 69 has value: 3547.2199707
Float register 71 has value: 0.0
Float register 73 has value: 0.0
Float register 75 has value: 0.0
Float register 77 has value: 0.0
Float register 79 has value: 1.40129846432e-42
```
</details>

And [now I'm using an ESP32 Saola Board](https://github.com/brainstorm/yigedianqi-modbus), for which I've had to change the Serial2-related code on eModBus (maybe worth parametrising for boards that do not have that many serial ports?):

```c++
src/main.cpp:33:20: error: 'Serial2' was not declared in this scope
 ModbusClientRTU MB(Serial2, REDEPIN);
                    ^~~~~~~
src/main.cpp:33:20: note: suggested alternative: 'Serial1'
 ModbusClientRTU MB(Serial2, REDEPIN);
                    ^~~~~~~
                    Serial1
(...)
```

Hacky change in need for proper board detection and handling code:
 
```diff
% diff -u RTUutils.cpp.orig RTUutils.cpp
--- RTUutils.cpp.orig	2022-03-15 13:44:08.900875538 +0000
+++ RTUutils.cpp	2022-03-15 13:48:36.957559540 +0000
@@ -138,6 +138,7 @@
       rc = UART1.conf1.rxfifo_full_thrhd;
       UART1.conf1.rxfifo_full_thrhd = thresholdBytes;
       LOG_D("Serial1 FIFO threshold set to %d (was %d)\n", thresholdBytes, rc);
+    #if 0
     // No, but it may be Serial2
     } else if (&serial == &Serial2) {
       // Found it. Get the current value and set ours.
@@ -145,6 +146,7 @@
       UART2.conf1.rxfifo_full_thrhd = thresholdBytes;
       LOG_D("Serial2 FIFO threshold set to %d (was %d)\n", thresholdBytes, rc);
     // None of the three, so we are at an end here
+    #endif
     } else {
       LOG_W("Unable to identify serial\n");
     }
```

Sorry for those few offtopic bits on this issue (might be worth opening a new one), but I'm just glad your project exists and wanted to share my past and present experience ;)

brainstorm, 2022-03-16:
Not really:

[image]

But anyway, I got it working with that `#if 0` on `RTUUtils.cpp` :)

brainstorm, 2022-03-24:
> Ah, I see - the UART2 is missing on your "Saola" board? How about the further UARTs 3 and 4?

Quoting [espressif docs](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/api-reference/peripherals/uart.html#universal-asynchronous-receiver-transmitter-uart): 

> The ESP32-S2 chip has two UART controllers (also referred to as port), each featuring an identical set of registers to simplify programming and for more flexibility.

Miq1, 2022-03-25:
> > Ah, I see - the UART2 is missing on your "Saola" board? How about the further UARTs 3 and 4?
> 
> Quoting [espressif docs](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/api-reference/peripherals/uart.html#universal-asynchronous-receiver-transmitter-uart) w.r.t to my particular board which has an ESP32S2:
> 
> > The ESP32-S2 chip has two UART controllers (also referred to as port), each featuring an identical set of registers to simplify programming and for more flexibility.

Yeah, I read that, too, but I am wondering why the statements referring to UART3 and UART4 in ``UARTinit()`` are not causing errors from the compiler as well.

brainstorm, 2022-03-26:
Anyway, more related to this issue and after my excellent experience with your lib, I'd love to see a "discovery" sketch instead of a sniffer. The discovery code would send read-only queries and "map" what the device is capable of, perhaps even spitting out a Data Object/CID table as a result, [like this one I have in another branch of my repo](https://github.com/brainstorm/yigedianqi-modbus/blob/8f394dc3c61a1e816e9891aac9759eb6bc2fb346/src/app_modbus.c#L92-L155).

Miq1, 2022-03-26:
My bad, of course it is UART0, UART1 and UART2 only. Do you know any ``#define`` to identify the ESP you are using? then I could do a conditional based on that. Else I would have to have it set in ``options.h``.

Yeah, I also considered such a discovery sketch, but depending on the timeout that would have to run a while - given the 247 possible IDs. And it will not be sufficient to send some undefined FC to tell from the error returns which IDs are used on a certain local bus, because some servers are simply ignoring requests with unknown FCs, relying on the client's timeout.

brainstorm, 2022-03-26:
Something like [this define, which is based on sdkconfig I believe](https://github.com/espressif/arduino-esp32/blob/master/tools/sdk/esp32/include/driver/include/driver/adc_deprecated.h#L21)?

w.r.t the discovery sketch, I'd go with reasonable small subset of default settings and then offer configurable options. For instance, out of the possible 247 IDs, I'd stick with a much smaller 2-10 IDs... on the lines of what [Max Howell does/did for Homebrew](https://www.quora.com/Whats-the-logic-behind-Google-rejecting-Max-Howell-the-author-of-Homebrew-for-not-being-able-to-invert-a-binary-tree).

Miq1, 2022-03-27:
I suppose you are using the yet-to-be-published ``master`` branch of arduino-esp32? Like in
```
[env:arduino-esp32s2]
platform = https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream
board = esp32-s2-saola-1
framework = arduino
platform_packages =
   framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32#master
```
I am still using the last official framework version that does not yet support the newer ESP32 types, so I am in vain looking for matching defines. I suppose there will be something like ``ESP32_S2`` or such that I would like to use to map out the UART2 code.

Miq1, 2022-06-26:
Sniffer functionality added with ``tiny-server`` branch.

```
ModbusServerRTU mbsr;
...
void mySniffer(ModbusMessage m) {
  // do something
}
...
mbsr.registerSniffer(mySniffer);
```

closed 2022-06-26 by Miq1

kuifje123, 2023-12-16:
@fluppie I was wondering, did you succeed to solve your use case? I am also having the exact same idea and was wondering how to solve it.

Bram

fluppie, 2023-12-16:
I did not look further into it, but we tried to use the TCP server from de eModbus lib as a bridge.
https://github.com/serkri/SmartEVSE-3/issues/132 see here