eModbus/eModbus #137: Using library to read soil sensor

Tsaadon, 2022-01-14:
Hello,
I am using ESP32 and RS485 Modbus to read a soil sensor, I have used the esp32ModbusRTU library successfully to do it, however, recently I have changed the RS485 communications chip and now it doesn't work anymore, from what I understand the chips are nearly identical, only the new one I'm using is more reliable. The old chip was SP3485EN-L and new one is MAX3485ESA+.
https://www.findic.us/compare/max3485esa-vs-sp3485en-l-40p6xA93G.html

When I tried examining the pulses coming out of the RO pin I noticed the pulses are reversed between the chips.

I tried using the eModbus library and example 4RTU but couldn't get it to work with either chip. 
SP3485EN-L RO pin pulse
[image]

MAX3485ESA+ RO pin pulse
[image]

This is the code that works successfully for me with esp32ModbusRTU library using SP3485EN-L:
[SoilSensor.txt](https://github.com/eModbus/eModbus/files/7869634/SoilSensor.txt)

Anyone has any ideas as to why it doesn't work for me and how to adapt the example 4RTU so I can read the relevant parameters, with the eModbus library, in my soil sensor?

Thanks!

Miq1, 2022-01-14:
Hi,

What is the RO pin you are mentioning? The control pin of the RS485 chip (usually called RE) or the data receive pin (usually RX)?

Tsaadon, 2022-01-14:
According to the electrical diagram it's RO, but yes it's the data receive pin (RX):
[image]
[image]

Miq1, 2022-01-14:
I see. If the signal is inverted, you may be lucky exchanging the A and B wires at the chip. If all else fails, the ``Serial.begin()`` call has an optional parameter after the timeout value, that when set to true will invert the Serial signals. But I would bet on the first solution with A/B switched.

Tsaadon, 2022-01-14:
Unfortunately switching A and B didn't help. 
Can you give me an example of how to use the Serial.begin() command to switch the signals?

Miq1, 2022-01-16:
Any news on this issue?

Tsaadon, 2022-01-16:
Yea I tried and unfortunately, it didn't help.

Miq1, 2022-01-16:
Too bad. I am still almost sure it is an issue with the cabling...

Tsaadon, 2022-01-23:
Appears you were right, I had 2 resistors missing, when I put them on the signal flipped and now it's reading perfect, thanks!

closed 2022-01-23 by Tsaadon

Miq1, 2022-01-23:
👍🏻