eModbus/eModbus #142: UARTinit() not called with arduino-esp32 core 2.0

Miq1, 2022-01-16:
### Discussed in https://github.com/eModbus/eModbus/discussions/141

<div type='discussions-op-text'>

<sup>Originally posted by **mstegen** January 16, 2022</sup>
Hi again, 

I'm trying arduino core 2.0.2 on my project, and noticed something strange.

In my project the ModbusServerRTU is listening for modbus data from other devices on the bus,  and processing data without sending any data back.
I noticed that the request and response interval time was not working correctly. Both messages were processed as one block of data, and the CRC check failed.
This was not a problem on arduino core 1.0.6. 

After some tests, i noticed that somehow `UARTinit()` is not called. At least i can not see the debug output "UARTinit: Serial1 FIFO threshold set to 1 (was 112)" this is for both 1.0.6 and 2.0.2 versions of arduino core.

So i added the call to `UARTinit()` in the `ModbusServerRTU::start` code. 
Now it's working on core2.0.2

</div>

Miq1, 2022-01-17:
@mstegen I doubt the call is not made, but the initialization we do in there may be overwritten later again.
Which development environment are you using? In PlatformIO, that I usually am using, the 2.0 core has not been deployed yet.

mstegen, 2022-01-17:
Hi @Miq1, i'm using PlatformIO. These are the settings i used for core 2.0.2:
```platform  = https://github.com/platformio/platform-espressif32.git#feature/arduino-idf-master
platform_packages = framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#2.0.2
     platformio/tool-esptoolpy @ https://github.com/tasmota/esptool/releases/download/v3.2/esptool-v3.2.zip
```                    
But even when using core 1.0.6 i can not see the debug output that sets the FIFO threshold to 1.
Do you see it?

mstegen, 2022-01-17:
Did some more tests on core 1.0.6.
You were right, the call to `UARTinit` was made.
But at that time the Uart is not initialized yet, so i guess the changes to the FIFO threshold get overwritten when the Serial port is setup.
Somehow this did not affect the library on 1.0.6 (at least for me), but it does on 2.0.2 
Moving the call to `UARTinit` to `ModbusServerRTU::start` will fix this for both core versions.

Miq1, 2022-01-17:
Thank you for testing that. I think for the time being moving the call as you are suggesting will do no harm. 

I still am waiting to learn what has been improved with the different UART handling in 2.0. The ``UARTinit()`` stuff was only put in to get around the lazy FIFO handling that is totally senseless for applications with small messages like Modbus. The core maintainers were completely ignorant about that...

Miq1, 2022-01-17:
I provided a [commit](https://github.com/eModbus/eModbus/commit/8b292a872b542ea71d0b1472cb228a1060c36e4f) for ``ModbusServerRTU`` and ``ModbusClientRTU``. Would be great if you could do some tests with this.

mstegen, 2022-01-17:
Thanks. I do see some more packet length errors though, when it thinks it receives a single 'E5' byte. 
But Uart handling has improved a lot in 2.0.2 from what i can see. I'm working on a project that uses all three Uarts simultaneously, and i used to see lot's of gaps in reception, that resulted in checksum errors on the data (checksum was required on all three Uarts because of this behavior). 
It's now running for a few hours, and i have not seen one checksum error yet.

Miq1, 2022-01-17:
Can you provide us a log of the E5 errors occurring? Best with the ``-DLOG_LEVEL=6`` compiler flag to see the raw data as well.
E5 can be an indication of bad timing on the bus, most probably done by us, if it worked before. If you got industrial servers on the bus with high baud rates, chances are that the lib is losing pace. We for instance do not care at all about the intra-message gap control that the Modbus standard advises, because we are depending from the UART to deliver the data without any chance to interfere.

mstegen, 2022-01-18:
Here's the logfile, it looks like the E5 error is only happening after the software writes a response to the bus.

[logfileE5.txt](https://github.com/eModbus/eModbus/files/7889982/logfileE5.txt)

Miq1, 2022-01-18:
Thanks!

There is always a stray 0x00 byte on the bus that is causing the E5 error (as I suspected...). In combination with the fact that you had sent a response immediately before that, that strongly points to the signal issue @bertmelis analyzed and has explained thoroughly in the pinned thread in Discussions.

In short, the switch of the DE/RE pin after sending back to listening mode in combination with a suboptimal biasing on the Modbus leads to a three-state effect that is registered as a "ghost 0x00" on UART.

mstegen, 2022-01-18:
Yes, i "fixed" it for now by adding 
`serial.read();` just after the Rts pin of the RS485 transceiver is set to low. 
so it absorbs the ghost 0x00.
If there is nothing to read, it will return immediately, so it should not block or harm in any way i think ?

All E5 errors are gone now..

Miq1, 2022-01-18:
That is a fix of the symptoms, with the threat of more to happen later on. I rather would try to exactly bias my bus with Bert's example. If you have proper biasing and termination on the bus, you will be safe.

mstegen, 2022-01-18:
Yes, i understand that exactly biasing the bus is preferred, however i think it's a bit to sensitive now. It's not that there is random data on the bus that can be removed by biasing the bus. It's always a 0x00.
It's caused by the transceiver switching from RX to TX, and this spike is then mistakenly read by the uart as a 0x00.

The thing is: the uart should -not- listen to the bus at this time, as it is half duplex, it should only start listening again when the transmission has finished. So i think it's ok to reject any data that is received when the transmission is in progress...

And that brings me to this: It also suppresses the 0x00 if you move the `serial.read()` just before the `rts(LOW);` (so before the tx ends)

Btw, on core 1.0.6 the E5 error appeared only once every 10 minutes or so, not every 2 seconds. So they definitely 'improved' the uart code ;-)

Miq1, 2022-01-18:
The 0x00 is caused by toggling RE/DE only, so I am wondering why the UART should have changed anything about that? The time interval would only be affected by the sketch doing the toggle, right?

I agree the UART is not particularly well-suited to do our Modbus communications, but writing our own low-level serial code was no feasible option.

mstegen, 2022-01-18:
With core 2.0.2 it's now much more reliable at receiving all data at all three uarts, at the same time. That might have something to do with it.

I noticed they just added a callback, so that probably means that the polling loop can be removed in the future?
https://github.com/espressif/arduino-esp32/pull/6134

Miq1, 2022-01-18:
That is sounding promising. If only the PlatformIO guys woul lift to 2.0 at least.

Miq1, 2022-02-01:
Closing this, as it seems to work with the moved UARTinit call.

closed 2022-02-01 by Miq1