eModbus/eModbus #153: Sending Server Messages longer than 256 bytes

salakrzy, 2022-02-07:
**Describe the bug**
Hi,
I use ESP32 to addreses translator between DTSU666 and DTSU666H Power Meter. I use in parallel ModbusServerRTU.cpp and  RTUutils.cpp	from  eMODBUS librrary . Everything going well when i read les than 28 float registers. Unfortunetly the SUN2000 Huawei inwertor asking in one request for 80 float registers, this causes the program to crash and watchdogs to restart the system. The problem is probably due to the serial port buffer length limitation of 256 bytes . The MODBUS standard says , **maximum message length is 256 bytes** but as you can see not everyone follows the standard...
I made changes to the ModbusServerRTU.cpp and  RTUutils.cpp files and now my translator works properly.
**ModbusServerRTU.cpp changed from "delay(1)" to "delay(60)"  	row 192**
**RTUutils.cpp	added "delay(70)"     row 223**
Please find attached a screenshot of the monitor before and after this changes.  

[eMODBU_Error 256 bytes.txt](https://github.com/eModbus/eModbus/files/8012926/eMODBU_Error.256.bytes.txt)

**Expected behavior**
Doesn't my solution break something in the other components of the library? My solution works well for 9600 but for other serial port speeds you probably need to set this delay more intelligently.

**To Reproduce**
I have posted the source code and my project description on github

[https://github.com/salakrzy/DTSU666_CHINT_to_HUAWEI_translator](https://github.com/salakrzy/DTSU666_CHINT_to_HUAWEI_translator)

**Example code**
```
Put code here to reproduce the bug, if possible
```

**Additional context**

### Best Regards,

Miq1, 2022-02-07:
Hi,
This strictly speaking is no bug, as you already have seen the Modbus specs do not allow messages longer than 256 bytes.
The timing of Modbus RTU is critical and even more so if increasing the bus speed. So the overflow of the serial FIFO and need to recover or compensate (as you did with the longer delays) will definitely break the bus timing as Modbus RTU requires it.
The better way was to completely write a low level bus handler from scratch that does not rely on the serial implementation of the arduino-esp32 core, but we so far were content with the solution we had.
If your modification does work for you - fine, but for the reasons given we are not going to implement it in the library.

closed 2022-02-07 by Miq1

Miq1, 2022-02-07:
One addendum: there must be something else happening in your case. 

I checked again the Test sketch, since I remembered having implemented oversized messages for user defined function codes. In the test sketch a 400+ byte message is sent and received by function code 0x44 over RTU and is passing the RTU loop test without problems. 

So there is something different stalling your receive in your setup.