eModbus/eModbus #157: Sending message in RTU mode not working (1.4.0)

mstegen, 2022-02-11:
After upgrading to the latest version, sending a modbus message did not work anymore.

This is causing the issue:

https://github.com/eModbus/eModbus/blob/7a5f8877e5ee426ae8eca634e96ec1dfd5724209/src/RTUutils.cpp#L209

moving it to it's original location (before  rts(LOW);  ) fixed it.

mstegen, 2022-02-11:
No problem!

mstegen, 2022-02-11:
Just tested the latest master, and the is fixed.
Thanks.

closed 2022-02-11 by mstegen

Miq1, 2022-02-11:
Thanks a lot! I will publish a hotfix release now.

mstegen, 2022-02-11:
Yes, was expecting that the library manager(?) would pick up the new version in a hour max, but that does not seem to be the case..

Miq1, 2022-02-11:
I would need to ``pio publish`` the release to the Platformio library manager, but pio internally is aborting due to some port error. As I wrote, there are dozens of bug reports for that already.

mstegen, 2022-02-13:
Still no new version available from the Library manager, however the 'changelog' tab in the manager shows the latest release notes.. Weird.

Miq1, 2022-02-13:
I found a workaround - setting the pio port fixed to 9013 - and was able to publish now. The revised version should be available in a few minutes.

mstegen, 2022-02-13:
Yes! working now.

Miq1, 2022-02-13:
:1