eModbus/eModbus #158: Modbus lib causes reboot of device.

EmbeddedDevver, 2022-02-11:
I am using this library for several months now.
I was seeing random reboots every 1 to 48 hours.
This weekend I put a logging device to capture all the serial output on the terminal to pinpoint the random reboots. With this log I discovered that the problem was bc of this library.

It seems that ```Modbus01RTU``` can sometimes block a bit too long causing that the Watchdog does not get serviced in time. 

Note that: 
- The Modbus01RTU is by default at priority 6 at core 0. 
- The WDT is at priority 0 and also core 0. 
- The WiFi library is also (always) at core 0.

The problem is that Modbus01RTU is blocking loop with only a short  `` delay(1);``  So Modbus01RTU  consumes most of the CPU time, then Modbus01RTU  releases the CPU for only 1mS. During this single 1mS the WiFi will be serviced and after that directly the Modbus01RTU gets serviced again and the WDT never gets serviced during this time.

Solution:

1) Move Modbus01RTU to core 1 like this
```    // Start ModbusClientRTU background task
    MBclient.begin(1);
```
This I've tested and works very well for more than 48 hours already. Also by stressing WiFi, no problems happen. Do note that the mainloop and everything else at core 1 will now sometimes have to wait for the Modbus01RTU  to finish his tasks. 

The items below might also work, but did not test:
2) increase the delay from 1mS to e.g. 50mS. Then WiFi can get serviced and the WDT. 
3) disable the WDT (not recommended)
4) Lower the priority of the task to 0 instead of 6. 

Here the relevant log
```
"09/02/2022 18:46:04",E (2386223) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
"09/02/2022 18:46:04",E (2386223) task_wdt:  - IDLE0 (CPU 0)
"09/02/2022 18:46:04",E (2386223) task_wdt: Tasks currently running:
"09/02/2022 18:46:04",E (2386223) task_wdt: CPU 0: Modbus01RTU
"09/02/2022 18:46:04",E (2386223) task_wdt: CPU 1: loopTask
"09/02/2022 18:46:04",E (2386223) task_wdt: Aborting.

"08/02/2022 13:10:55",E (6815517) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
"08/02/2022 13:10:55",E (6815517) task_wdt:  - IDLE0 (CPU 0)
"08/02/2022 13:10:55",E (6815517) task_wdt: Tasks currently running:
"08/02/2022 13:10:55",E (6815517) task_wdt: CPU 0: Modbus01RTU
"08/02/2022 13:10:55",E (6815517) task_wdt: CPU 1: loopTask
"08/02/2022 13:10:55",E (6815517) task_wdt: Aborting.
"08/02/2022 13:10:55",abort() was called at PC 0x400f59a0 on core 0

"08/02/2022 15:14:49",E (7432388) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
"08/02/2022 15:14:49",E (7432388) task_wdt:  - IDLE0 (CPU 0)
"08/02/2022 15:14:49",E (7432388) task_wdt: Tasks currently running:
"08/02/2022 15:14:49",E (7432388) task_wdt: CPU 0: Modbus01RTU
"08/02/2022 15:14:49",E (7432388) task_wdt: CPU 1: loopTask
"08/02/2022 15:14:49",E (7432388) task_wdt: Aborting.
"08/02/2022 15:14:49",abort() was called at PC 0x400f59a0 on core 0

"08/02/2022 15:24:12",E (563585) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
"08/02/2022 15:24:12",E (563585) task_wdt:  - IDLE0 (CPU 0)
"08/02/2022 15:24:12",E (563585) task_wdt: Tasks currently running:
"08/02/2022 15:24:12",E (563585) task_wdt: CPU 0: Modbus01RTU
"08/02/2022 15:24:12",E (563585) task_wdt: CPU 1: loopTask
"08/02/2022 15:24:12",E (563585) task_wdt: Aborting.
"08/02/2022 15:24:12",abort() was called at PC 0x400f59a0 on core 0
```

Miq1, 2022-02-11:
You found the solution I would have recommended - running the Modbus task on core 1.

Extending the delay will break the RTU timing; the lib has problems already with high baud rates, but does work okay up to 19200 or even 38400 so far. Higher baud rates only will work for a client (as is the bus master) or a server with a client ignorant to intra-message gaps. 

The reason is the UART handling in the arduino-esp32 core. In a better world one would write a dedicated low level driver just for RTU without the overhead. You may volunteer... 😉

closed 2022-02-11 by Miq1

EmbeddedDevver, 2022-02-11:
Just came to my mind.

I do understand the problem that you have currently. It is mostly related to the OS (FreeRtos) that has no option (by default) to have less delays than 1mS. Thus the hack/solution that the lib is using now is making the task highest priority so no other task can interrupt and then blocking poll for an answer to see if the response come within the RTU specified timing.

But how about this; use dedicated interrupt timer from the ESP32. 
You know the baud rate, then you know at what interval you need to scan for new data. The interrupt fires without any delay, you can set it to 100uS for example and it will be exactly like that.

You only need to know if there was data in the specified interval. That is what the interrupt routine will do. Then if the interrupt routine know there is data, it can send a semaphore to the same thread that you are using now that data is finished.
By using this, the RTU routine priority can be lowered to just 0.

What do you think?

```
volatile int interruptCounter;
int totalInterruptCounter;
 
hw_timer_t * timer = NULL;
portMUX_TYPE timerMux = portMUX_INITIALIZER_UNLOCKED;
 
void IRAM_ATTR onTimer() {
  portENTER_CRITICAL_ISR(&timerMux);
// check for new data in the uart buffer.
// if true, use semaphore to annouce there is new data.
  interruptCounter++;
  portEXIT_CRITICAL_ISR(&timerMux);
 
}
 
void setup() {
  timer = timerBegin(0, 80, true);
  timerAttachInterrupt(timer, &onTimer, true);
  timerAlarmWrite(timer, 1000000, true);
  timerAlarmEnable(timer);
}
 
void loop() {
 
  if (interruptCounter > 0) {
    portENTER_CRITICAL(&timerMux);
    interruptCounter--;
    portEXIT_CRITICAL(&timerMux);
  }
}
```