eModbus/eModbus #167: Choice of reversing rtsPin

atlesg, 2022-03-17:
Hi! I am using my custom ESP32 board and I want to test communicating with Modbus RTU using TTL-to-RS485 module. 
My problem is that my only possible gpio output pin (that controls the rtsPin) is reversed - because it's routed through an Op-Amp, such that:

**digitalWrite(rstPin, LOW)** equals to **digitalWrite(rstPin, HIGH)** 
(if rstPin is considered as a LED)

Is there anyways to overcome this without touching the library source code? If no, how should I modify the library to suit my need?

Thank you so much!

Miq1, 2022-03-17:
Of course there is a way 😉 

You may use the RTS callback, as described [here in the docs](https://emodbus.github.io/modbusclient-rtu-api#modbusclientrtuhardwareserial-serial-rtscallback-func-and-modbusclientrtuhardwareserial-serial-rtscallback-func-uint16_t-queuelimit). 
This function will be called every time the toggle pin is about to be set and you may do whatever you find sensible in it. The "normal" level value will be given to the callback as a ``bool`` parameter.

closed 2022-03-17 by Miq1

atlesg, 2022-03-18:
Thank you Sir! Works like a charm 👏

Miq1, 2022-03-18:
👍🏻