eModbus/eModbus #184: in need for non-standard function code 0x13

TLS1000, 2022-04-25:
Hi, I've been using this library for about two years now on ESP32, to my great content. But now for some chinese device, I need to write to the input registers (typically the 0X06 is used for this) via code 0x13, according to the manufacturer modbus protocol.  But as expected the addRequest of this message returns 'Error creating request: 01 - Illegal function code'. What would be the easiest way to make an alteration to the library source? I tried commenting the 'if (functionCode == 13) return ILLEGAL_FUNCTION; // FC 13 does not exist' in modbusmessage.ccp but no luck....

Miq1, 2022-04-25:
Augh! 0x13 is explicitly set to ILLEGAL_FUNCTION, because the Modbus Standard does not allow user defined codes in that range ("intentionally left blank"). These Chinese...
It may work to map the code as a user defined code, though, but that will require some more changes.
I could offer to write a service call to explicitly add any FC to the USER_DEFINED group, so you could write your own check code for these codes as you need to.

TLS1000, 2022-04-26:
Hi, the device in question is not one I can easily swap with another one that follows the modbus standard. I guess the manufacturer wanted to incorporate some sort of tamper protection by deviating from the modbus standard. The 0x13 function call is not the only one they've taken the liberty to work with, they've also used other strange function call-numbers. Ideally would be a solution where the user can freely choose a function call number and tell the library 'make the modbus message as if it were 0x05,0x06, 0x10 function call. But I guess that's already determined by the passed arguments to the 'addMessage' function? A free FC to the user defined group could work for me, I have someone working on the project with good programming skills for the check code-part.

Miq1, 2022-04-26:
I am thinking of some call to release any arbitrary FC from the bounds of the standard to have it handled completely by the application.
A message then will have to be set up with ``add()`` calls, starting with the server ID and function code proper. There is one ``addMessage()`` call already that takes a prebuilt ``ModbusMessage`` as a parameter. Of course no message internal parameter checks will be done in it.
Give me a day or so to implement it.

TLS1000, 2022-04-26:
that would be great, thank you.

Miq1, 2022-04-26:
On the way. There is a "FC-redefinition" branch now that already has the required functionality, but is completely untested yet. It compiles and passes the standard test sketch, but the new feature has not been tested.

If you are adventurous enough, you may try it 😉 

There is a set of FC types now:
```
enum FCType : uint8_t {
  FC01_TYPE,              // Two uint16_t parameters (FC 0x01, 0x02, 0x03, 0x04, 0x05, 0x06)
  FC07_TYPE,              // no additional parameter (FCs 0x07, 0x0b, 0x0c, 0x11)
  FC0F_TYPE,              // two uint16_t parameters, a uint8_t length byte and a uint16_t* pointer to array of bytes (FC 0x0f)
  FC10_TYPE,              // two uint16_t parameters, a uint8_t length byte and a uint8_t* pointer to array of words (FC 0x10)
  FC16_TYPE,              // three uint16_t parameters (FC 0x16)
  FC18_TYPE,              // one uint16_t parameter (FC 0x18)
  FCGENERIC_TYPE,         // for FCs not yet explicitly coded (or too complex)
  FCUSER_TYPE,            // No checks except the server ID
  FCILLEGAL_TYPE,         // not allowed function codes
};
```
You can assign a new type to any function code by
```
Error ModbusMessage::redefineFunctionCodeType(uint8_t fc, FCType t);
```
So to use your 0x13 code exactly as 0x10, you will have to put
```
ModbusMessage::redefineFunctionCodeType(0x13, FC10_TYPE);
```
in your ``setup()``. A call with ``addMessage(1, 0x13...`` will (should, as of now) behave exactly like the ``addMessage(1, WRITE_MULT_REGISTERS...`` now.

TLS1000, 2022-04-27:
<html xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:w="urn:schemas-microsoft-com:office:word"
xmlns:m="http://schemas.microsoft.com/office/2004/12/omml"
xmlns="http://www.w3.org/TR/REC-html40">

<head>

<meta name=ProgId content=Word.Document>
<meta name=Generator content="Microsoft Word 15">
<meta name=Originator content="Microsoft Word 15">
<link rel=File-List
href="file:///C:/Users/wouter/AppData/Local/Packages/oice_16_974fa576_32c1d314_2aca/AC/Temp/msohtmlclip1/01/clip_filelist.xml">

<link rel=themeData
href="file:///C:/Users/wouter/AppData/Local/Packages/oice_16_974fa576_32c1d314_2aca/AC/Temp/msohtmlclip1/01/clip_themedata.thmx">
<link rel=colorSchemeMapping
href="file:///C:/Users/wouter/AppData/Local/Packages/oice_16_974fa576_32c1d314_2aca/AC/Temp/msohtmlclip1/01/clip_colorschememapping.xml">

<style>

</style>

</head>

<body lang=en-BE style='tab-interval:36.0pt;word-wrap:break-word'>

<p class=MsoNormal style='line-height:11.95pt;mso-line-height-rule:exactly;
text-autospace:none'><span lang=EN-US style='mso-ascii-font-family:Calibri;
mso-ascii-theme-font:minor-latin;mso-hansi-font-family:"MS Gothic";color:black'>Master
Request message format<o:p></o:p></span></p>

<p class=MsoNormal style='line-height:11.95pt;mso-line-height-rule:exactly;
text-autospace:none'><span lang=EN-US style='mso-ascii-font-family:Calibri;
mso-ascii-theme-font:minor-latin;mso-fareast-font-family:"MS Gothic";
mso-hansi-font-family:"MS Gothic";color:black'><o:p>&nbsp;</o:p></span></p>

Slave  Address | Function Code | Starting Address | Numbers of the register | Register-1 value | 。。。 | Register-N value | CRC16
-- | -- | -- | -- | -- | -- | -- | --
1 byte | 1 byte | 2 bytes | 2byte | 2bytes | N-2 | 2bytes | 2 bytes
Byte | Byte | Hi   Byte Lo   Byte | Hi   Byte Lo   Byte | Hi   Byte  Lo Byte | 。。。 | Hi Byte  Lo Byte | Hi Byte  Lo Byte

</body>

</html>

TLS1000, 2022-04-27:
Hi, no luck yet; I get timeout.
Below is the protocol description of the device in question. I'm a bit confused by the lack of a byte count as specified in the modbus.org standard. Could it be that this is not like 0x10 function code?
<html xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:w="urn:schemas-microsoft-com:office:word"
xmlns:m="http://schemas.microsoft.com/office/2004/12/omml"
xmlns="http://www.w3.org/TR/REC-html40">

<head>

<meta name=ProgId content=Word.Document>
<meta name=Generator content="Microsoft Word 15">
<meta name=Originator content="Microsoft Word 15">
<link rel=File-List
href="file:///C:/Users/wouter/AppData/Local/Packages/oice_16_974fa576_32c1d314_2aca/AC/Temp/msohtmlclip1/01/clip_filelist.xml">

<link rel=themeData
href="file:///C:/Users/wouter/AppData/Local/Packages/oice_16_974fa576_32c1d314_2aca/AC/Temp/msohtmlclip1/01/clip_themedata.thmx">
<link rel=colorSchemeMapping
href="file:///C:/Users/wouter/AppData/Local/Packages/oice_16_974fa576_32c1d314_2aca/AC/Temp/msohtmlclip1/01/clip_colorschememapping.xml">

<style>

</style>

</head>

<body lang=en-BE style='tab-interval:36.0pt;word-wrap:break-word'>

<p class=MsoNormal style='line-height:11.95pt;mso-line-height-rule:exactly;
text-autospace:none'><span lang=EN-US style='mso-ascii-font-family:Calibri;
mso-ascii-theme-font:minor-latin;mso-hansi-font-family:"MS Gothic";color:black'>Master
Request message format<o:p></o:p></span></p>

<p class=MsoNormal style='line-height:11.95pt;mso-line-height-rule:exactly;
text-autospace:none'><span lang=EN-US style='mso-ascii-font-family:Calibri;
mso-ascii-theme-font:minor-latin;mso-fareast-font-family:"MS Gothic";
mso-hansi-font-family:"MS Gothic";color:black'><o:p>&nbsp;</o:p></span></p>

Slave  Address | Function Code | Starting Address | Numbers of the register | Register-1 value | 。。。 | Register-N value | CRC16
-- | -- | -- | -- | -- | -- | -- | --
1 byte | 1 byte | 2 bytes | 2byte | 2bytes | N-2 | 2bytes | 2 bytes
Byte | Byte | Hi   Byte Lo   Byte | Hi   Byte Lo   Byte | Hi   Byte  Lo Byte | 。。。 | Hi Byte  Lo Byte | Hi Byte  Lo Byte

</body>

</html>

Miq1, 2022-04-27:
Yeah, looks like the message format is different. You only have the method to use
```
ModbusMessage::redefineFunctionCodeType(0x13, FCUSER_TYPE);
```
and set up the message all by yourself like
```
ModbusMessage m;
m.add(serverID, 0x13, address, numberOfRegisters);
m.add(registerValue1, registerValue2,...);
MB.addRequest(m);
```

TLS1000, 2022-04-27:
yess, succes! After messing around a bit to find out that the token needs to be passed as second argument in the addrequest when using self composed MBmessage, I was able to write to the register and readback the altered register content.

Miq1, 2022-04-27:
Glad to read it worked. Sorry, I wrote the post on my mobile and do not know all the calls by heart...

Miq1, 2022-05-22:
@TLS1000 Please note that the ``RTU-issues`` branch ist supposedly getting merged with ``master``. The way to redefine function codes has slightly changed, so you will have to modify your code once again. Sorry for that, I was not happy with the way I had implemented it before.

There is a table of allowed function code types:
```
enum FCType : uint8_t {
  FC01,              // Two uint16_t parameters (FC 0x01, 0x02, 0x03, 0x04, 0x05, 0x06)
  FC07,              // no additional parameter (FCs 0x07, 0x0b, 0x0c, 0x11)
  FC0F,              // two uint16_t parameters, a uint8_t length byte and a uint16_t* pointer to array of bytes (FC 0x0f)
  FC10,              // two uint16_t parameters, a uint8_t length byte and a uint8_t* pointer to array of words (FC 0x10)
  FC16,              // three uint16_t parameters (FC 0x16)
  FC18,              // one uint16_t parameter (FC 0x18)
  FCGENERIC,         // for FCs not yet explicitly coded (or too complex)
  FCUSER,            // No checks except the server ID
  FCILLEGAL,         // not allowed function codes
};
```
You will have to use one of these to redefine a function code. Redefinition is done by
```
const FCType redefineType(uint8_t functionCode, const FCType type);
```
If you will omit the ``type`` parameter, type ``FCUSER`` is used as a default.

This is possible only for function codes not covered by the Modbus standard, that initially will have a type of ``FCILLEGAL``. After the modification, the new function code may not change its type again.
The function call will return the now active type - if it is identical to that you attempted to set, you were successfull.

Any function code's type can be read with
```
const FCType getType(uint8_t functionCode);
```

Hope that will work for you.

TLS1000, 2022-05-23:
Hi Michael,

Sorry for my late response, I'm in the middle of updating PCB's in order to
get them to the factory....

'After the modification, the new function code may not change its type
again'.  I'm using multiple strange FC's in succession 0x42,0x49. You mean
I cannot do multiple 'redefinitions+addrequests'?

regards,
Wouter

Miq1, 2022-05-23:
No worries. That was supposed to explain that a function code like your 0x42 and 0x49 can be set to type FCUSER or whatever in setup() and will keep that type throughout the remainder of the program. You will not be able to some time later change again to, say: FCGENERIC, the code will keep the first type you assigned.

Miq1, 2022-05-28:
Closing now again, as the change is mainstream now.

closed 2022-05-28 by Miq1