eModbus/eModbus #185: Mark getter methods constant.

OekoSolveMG, 2022-04-28:
**Is your feature request related to a problem? Please describe.**
Getter eModbus methods can't be used on a constant ```ModbusMessage```, because the getter calls are not marked ```const``` even tough they could be.

**Describe the solution you'd like**
Simply make all getter methods ```const```, because we don't actually change anything in the class itself and this additionaly makes that clear to the user, when interfacing with the methods and using them.

**Describe alternatives you've considered**
There are no alternatives to just marking the methods ```const```.

**Example with the ```get()``` method**
```cplusplus
// Template function to extend getOne(index, A&) to get(index, A&, B&, C&, ...)
template <class T, class... Args>
typename std::enable_if<!std::is_pointer<T>::value, uint16_t>::type
get(uint16_t index, T& v, Args&... args) const {
  uint16_t pos = getOne(index, v);
  return get(pos, args...);
}

// getOne() - read a MSB-first value starting at byte index. Returns updated index
  template <typename T> uint16_t getOne(uint16_t index, T& retval) const {
    uint16_t sz = sizeof(retval);    // Size of value to be read

    retval = 0;                      // return value

    // Will it fit?
    if (index <= MM_data.size() - sz) {
      // Yes. Copy it MSB first
      while (sz) {
        sz--;
        retval <<= 8;
        retval |= MM_data[index++];
      }
    }
    return index;
  }

inline uint16_t get(uint16_t index) const { return index; }
```

Miq1, 2022-04-28:
Thanks for the suggestion.

One question, please: if the ModbusMessage is ``const`` anyway, why are you in need of ``get()`` at all?

OekoSolveMG, 2022-04-28:
Thanks for the fast response, I still want to be able to read back the values in the ModbusMessage even if it is constant.

But In my use case I don't need to change them, because the message has served it's purpose.

What I currently do is stash the request in a map with the token so I know which ```ModbusMessage``` request actually got a response in the callback, so I can then get the amount of registers and start register I asked for and use that to read out the response. Currently I can't make that request ```ModbusMessage``` ```const```, because If i want to ```get()``` from it once the callback has been called I can't do that.

Miq1, 2022-04-28:
I see. I am not sure, though, if making the ``get()`` methods ``const`` will break some existing code. I will have to check.

For the time being: can you have your stored messages non-const?

OekoSolveMG, 2022-04-28:
Yeah sure that's no problem that's why I made this issue a feature request, there isn't a high priority for this. Just thought that making getter methods ```const``` would somewhat make sense :D.

Miq1, 2022-04-28:
👍🏻 

I am currently involved in another (internally major) change, so it may take some time for me to try that.

OekoSolveMG, 2022-04-28:
No problem, thanks a lot for this amazing library :D and I hope the change goes smoothly.

Miq1, 2022-05-28:
Done it for the ``Tiny-server`` branch. If you like, have a try.

OekoSolveMG, 2022-06-14:
Sorry for the late reply tested the changes and it worked perfectly. There are some getter functions that haven't been marked const tough. Thoose would be:

```getFunctionCode()```
```getError()```
```getServerID()```

Would be nice if these could be marked const as well.

Miq1, 2022-06-14:
Sorry for having missed those. I will add them these days.

OekoSolveMG, 2022-06-15:
Works perfectly thanks a lot. Would be nice if it could be commited into the next release.

closed 2022-06-15 by OekoSolveMG

Miq1, 2022-06-15:
Will do. It may have to wait until I finished my "tiny server" addition, though... 😉