eModbus/eModbus #194: modbus RTU @9600 baudrate causes asyncTCP watchdog reset - multithread application

TLS1000, 2022-06-14:
#### Do not use to discuss topics! 

**Describe the bug**
With the latest release, the ESP crashes every 15s, watchdog timout in asyncTCP task. I have quite a number of running tasks. The old version of emodbus I use only runs in 1.0.6 build, and causes communication error in 2.0.X; BUT: it doesn't make the ESP crash. With the latest release of emodbus, the modbus communication with the host works,  but makes the ESP crash.

My suspicion is with the slow 9600BPS communication I'm using the emodbus library doesn't allow the OS to jump to another task. One would expect the timeout to occur on the modbus task, but it always happens at asyncTCP. 

**Expected behavior**
A clear and concise description of what you expected to happen.

**To Reproduce**
Steps to reproduce the behavior:

**Example code**
```
Put code here to reproduce the bug, if possible
```

**Additional context**
arduino core 2.0.X

TLS1000, 2022-06-14:
changing in RTUutils.ccp:
if (micros() - lastMicros < interval) delayMicroseconds(interval - (micros() - lastMicros));   I think this takes up too much processor time, since delaymicroseconds doens't allow the OS to jump to another task.

I tried in line 205:
if (micros() - lastMicros < interval) delay((interval - (micros() - lastMicros))/1000+5);    now I don't have ESP crash, but communication error.

TLS1000, 2022-06-14:
I also witnessed in case of non-response from the host, the library floods the host with repeated requests, at a much higher rate than modbus timeout setting. Is this intentionally?

Miq1, 2022-06-14:
I am no expert at all for the AsyncTCP stuff, so: @bertmelis, please have a look!

On the other hand the statement you think to have identified, when dissected means:
```
if (micros() - lastMicros < interval) // If the time between the last message sent or received and now is shorter than the interval,

delayMicroseconds(interval - (micros() - lastMicros)); // then delay for the rest of the interval
```
Interval at 9600 baud is around 3700µs, so to reach the WDT timeout of 2 seconds, it would take much longer.

There should be no overflow happening either, since the difference ``micros() - lastMicros`` will work even when we had an overflow of ``micros()`` in the meantime.

The communication errors will be caused by ``delay()`` and tasks switching, that is why we had that changed to ``delyMicroseconds()`` instead.

15 seconds is a timeout much longer than the WDT's of 2s, so again there must be something else involved. What you wrote normally would directly point to something stalling in a task using AsyncTCP.

The other observation (flooding a non-responsive host with requests) is nothing the library is even able to do. I would like to have a look at the application code when the E0 TIMEOUT error is returned - which is what the library will do.

TLS1000, 2022-06-14:
sorry, it's not asyncTCP that causes the timeout, it's just one of the tasks running during the timeout:

E (29637) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (29637) task_wdt:  - async_tcp (CPU 1)
E (29637) task_wdt: Tasks currently running:
E (29637) task_wdt: CPU 0: Modbus01RTU
E (29637) task_wdt: CPU 1: IDLE
E (29637) task_wdt: Aborting.

abort() was called at PC 0x400f1d01 on core 0

Backtrace:0x40083651:0x3ffbeb6c |<-CORRUPTED

the flooding of the host is observed when watching the RS485 TX-LED

Miq1, 2022-06-14:
As I read that, the async_tcp task on CPU1 was not feeding the watchdog while being active. This is regardless of which other tasks may be on  the system. The Modbus01RTU task was not involved at all.

As I wrote, the library itself simply is not able to do repeated requests, so whatever is happening there must be caused by another piece of code.

TLS1000, 2022-06-14:
the communication is working and I no longer have crashes when doing this:

if (micros() - lastMicros < interval) delay((interval - (micros() - lastMicros))/1000+5);

I don't know if it would work on other baudrates, but just going to the next milisecond still causes crashes

Miq1, 2022-06-14:
You are doing a 5ms delay here at least, which is too long to properly maintain the standard cycle, even more, if a task switch is happening. Besides that the ``delay()`` is notoriously inexact in the low ms region, so we must not use it here.

This rather looks like a race condition or deadlock to me, that does happen between the tasks. What does your asyncTCP task attempt to do when it hits the barrier?

bertmelis, 2022-06-14:
> This rather looks like a race condition or deadlock to me, that does happen between the tasks. What does your asyncTCP task attempt to do when it hits the barrier?

I second this. Could you share some code?

TLS1000, 2022-06-14:
As far as asyncTCP goes, we're using the async mqtt library, remotedebug; everything event driven. We have about 5 constant running tasks, and some dynamically created tasks for WIFI and MQTT-client reconnect. I'm not the coding expert in our project as you might have noticed :-) 

For sending data, does it matter that some extra time is added between frames, as long as one set of SEND-data fits the UART? 

Code above is running now for one hour steadily.

Miq1, 2022-06-14:
> Code above is running now for one hour steadily.

May be, but it feels wrong...

I take it you are reading Modbus values to provide these via MQTT? So how is the interfacing done between the Modbus request/response part and the MQTT delivery? This is highly susceptible to cause the problem you are experiencing.

Miq1, 2022-06-14:
> about the flooding: my modbus timeout was set at just 100ms, that's wat I was seeing.

Still, the library will send this one request only, wait for 100ms longest and then return an E0 TIMEOUT error. Nothing else.

TLS1000, 2022-06-14:
modbus runs in a seperate task, responses are written in global variables. MQTT runs in it's own task too. Wouldn't it be possible to use timer interrupt for exact timing in multi-task environment? blocking the task for a microsecondsdelay consumes huge resources. I could ask our coding guy to contribute

bertmelis, 2022-06-14:
So it might be the async mqtt client that blocks triggers the watchdog. Coincidentally, I worked on that as well.

Anyway, Arduino works by default on core 1 and WiFi -which is time critical, same as modbus RTU- runs on core 0. Core 1 has by default no watchdog enabled. I assume you have a specific reson to enable the watchdog.

I still would like to know what code you are running in the callbacks.

Miq1, 2022-06-14:
> blocking the task for a microsecondsdelay consumes huge resources. I could ask our coding guy to contribute

Nah! We are talking about below 4ms maximum - which of your tasks is as time-critical to get into trobles if running on the same core?

Yeah, it may be worthwhile to have the specialist involved here.

bertmelis, 2022-06-14:
I have to correct myself: AsyncTCP adds a watchdog to its task.  But then again, if I recall correctly, the watchdog is 300msec.

Miq1, 2022-06-14:
If it strikes every 15s, there must be a coincidence in the intervals used in the different tasks. The Modbus server is not expected to respond at the exact same times, so the intersection must be internally to the code on the ESP. Which timeouts or loop counts in your code have a common multiple of 15s? What frequency are you using to issue requests?

TLS1000, 2022-06-14:
I'm sayin' 15s as a guess, I don't think it's constant. All the tasks run (the ones that are in constant use) in about 2-3seconds. mqtt,wifi,and emodbus we use the event-handlers. For the modbus we send out about 10 read/writes every 2s. At the end we add a variable delay to establish a stable loop-time. The arduino 'loop' is empty

Miq1, 2022-06-14:
If you cannot narrow it down else, I would suggest logging all the relevant activity spots in your code to see if two or more are crossing shortly before the system crashes.

TLS1000, 2022-06-14:
> I am no expert at all for the AsyncTCP stuff, so: @bertmelis, please have a look!
>
> On the other hand the statement you think to have identified, when dissected means:
>
> ```
> if (micros() - lastMicros [...]

The sole purpose of the microseconddelay is to slow down both the sending and receiving? And if we pauzed the RTUutils as a task until the interval has passed? Would that be a solution?

TLS1000, 2022-06-14:
yes, that's what I meant, just adding a gap. With 'AT LEAST 3.5 character lengths', can't we just choose to be on the safe side and add a DELAY based on baudrate(what you're doing), plus some extra to let the OS switch tasks during that interval? I realize it's sub-optimal in terms of speed, but on the other side 9600bps is lightning fast to begin with.

Miq1, 2022-06-14:
The gap is mandatory for anyone on the bus, regardless if the client would be able to pause longer at will. The same code is used in the library for client and server and would have to be duplicated to allow for different gap times anyway.

I still am convinced you are poking at the false end to find a cause. If you have a task malfunctioning because of a mere <4ms pause you should consider placing that task on the other core instead.

TLS1000, 2022-06-15:
yes indeed, 4ms seems very low to cause problems to other tasks. BUT, this firmware has been running steadily for several months on numerous devices, the only thing I changed was upgrading the core build from 1.0.6 to 2.0.3, and going to the latest release of emodbus because the old one I had didn't work anymore on this latest build.   I guess time will tell. Anyway, thank you for the support Miq1 and bertmelis!

Miq1, 2022-06-19:
Mhm, not finally solved IMHO. Closing this anyway as it seems to be a local problem only.

closed 2022-06-19 by Miq1