eModbus/eModbus #198: Split ModbusMessages

OekoSolveMG, 2022-06-23:
**Describe the bug**
RTUUtils causes certain messages to be cut into smaller parts and it seems to miss some bytes, causing E2 - CRC Check errors and E5 - Packet length error.

**For example:**

```cpp
  | 0000: 01 03 3E 00 60 00 00 00  00 00 00 00 00 00 00 02  |..>.`...........|
  | 0010: 79 01 02 00 7A 00 01 00  00 02 7B 00 01 00 00 00  |y...z.....{.....|
  | 0020: E2 00 00 02 B9 00 00 0B  D6 27 28 02 79 4C 9F 00  |.........'(.yL..|
  | 0030: 00 00 08 00 50 0F 71 0C  A8 00 01 00 01 00 64 00  |....P.q.......d.|
  | 0040: 01 
```

Gets cut into firstly this which is received as a full message even tough it is only a part of it.

```cpp
01 03 3E 00 71 02 7B 00 04 01 3D 00 00 00 00 02 79 01 
```

Following this part, which is ignored.

```cpp
02 00 7A 00 01 00  00 02 7B 00 01 00 00 00 E2 00 00 02 B9 00 00 0B  D6 27 28 02 79
```

and then this part, which probably gets interpreted as the response for the 2nd request.

```cpp
F9 53 06 00 00 08 00 50 0F 71 0C A8 00 01 00 01 00 64 00 03 A7 AD
```

**Log calls**

Successfull message:
```cpp
[N] Received packet: @3FFDE3C4/37:
  | 0000: 01 03 22 00 00 00 00 00  00 00 00 00 00 00 00 00  |..".............|
  | 0010: 00 00 01 0E 5C 00 00 78  3C 00 00 96 5C 00 96 AD  |....\..x<...\...|
  | 0020: 3C 00 21 E2 17                                    |<.!..           |
[N] Received packet: @3FFDE5CC/65:
  | 0000: 01 03 3E 00 60 00 00 00  00 00 00 00 00 00 00 02  |..>.`...........|
  | 0010: 79 01 02 00 7A 00 01 00  00 02 7B 00 01 00 00 00  |y...z.....{.....|
  | 0020: E2 00 00 02 B9 00 00 0B  D6 27 28 02 79 4C 9F 00  |.........'(.yL..|
  | 0030: 00 00 08 00 50 0F 71 0C  A8 00 01 00 01 00 64 00  |....P.q.......d.|
  | 0040: 01    
```

Error message:
```cpp
[3082][MB][E] (lib/modbus/ModbusMaster.cpp:304) - Error: (E2) - (CRC check error).
[N] Received packet: @3FFDE588/1:
  | 0000: E2  
```

Error data:
```cpp
[9261][E-MB][DUMP] (.pio/libdeps/m5stack-core-esp32/eModbus/src/RTUutils.cpp:497) - Size: [7], Content: [01 03 3E 00 71 02 7B ]
[9266][MB][E] (lib/modbus/ModbusMaster.cpp:304) - Error: (E2) - (CRC check error).

[9291][E-MB][DUMP] (.pio/libdeps/m5stack-core-esp32/eModbus/src/RTUutils.cpp:497) - Size: [4], Content: [71 D0 0D A8 ]
[9292][MB][E] (lib/modbus/ModbusMaster.cpp:304) - Error: (E2) - (CRC check error).

[34242][E-MB][DUMP] (.pio/libdeps/m5stack-core-esp32/eModbus/src/RTUutils.cpp:497) - Size: [18], Content: [01 03 3E 00 71 02 7B 00 04 01 3D 00 00 00 00 02 79 01 ]
[34247][MB][E] (lib/modbus/ModbusMaster.cpp:304) - Error: (E2) - (CRC check error).

[34292][E-MB][DUMP] (.pio/libdeps/m5stack-core-esp32/eModbus/src/RTUutils.cpp:497) - Size: [22], Content: [F9 53 06 00 00 08 00 50 0F 71 0C A8 00 01 00 01 00 64 00 03 A7 AD ]
[34297][MB][E] (lib/modbus/ModbusMaster.cpp:304) - Error: (E2) - (CRC check error).

[59244][E-MB][DUMP] (.pio/libdeps/m5stack-core-esp32/eModbus/src/RTUutils.cpp:497) - Size: [19], Content: [01 03 3E 00 71 02 7B 00 04 01 3C 00 00 00 00 02 79 01 00 ]
[59249][MB][E] (lib/modbus/ModbusMaster.cpp:304) - Error: (E2) - (CRC check error).

[59293][E-MB][DUMP] (.pio/libdeps/m5stack-core-esp32/eModbus/src/RTUutils.cpp:497) - Size: [20], Content: [06 00 00 08 00 50 0F 71 0C A8 00 01 00 01 00 64 00 03 9B AF ]
[59298][MB][E] (lib/modbus/ModbusMaster.cpp:304) - Error: (E2) - (CRC check error).

[144232][E-MB][DUMP] (.pio/libdeps/m5stack-core-esp32/eModbus/src/RTUutils.cpp:497) - Size: [8], Content: [01 03 3E 00 71 02 7B 00 ]
[144234][MB][E] (lib/modbus/ModbusMaster.cpp:304) - Error: (E2) - (CRC check error).

[144292][E-MB][DUMP] (.pio/libdeps/m5stack-core-esp32/eModbus/src/RTUutils.cpp:497) - Size: [32], Content: [02 B9 00 00 0B D6 27 28 02 79 4C 9F 00 00 00 08 00 50 0F 71 0C A8 00 01 00 01 00 64 00 
03 6D BA ]
[144300][MB][E] (lib/modbus/ModbusMaster.cpp:304) - Error: (E2) - (CRC check error).

[169242][E-MB][DUMP] (.pio/libdeps/m5stack-core-esp32/eModbus/src/RTUutils.cpp:497) - Size: [18], Content: [01 03 3E 00 71 02 7C 00 04 01 3C 00 00 00 00 02 79 01 ]
[169247][MB][E] (lib/modbus/ModbusMaster.cpp:304) - Error: (E2) - (CRC check error).

[169292][E-MB][DUMP] (.pio/libdeps/m5stack-core-esp32/eModbus/src/RTUutils.cpp:497) - Size: [23], Content: [FF 4C 9F 00 00 00 08 00 50 0F 71 0C A8 00 01 00 01 00 64 00 03 E9 E8 ]
[169297][MB][E] (lib/modbus/ModbusMaster.cpp:304) - Error: (E2) - (CRC check error).

[194232][E-MB][DUMP] (.pio/libdeps/m5stack-core-esp32/eModbus/src/RTUutils.cpp:497) - Size: [8], Content: [01 03 3E 00 71 02 7C 00 ]
[194234][MB][E] (lib/modbus/ModbusMaster.cpp:304) - Error: (E2) - (CRC check error).
[194292][E-MB][DUMP] (.pio/libdeps/m5stack-core-esp32/eModbus/src/RTUutils.cpp:497) - Size: [32], Content: [02 B9 00 00 0B D6 27 28 02 79 4C 9F 00 00 00 08 00 50 0F 71 0C A8 00 01 00 01 00 64 00 
03 B1 3E ]
[194300][MB][E] (lib/modbus/ModbusMaster.cpp:304) - Error: (E2) - (CRC check error).
[219233][E-MB][DUMP] (.pio/libdeps/m5stack-core-esp32/eModbus/src/RTUutils.cpp:497) - Size: [9], Content: [01 03 3E 00 71 02 7C 00 04 ]
[219236][MB][E] (lib/modbus/ModbusMaster.cpp:304) - Error: (E2) - (CRC check error).

[219292][E-MB][DUMP] (.pio/libdeps/m5stack-core-esp32/eModbus/src/RTUutils.cpp:497) - Size: [32], Content: [F3 28 03 00 0B D6 27 28 02 79 4C 9F 00 00 00 08 00 50 0F 71 0C A8 00 01 00 01 00 64 00 
03 27 7C ]
[219300][MB][E] (lib/modbus/ModbusMaster.cpp:304) - Error: (E2) - (CRC check error).

[244241][E-MB][DUMP] (.pio/libdeps/m5stack-core-esp32/eModbus/src/RTUutils.cpp:497) - Size: [18], Content: [01 03 3E 00 71 02 7C 00 04 01 3C 00 00 00 00 02 79 01 ]
[244246][MB][E] (lib/modbus/ModbusMaster.cpp:304) - Error: (E2) - (CRC check error).

[244292][E-MB][DUMP] (.pio/libdeps/m5stack-core-esp32/eModbus/src/RTUutils.cpp:497) - Size: [23], Content: [02 5E EA 06 00 00 08 00 50 0F 71 0C A8 00 01 00 01 00 64 00 03 2B A9 ]
[244300][MB][E] (lib/modbus/ModbusMaster.cpp:304) - Error: (E2) - (CRC check error).

[269243][E-MB][DUMP] (.pio/libdeps/m5stack-core-esp32/eModbus/src/RTUutils.cpp:497) - Size: [18], Content: [01 03 3E 00 71 02 7B 00 04 01 3C 00 00 00 00 02 79 01 ]
[269248][MB][E] (lib/modbus/ModbusMaster.cpp:304) - Error: (E2) - (CRC check error).

[269292][E-MB][DUMP] (.pio/libdeps/m5stack-core-esp32/eModbus/src/RTUutils.cpp:497) - Size: [23], Content: [F9 4C 9F 00 00 00 08 00 50 0F 71 0C A8 00 01 00 01 00 64 00 03 33 EA ]
[269298][MB][E] (lib/modbus/ModbusMaster.cpp:304) - Error: (E2) - (CRC check error).

[294243][E-MB][DUMP] (.pio/libdeps/m5stack-core-esp32/eModbus/src/RTUutils.cpp:497) - Size: [18], Content: [01 03 3E 00 71 02 7C 00 04 01 3C 00 00 00 00 02 79 01 ]
[294248][MB][E] (lib/modbus/ModbusMaster.cpp:304) - Error: (E2) - (CRC check error).

[294293][E-MB][DUMP] (.pio/libdeps/m5stack-core-esp32/eModbus/src/RTUutils.cpp:497) - Size: [22], Content: [01 EA 06 00 00 08 00 50 0F 71 0C A8 00 01 00 01 00 64 00 03 49 01 ]
[294299][MB][E] (lib/modbus/ModbusMaster.cpp:304) - Error: (E2) - (CRC check error).

[319242][E-MB][DUMP] (.pio/libdeps/m5stack-core-esp32/eModbus/src/RTUutils.cpp:497) - Size: [18], Content: [01 03 3E 00 71 02 7C 00 04 01 3C 00 00 00 00 02 79 00 ]
[319247][MB][E] (lib/modbus/ModbusMaster.cpp:304) - Error: (E2) - (CRC check error).

[319292][E-MB][DUMP] (.pio/libdeps/m5stack-core-esp32/eModbus/src/RTUutils.cpp:497) - Size: [23], Content: [FF 4C 9F 00 00 00 08 00 50 0F 71 0C A8 00 01 00 01 00 64 00 03 8C 91 ]
[319297][MB][E] (lib/modbus/ModbusMaster.cpp:304) - Error: (E2) - (CRC check error).

[339351][E-MB][DUMP] (.pio/libdeps/m5stack-core-esp32/eModbus/src/RTUutils.cpp:497) - Size: [3], Content: [D3 12 FF ]
[339353][MB][E] (lib/modbus/ModbusMaster.cpp:304) - Error: (E5) - (Packet length error).

[344233][E-MB][DUMP] (.pio/libdeps/m5stack-core-esp32/eModbus/src/RTUutils.cpp:497) - Size: [7], Content: [01 03 3E 00 71 02 7C ]
[344235][MB][E] (lib/modbus/ModbusMaster.cpp:304) - Error: (E2) - (CRC check error).

[344294][E-MB][DUMP] (.pio/libdeps/m5stack-core-esp32/eModbus/src/RTUutils.cpp:497) - Size: [35], Content: [7A 00 00 02 B9 00 00 0B D6 27 28 02 79 4C 9F 00 00 00 08 00 50 0F 71 0C A8 00 01 00 01 
00 64 00 03 A2 09 ]
[344303][MB][E] (lib/modbus/ModbusMaster.cpp:304) - Error: (E2) - (CRC check error).

[364269][E-MB][DUMP] (.pio/libdeps/m5stack-core-esp32/eModbus/src/RTUutils.cpp:497) - Size: [42], Content: [01 03 3E 00 71 02 7B 00 04 01 3D 00 00 00 00 02 79 01 01 00 78 FF FF 00 00 02 72 00 01 
00 01 01 84 00 00 02 B9 00 00 0B D6 27 ]
[364281][MB][E] (lib/modbus/ModbusMaster.cpp:304) - Error: (E2) - (CRC check error).

[389244][E-MB][DUMP] (.pio/libdeps/m5stack-core-esp32/eModbus/src/RTUutils.cpp:497) - Size: [19], Content: [01 03 3E 00 71 02 7C 00 04 01 3C 00 00 00 00 02 79 01 00 ]
[389249][MB][E] (lib/modbus/ModbusMaster.cpp:304) - Error: (E2) - (CRC check error).

[389295][E-MB][DUMP] (.pio/libdeps/m5stack-core-esp32/eModbus/src/RTUutils.cpp:497) - Size: [20], Content: [06 00 00 08 00 50 0F 71 0C A8 00 01 00 01 00 64 00 03 D2 F8 ]
[389300][MB][E] (lib/modbus/ModbusMaster.cpp:304) - Error: (E2) - (CRC check error).
```

OekoSolveMG, 2022-06-23:
I've attached the LogicAnaylzer to see if I can see anything on the Hardware level.

As far as I could debug I can see 2 errors. One of which occurs randomly. That would be that we don't toggle the pin to LOW fast enough and cause as to loose a lot of the message.

[image]

And the second one is sistematic and always seems to occurs on the 6th read and is toggling the pin to HIGH and writing a request even tough the receive hasn't been finished.

[image]

[image]

OekoSolveMG, 2022-06-23:
I've added the 10 minutes I recorded as a .sr file in the .zip.

[Session.zip](https://github.com/eModbus/eModbus/files/8968013/Session.zip)

Really confused why it would be able to request, when the first receive hasn't been processed yet tough.

OekoSolveMG, 2022-06-23:
Baud rate of 9600

Miq1, 2022-06-23:
There must be something more to that. This send:
[image]

must not happen at all due to the intended logics. Only if the gap time has passed by all means the send method shall send:
```
// RTU mode
    uint16_t crc16 = calcCRC(data, len);

    // Respect interval - we must not toggle rtsPin before
    if (micros() - lastMicros < interval) delayMicroseconds(interval - (micros() - lastMicros));

    // Toggle rtsPin, if necessary
    rts(HIGH);
```

You did not mess with the interval etc., or did you? 😉 

Can you print out the three values of ``micros()``, ``interval`` and ``lastMicros`` here (``RTUutils.cpp, send(), in RTU mode``)

OekoSolveMG, 2022-06-23:
> You did not mess with the interval etc., or did you? 😉

No but I can add the log calls as well. Have to produce the errors again, because it doesn't happen all the time, but if it does it happens like shown above.

Miq1, 2022-06-23:
Another oddity: even if the logic has determined the response was complete and does start another send(), the first it does is reading everything left in the UART buffer. So the remainder of the data can not continue being received after the send is over, as the LA file is suggesting?!?

OekoSolveMG, 2022-06-23:
Same behaviour as before the 6th request always fails with reading 2 blocks each.

```cpp
// Block 1 / Read 1
micros(): 75643485
interval: 3645
lastMicros: 55725944
// Block 2 / Read 1
micros(): 75729346
interval: 3645
lastMicros: 75724811
// Block 1 / Read 2
micros(): 76641774
interval: 3645
lastMicros: 75782499
// Block 2 / Read 2
micros(): 76727874
interval: 3645
lastMicros: 76723125
// Block 1 / Read 3
micros(): 77641688
interval: 3645
lastMicros: 76780104
// Block 2 / Read 3
micros(): 77727897
interval: 3645
lastMicros: 77723084
// Block 1 / Read 4
micros(): 78641684
interval: 3645
lastMicros: 77780077
// Block 2 / Read 4
micros(): 78727675
interval: 3645
lastMicros: 78723041
// Block 1 / Read 5
micros(): 79641708
interval: 3645
lastMicros: 78780033
// Block 2 / Read 5
micros(): 79727504
interval: 3645
lastMicros: 79722996
// Block 1 / Read 6
micros(): 80644552
interval: 3645
lastMicros: 79779568
// Block 2 / Read 6
[80661][MB][E] (lib/modbus/ModbusMaster.cpp:304) - Error: (E2) - (CRC check error).
micros(): 80672250
interval: 3645
lastMicros: 80663446
```
```cpp
micros(): 100643717
interval: 3645
lastMicros: 80717627
micros(): 100729889
interval: 3645
lastMicros: 100725189
micros(): 101641323
interval: 3645
lastMicros: 100782607
micros(): 101727246
interval: 3645
lastMicros: 101722590
micros(): 102641378
interval: 3645
lastMicros: 101779300
micros(): 102727210
interval: 3645
lastMicros: 102722631
micros(): 103641617
interval: 3645
lastMicros: 102779282
micros(): 103727784
interval: 3645
lastMicros: 103723063
micros(): 104642628
interval: 3645
lastMicros: 103780048
micros(): 104728556
interval: 3645
lastMicros: 104724019
micros(): 105644774
interval: 3645
lastMicros: 104780616
[105664][MB][E] (lib/modbus/ModbusMaster.cpp:304) - Error: (E2) - (CRC check error).
micros(): 105675511
interval: 3645
lastMicros: 105663557
```
```cpp
micros(): 150644341
interval: 3645
lastMicros: 130692701
micros(): 150730374
interval: 3645
lastMicros: 150725702
micros(): 151641517
interval: 3645
lastMicros: 150782654
micros(): 151727626
interval: 3645
lastMicros: 151722804
micros(): 152643295
interval: 3645
lastMicros: 151779738
micros(): 152728961
interval: 3645
lastMicros: 152724578
micros(): 153641664
interval: 3645
lastMicros: 152781130
micros(): 153727739
interval: 3645
lastMicros: 153723106
micros(): 154641625
interval: 3645
lastMicros: 153780085
micros(): 154727707
interval: 3645
lastMicros: 154723056
micros(): 155644655
interval: 3645
lastMicros: 154780048
[155660][MB][E] (lib/modbus/ModbusMaster.cpp:304) - Error: (E2) - (CRC check error).
micros(): 155671828
interval: 3645
lastMicros: 155663499
```

OekoSolveMG, 2022-06-23:
I'm using the TinyServer branch currently just so you know. Pretty sure there aren't any breaking changes tough.

Miq1, 2022-06-23:
I did not find a missing update of ``lastMicros`` unfortunately. It is just a single task in the client doing the RTU handling, so we will not have any concurrency issue here either.
The DE/RE line is pulled in the middle of the receive, with no gap at all, although the log tells it differently:
[image]

Can you spot the last byte that is reported as received in comparison to those on the LA dump? Could it be that the gap was detected a few ms earlier somehow?

Miq1, 2022-06-23:
Since we can not use log statements in the receive loop without affecting the timing, can you alternatively do a ``digitalWrite(pin, HIGH);digitalWrite(pin, LOW);`` after a byte has been read just before this line
```
          // Do we need to skip it, if it is zero?
```
in ``RTUutils.cpp`` and connect that pin to another LA channel?
Then we may get an impression how the Serial read relates to the UART traffic.

OekoSolveMG, 2022-06-24:
> Can you spot the last byte that is reported as received in comparison to those on the LA dump? Could it be that the gap was detected a few ms earlier somehow?

```cpp
 [01 03 3E 00 71 02 7B ]
```

[image]

```cpp
Content: [01 FF 40 0C 00 02 B9 00 00 0B D6 27 28 02 79 4C 9F 00 00 00 08 00 50 0F 71 0C A8 00 01 00 01 00 64 00 03 C3 EF ]
```

[image]

[image]

Miq1, 2022-06-24:
So the gap is detected several ms before actually. 
This hints to the Serial FIFO handling again - the ``read()`` does not see the bytes arriving at the pins in time. Can you add the pin toggle that I suggested in my last post? If I am right, the toggle will fire with the last byte detected, before starting again after the misplaced ``send()``.

OekoSolveMG, 2022-06-24:
Will do, need a second to sotter the pin tough. Will come back once I collected the data.

Miq1, 2022-06-24:
Unfortunately I will have to leave in about an hour for most of today, so whatever you can provide to help me analyze over the weekend is welcome.

OekoSolveMG, 2022-06-24:
> Connect that pin to another LA channel?

**RTUutils.cpp at both line 308 / 265**

```cpp
digitalWrite(5, HIGH);
// Buffer has space left - try to read another byte
b = serial.read();
digitalWrite(5, LOW);
```

The reads seem to stop in the middle of the first message the same as I can see in the log calls above.

[image]

[image]

Attached you can find the sigrok file (D0 is the CS toggle, D1 is RX, D2 is the pin that I toggle each read and D3 is the TX)
[SessionWithPinToggle.zip](https://github.com/eModbus/eModbus/files/8974467/SessionWithPinToggle.zip)

bertmelis, 2022-06-24:
I find it utterly strange that D2 **stays** high for more than a moment. You made it HIGH, read single byte, LOW. This means there must be something going on in the `serial.read()` function.

Miq1, 2022-06-24:
That was my bet from start.

bertmelis, 2022-06-24:
It may not be in the read itself but somewhere else though. Read is guarded by a mutex.

Miq1, 2022-06-26:
[image]
Looks like the ``read()`` is doing something under the hood indeed.

bertmelis, 2022-06-26:
It also happens when reading successfully. But as long as the read operation time stays below the gap it happily moves on.

Miq1, 2022-06-26:
I had a look at the RMT in the meantime. It would be usable up to about 250000 baud, but it requires substantial pre- and postprocessing and a set of large ring buffers. A kind of last resort I would say and only for the ESP32.

Miq1, 2022-06-27:
Retrospectively I now can see why I had to put seemingly arbitrary delays into my test sketch to get the >100 RTU tests going in a row. Since the effect is worsening with lower baud rates, the interfering code (if there is some) must run in timed cycles, not based on other events. Plus it must have a very high priority to be run while the serial read is on.
I found some evidence this has been noticed before by others as well, but no one seems to have cared enough to get it finally analyzed or even resolved. All were related to the arduino-esp32 environment, by the way.

OekoSolveMG, 2022-06-27:
> Retrospectively I now can see why I had to put seemingly arbitrary delays into my test sketch to get the >100 RTU tests going in a row.

That would also explain why enabling logging decreases the occurrences of the error.

Miq1, 2022-06-27:
> Would it be possible to write an issue on the main espressif32 branch?

I must confess I have had not the best of experiences doing so in the past. In most cases one is ignored, sometimes after having been taught a lesson that did not relate at all to the issue described.

OekoSolveMG, 2022-06-27:
> @bertmelis A wild idea: we should be able to use the IDF API to read out the configured values with the get methods, then delete the driver the arduino-esp32 core had installed for the selected UART. Following we can install our own driver with a ring buffer large enough to hold a complete RTU message and a RX timeout threshold of 3.5 characters, RX interrupts enabled.

At that point isn't it simpler to just use the esp-idf base instead of arduino espressif32.

OekoSolveMG, 2022-06-27:
I'll just try to create an issue for now and I'll see where it leads. The worst that can happen is that nobody cares.

bertmelis, 2022-06-27:
I believe the Arduino is at this moment nothing more than a wrapper around IDF. Buffer size and timeout is already settable. I'll try some options but we're leaving on a 14 day holiday and I've got a load of other tasks. I may be able to squeeze this one in in the coming days.

Miq1, 2022-06-27:
> I'll just try to create an issue for now and I'll see where it leads. The worst that can happen is that nobody cares. Might you know any good title? Because the issue seems to be the read If I understand it correctly?

"No real-time serial.read() possible " came to my mind.

OekoSolveMG, 2022-06-28:
Any idea on what sketch I could provide?

OekoSolveMG, 2022-06-28:
Would be nice if you could look at the issue request if you think I should add anything else.

Miq1, 2022-06-28:
I am busy today in the morning unfortunately and away from my PC.
I would suggest to write a sketch with two tasks, one writing to a serial chunks of >200 bytes, the other trying to read the data byte-wise from another serial and connect these crosswise. The reading serial will have to be set to a 1-byte threshold as is done in UARTinit().
Baud rate low, like 2400.

Miq1, 2022-06-28:
Very good! How does it behave then?

bertmelis, 2022-06-28:
Well, it generates the callback when the whole message is received and only then. No onError callback is called because the application here doesn't generate an error. Timeout calculation is not on target though. Might also be the way it is measured (toggling pin in the callback, which is slow in itself). But I think it can be used in our context.

bertmelis, 2022-06-28:
Of course, I moved to pin toggle to the very beginning on the onReceive callback and now it seems spot on 4 chars.

Miq1, 2022-06-28:
👍🏻 
I will try to integrate that into RTUutils today.

Miq1, 2022-06-28:
> The 4t char timeout is a pity - if we are a server and the client on the
> bus is using exactly 3.5t between the last response and another request,
> the server would not detect that. If we took 3t instead, we might be
> missing slow servers instead.
>
> Yes, float values are not supported. I'm not sure if we can call one of
> the ESP-IDF methods directly to finetune. I don't see a possibility yet.

I am completely cut off the cable network again, so I doubt I will be able
to do anything today.
Whenever you hear Vodafone, simply run... :angry

Miq1, 2022-06-28:
I checked the ESP-IDF calls the arduino-esp32 core is using: the RX timeout threshold is a ``uint8_t`` there as well, so no chance to have a finer granularity. But you are right, Bert, a value of 3 should do.
Hopefully I get back into the net tomorrow or so, so I can try out the revised receive.

Miq1, 2022-07-01:
Dang. The ``onReceive()`` callback does not know which serial it is attached to and has no way to find out. Or has it?

Problem is, if we have more than one RTU client or server in one sketch, these will be using different serials and have different target buffers to fill with the (mostly received) data. 

My current approach is to have instance-specific callbacks, but that is less efficient, because the code has to be duplicated for client and server.

Miq1, 2022-07-01:
Note: we are creating a fixed dependency from Platformio platform version >=5.0.0 here. Older platform versions are not compiling any more.

Miq1, 2022-07-01:
Next obstacle: in ASCII mode we can have messages up to 513 bytes long and have a much longer timeout that is not to be covered by the ``RxTimeout``. So classic RTU mode and ASCII mode are no more switchable at will, but will have to be defined at the time the UART is initialized.

bertmelis, 2022-07-01:
Is there ever more then 1 leading zero? I never investigated this but I'd expect only one. If there are more, it's fishy at least.

Miq1, 2022-07-01:
> Is there ever more then 1 leading zero? I never investigated this but I'd expect only one. If there are more, it's fishy at least.

AFAIAC even the first is fishy 🤣 

I have stopped for today and will move on tomorrow. I do not like the different handling yet, but that may change over time...

(Aren't you supposed to be vacationing? 😉 )

bertmelis, 2022-07-01:
Leaving on Sunday but preparing... And it's 35° here in Warsaw. Too hot to think.

Miq1, 2022-07-01:
> Is there ever more then 1 leading zero? I never investigated this but I'd expect only one. If there are more, it's fishy at least.

Worse... Since we are supporting broadcasts now, the ``skipLeadingZero`` handling will effectively prevent broadcasts from being received at a server using it. That has no dependencies from the new serial handling, but already is so with the existing code. 

One more argument to insist to find the root cause for the zeroes and not filtering them.

bertmelis, 2022-07-01:
> I still believe the zero is generated by the TX->RX switch.

It is. But if we're toggling right after sending, and the server/client respects the 3.5 chars it will come in as a separate message because of the 3 char timeout.

Miq1, 2022-07-02:
ASCII mode really is a show stopper. The larger buffer required is a minor thing, but the allowable timout of 1s _minimum_ is not. The RxTimeout value is of type ``uint8_t``, so we have 255 characters passing for a timeout _maximum_. With 10 bits per character we already will break that limit with 4800 baud.

I am afraid we will have to keep track of the timeouts in ASCII mode by regularly checking the fill level of the RX buffer. If it did not change any more for the timeout time, we can state a timeout. At the same time we must actively read and check the read characters to detect the CRLF message trailer in time to not waste time by waiting for the timeout. The pattern-matching interrupt the IDF provides is not taken over into arduino-esp32, so we can not make use of that. 😠

bertmelis, 2022-07-02:
This might come to the rescue, but it isn't merged and it's certainly not released yet: https://github.com/espressif/arduino-esp32/pull/6930

Miq1, 2022-07-02:
Sounds good indeed - if only this really will trigger the interrupt in real time. We hoped that to happen bvefore with our patch of the UART threshold and it did not work for reasons unknown, as we all know.

Our two options are:
1. continue doing the substantial modifications I am working on
2. hold it and wait for the PR to be integrated

Option 1 is much more effort than I had thought initially and will break our interface at least internally. Option 2 is undetermined when to be released and will leave the current issue unresolved, but will keep the interface as before.

What would be your advice?

Miq1, 2022-09-16:
@OekoSolveMG In case you are still interested, the long awaited PR was merged into the arduino-esp32 core yesterday that now opens the configuration options for the UARTs to legally set the FIFO full threshold.

I tested it and found it working for me. I did not have the exact same effects that you experienced, though, so you better should check your code with it yourself.

In the library's ``RTUutils.cpp`` you will need to replace the ``UARTinit()`` function with the following:
```
int RTUutils::UARTinit(HardwareSerial& serial, int thresholdBytes) {
  int rc = 0;
#if NEED_UART_PATCH
  serial.setRxFIFOFull(thresholdBytes);
#endif
  // Return the previous value in case someone likes to see it.
  return rc;
}
```

To use the recent core you will want to add the line 
```
platform_packages =
  framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32
```
to the ``[env]`` section of your ``platformio.ini`` file.

Miq1, 2022-09-23:
@OekoSolveMG @TLS1000
Guys? No reaction yet?

TLS1000, 2022-09-23:
Hi Michael, in our project where we run a dedicated task for the modbus communication, we currently use a simple modbus library from adafruit. It doesn't matter if it's blocking, it has it's own task. For the moment this works for us. But thank you for reaching out.

Miq1, 2022-09-23:
@TLS1000 Fine with me - thank you for the response!

(I had loved to learn if the new core UART handling will fix your original issue, but never mind. 😉 )

zivillian, 2022-09-26:
I recompiled my gateway with the mentioned patch, but I'm still seeing an error rate of ~0.5 - 1.5%.

Miq1, 2022-09-26:
> I recompiled my gateway with the mentioned patch, but I'm still seeing an error rate of ~0.5 - 1.5%.

I suppose you took the most recent core from Github? 0.5 to 1.5 is not that much for RTU in fact. Can you sort out other potential influences? Do you have some figures on the error codes encountered (the UART issues mostly are producing E0 and E5 errors)?

zivillian, 2022-09-26:
First some facts:

platformio.ini:
```ini
[env]
platform = espressif32
framework = arduino
lib_deps =
    https://github.com/tzapu/WiFiManager.git
    https://github.com/me-no-dev/ESPAsyncWebServer.git
    eModbus
build_flags = -DLOG_LEVEL=LOG_LEVEL_DEBUG
monitor_speed = 115200
platform_packages =
  framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32
  
[env:esp32dev]
board = esp32dev
```

I currently have two modbus devices on different gateways. First is an Eastron SDM72DM-v2 meter using 38400,8N1 polled by [mbmd](https://github.com/volkszaehler/mbmd), the other is a Heidelberg Wallbox using 19200,8E1 polled by [evcc](https://github.com/evcc-io/evcc). The hardware is a custom layout, but the same for both.

The error rate is a rough guess, based on the internal counters of eModbus.

My evcc log shows only the error codes `229 (0xE5)` (464 times) and `226 (0xE2)` (529 times). My mbmd log also shows only those to error codes - `0xE5` 513 times and `0xE2` 3436 times.

My gateway has a web debug interface, where I can select the slave id, function, register and register count. After submitting it enables and redirects the eModbus debug log to the http response stream, sends the request, waits for the response and disables the debug log.
From there I can send arbitrary modbus requests and by testing a few things I can tell, that the error rate correlates to the length of the response. Also the interval between the requests seems to play a role.

If I send 10 requests to query 20 registers and wait 2 seconds between each request, they all work. If I don't wait, only 50% succeed, while the other half reports `0xE2` and from the debug output I can tell that the response is truncated.

If I reduce the number of registers to one, two or even ten, of all of them succeed.

Miq1, 2022-10-04:
@zivillian Any new insights or data?

zivillian, 2022-10-05:
@Miq1 it took me a while to setup my hardware, but I've successfully captured a broken frame:
```
[D] 331590| ModbusClientRTU.cpp  [ 222] addToQueue: RC=01
[D] 331591| ModbusClientRTU.cpp  [ 240] handleConnection: Pulled request from queue
[D] Sent packet: @3FFDF588/6:
  | 0000: 01 03 00 01 00 28                                 |.....(          |
[D] 331606| ModbusClientRTU.cpp  [ 245] handleConnection: Request sent.
[D] 331645| RTUutils.cpp         [ 265] receive: 2052us without data
[D] Raw buffer received: @3FFE15C8/51:
  | 0000: 01 03 50 00 00 42 70 00  00 00 00 00 00 43 5C 00  |..P..Bp......C\.|
  | 0010: 00 41 20 00 00 40 40 00  00 42 C8 00 00 00 00 00  |.A ..@@..B......|
  | 0020: 00 00 00 00 00 00 00 00  00 3F 80 00 00 00 00 00  |.........?......|
  | 0030: 00 44 7A                                          |.Dz             |
[D] Received packet: @3FFDF588/1:
  | 0000: E2                                                |.               |
[D] 331691| ModbusClientRTU.cpp  [ 259] handleConnection: Error response (1 bytes) received.
[D] 331691| ModbusClientRTU.cpp  [ 280] handleConnection: Response generated.

Error: 0xe2 (CRC error)
```
I've used the following patch:
```patch
@@ -256,7 +256,7 @@ ModbusMessage RTUutils::receive(HardwareSerial& serial, uint32_t timeout, unsign
         // Are we past the interval gap without another byte?
         if (micros() - lastMicros >= interval) {
           // Yes, terminate reading
-          LOG_V("%ldus without data\n", micros() - lastMicros);
+          LOG_D("%ldus without data\n", micros() - lastMicros);
           state = DATA_READ;
         } else {
           // No, still in reading sequence
@@ -275,7 +275,9 @@ ModbusMessage RTUutils::receive(HardwareSerial& serial, uint32_t timeout, unsign
             }
           }
           // Buffer has space left - try to read another byte
+          digitalWrite(32, HIGH);
           b = serial.read();
+          digitalWrite(32, LOW);
         }
         break;
       // DATA_READ: successfully gathered some data. Prepare return object.
@@ -304,7 +306,11 @@ ModbusMessage RTUutils::receive(HardwareSerial& serial, uint32_t timeout, unsign
       case FINISHED:
         // CLear serial buffer in case something is left trailing
         // May happen with servers too slow!
-        while (serial.available()) serial.read();
+        while (serial.available()) {
+          digitalWrite(25, HIGH);
+          serial.read();
+          digitalWrite(25, LOW);
+        }
         break;
       }
     }
```

It look's like the `serial.read()` is interupted:
[image]
[modbus_truncated.zip](https://github.com/eModbus/eModbus/files/9718864/modbus_truncated.zip)

Miq1, 2022-10-06:
Dang. This is exactly the behaviour that was seen before with the older UART core handling. Thank you for taking the effort to check it.

There must be something going on in the core that I can not grab yet and is independent from the public Serial interface. The way we are using it is calling ``read()`` quite often - perhaps that is internally running against some limit?

I asked you before, but did not see an answer yet: what kind of ESP32 are you using? The two guys having the same issue had a M5stack, so just in case...

zivillian, 2022-10-06:
I'm using an ESP32-WROOM-32-U.

Miq1, 2022-10-06:
Great 😜 There goes my theory...

@bertmelis Do you have any idea left?

[15 more replies]

closed 2023-02-27 by Miq1