eModbus/eModbus #199: can someone help me to use the emodbus libarry

Frans400, 2022-06-26:
#include <ModbusRTU.h>
#include <esp_now.h>
#include <esp_wifi.h>
#include <WiFi.h>

#define board_ID 1

//uint8_t broadcastAddress[] = {0x34, 0xAB, 0x95, 0x65, 0xC9, 0x50};//  wrover    //MAC Address of the receiver 
uint8_t broadcastAddress[] = {0x94, 0xb9, 0x7e, 0xe6, 0x11, 0x24};//  esp cam   //MAC Address of the receiver 

typedef struct struct_message {
    int board_id;
    int a;
    int b;
    int c;
    int d;
    int e;
    int f;
    int Num_Readings;
} struct_message;

struct_message sensor_readings;

unsigned long previous_time = 0;   
const long interval = 1000;    

unsigned int Num_Readings = 0;

constexpr char wifi_SSID[] = "zzzz";

int32_t obtain_wifi(const char *ssid) {
  if (int32_t n = WiFi.scanNetworks()) {
      for (uint8_t i=0; i<n; i++) {
          if (!strcmp(ssid, WiFi.SSID(i).c_str())) {
              return WiFi.channel(i);
          }
      }
  }
  return 0;
}

void data_sent(const uint8_t *mac_addr, esp_now_send_status_t status) {
  Serial.print("\r\nLast Readings Send Status:\t");
  Serial.println(status == ESP_NOW_SEND_SUCCESS ? "Delivery Success" : "Delivery Fail");
}
int int_counter=1;

ModbusRTU mb;

int int_solar;
uint16_t uint16_t_solar;
bool bool_solar(Modbus::ResultCode event, uint16_t transactionId, void* data) {
  return true;
}

int int_load;
uint16_t uint16_t_load;
bool bool_load(Modbus::ResultCode event, uint16_t transactionId, void* data) {
  return true;
}

int int_grid;
int int_grid_fix;
uint16_t uint16_t_grid;
bool bool_grid(Modbus::ResultCode event, uint16_t transactionId, void* data) {
  return true;
}

int int_battery;
int int_battery_fix;
uint16_t uint16_t_battery;
bool bool_battery(Modbus::ResultCode event, uint16_t transactionId, void* data) {
  return true;
}

int int_soc;
uint16_t uint16_t_soc;
bool bool_soc(Modbus::ResultCode event, uint16_t transactionId, void* data) {
  return true;
}

int int_sog;
uint16_t uint16_t_sog;
bool bool_sog(Modbus::ResultCode event, uint16_t transactionId, void* data) {
  return true;
}

//int aa;
//int bb;
//int cc;
//int dd;
//int ee;
//int ff;

TaskHandle_t Task1;
TaskHandle_t Task2;

void setup() {
  WiFi.mode(WIFI_AP_STA);
  int32_t channel = obtain_wifi(wifi_SSID);
  WiFi.printDiag(Serial); 
  esp_wifi_set_promiscuous(true);
  esp_wifi_set_channel(channel, WIFI_SECOND_CHAN_NONE);
  esp_wifi_set_promiscuous(false);
  WiFi.printDiag(Serial);

  if (esp_now_init() != ESP_OK) {
    Serial.println("Error initializing ESP-NOW");
    return;
  }

  esp_now_register_send_cb(data_sent);
  
  esp_now_peer_info_t peerInfo;
  memset(&peerInfo, 0, sizeof(peerInfo));
  memcpy(peerInfo.peer_addr, broadcastAddress, 6);
  peerInfo.encrypt = false;       
  if (esp_now_add_peer(&peerInfo) != ESP_OK){
    Serial.println("Failed to add peer");
    return;
  }
  Serial.begin(115200);
  Serial2.begin(9600, SERIAL_8N1, 16, 17);//16 17 or 2 15
  mb.begin(&Serial2);
  mb.master();

  //create a task that will be executed in the Task1code() function, with priority 1 and executed on core 0
  xTaskCreatePinnedToCore(
                    Task1code,   /* Task function. */
                    "Task1",     /* name of task. */
                    10000,       /* Stack size of task */
                    NULL,        /* parameter of the task */
                    1,           /* priority of the task */
                    &Task1,      /* Task handle to keep track of created task */
                    0);          /* pin task to core 0 */                  

  //create a task that will be executed in the Task2code() function, with priority 1 and executed on core 1
  xTaskCreatePinnedToCore(
                    Task2code,   /* Task function. */
                    "Task2",     /* name of task. */
                    10000,       /* Stack size of task */
                    NULL,        /* parameter of the task */
                    1,           /* priority of the task */
                    &Task2,      /* Task handle to keep track of created task */
                    1);          /* pin task to core 1 */
}

//Task1code: all other stuff not modbus rtu read//
void Task1code( void * pvParameters ){
  for(;;){
    
      unsigned long current_time = millis();
      if (current_time - previous_time >= interval) {
        previous_time = current_time;
        sensor_readings.board_id = board_ID;
        sensor_readings.a = int_solar;
        sensor_readings.b = int_load;
        if (int_battery > 32767) {
          int_battery_fix =int_battery - 65535; 
        }else {
          int_battery_fix =int_battery ;
        } 
        sensor_readings.c = int_grid_fix;
        if (int_grid > 32767) {
        int_grid_fix =int_grid - 65535; 
        }else {
          int_grid_fix =int_grid ;
        }
        
        sensor_readings.d = int_grid_fix;
        sensor_readings.e = int_soc;
        sensor_readings.f = int_sog;
        sensor_readings.Num_Readings = Num_Readings++;
    
      esp_err_t result = esp_now_send(broadcastAddress, (uint8_t *) &sensor_readings, sizeof(sensor_readings));
    }
    delay(1000);
    } 
}

//Task2code: modbus rtu read//
void Task2code( void * pvParameters ){
  for(;;){

      mb.task();
      yield();
      if (!mb.slave()) {
        mb.readHreg(1, 186, &uint16_t_solar, 1, bool_solar);
      }
      int_solar = (int(uint16_t_solar));

      mb.task();
      yield();  
      if (!mb.slave()) {
        mb.readHreg(1, 178, &uint16_t_load, 1,  bool_load);
      }
      mb.task();
      yield();
      int_load = (int(uint16_t_load));

      mb.task();
      yield();    
      if (!mb.slave()) {
        mb.readHreg(1, 172, &uint16_t_grid, 1, bool_grid);
      }
      mb.task();
      yield();
      int_grid = (int(uint16_t_grid));

      mb.task();
      yield();
      if (!mb.slave()) {
        mb.readHreg(1, 190, &uint16_t_battery, 1,  bool_battery);
      }
      mb.task();
      yield();
      int_battery = (int(uint16_t_battery));
 
      mb.task();
      yield();   
      if (!mb.slave()) {
        mb.readHreg(1, 194, &uint16_t_sog, 1, bool_sog);
      }
      mb.task();
      yield();
      int_sog = (int(uint16_t_sog));

      mb.task();
      yield();
      if (!mb.slave()) {
        mb.readHreg(1, 184, &uint16_t_soc, 1,  bool_soc);
      }
      mb.task();
      yield();
      int_soc = (int(uint16_t_soc));
    }
}

void loop() {
  delay(100000);
  
}

Frans400, 2022-06-26:
As you can see in my code above I'm currently using the Arduino modbus rtu lib. Its working, but my esp32 restarts every 10 sec or so when i run this code. 

I would appreciate it if someone can do a small sample code showing how to read a holding register(fc3) one word, worker one 
holding register number 186 to a integer value using the emodbus lib.

Miq1, 2022-06-26:
You seem to be using a different library, the objects and methods you are using are not part of the ``eMosdbus`` lib.

closed 2022-06-26 by Miq1