eModbus/eModbus #204: ASCII Error response: EF - Invalid ASCII character

domotica-undici, 2022-07-26:
Hello, 
trying to read from an ASCII device. using modpoll everithing is working fine, but nothing here.
modpoll command working properly:
     modpoll -b 9600 -m ascii -p even -d 7 -a 4 -r 201 -c 1 /dev/ttyUSB2

emodbus master code:

 Serial2.begin(9600, SERIAL_7E1, GPIO_NUM_16, GPIO_NUM_17);
 MB.useModbusASCII();
 MB.onDataHandler(&handleData);
 MB.onErrorHandler(&handleError);
 MB.setTimeout(100);
 MB.begin(1);
 Error err = MB.addRequest(Token++, 4, READ_HOLD_REGISTER, 201, 1);

Log says:
handleError(): Error response: EF - Invalid ASCII character

any help?

Miq1, 2022-07-26:
It would be helpful to see the full packet data sent and received. To get that, you will want to set ``-DLOG_LEVEL=6`` as a compiler flag or temporarily set the default log level in ``Logging.h`` to ``LOG_LEVEL_VERBOSE``.
Please then post the log here for us to analyze.

The EF error normally will be returned if a byte in the received message is not a ASCII number, nor a ``:`` (colon) nor CR or LF. All else bytes are forbidden in Modbus ASCII.

Miq1, 2022-07-26:
Oops: this ``SERIAL_7E1`` is not possible for Modbus in general, 8 data bits are mandatory.

domotica-undici, 2022-07-26:
first of all, thanx for your quck reply! well, my device is 7E1 , it is a shame to not be able to use you great library. thank you anyway

Miq1, 2022-07-26:
> first of all, thanx for your quck reply! well, my device is 7E1 , it is a shame to not be able to use you great library. thank you anyway

Do not give up that quickly! If it is non-standard, it may work nevertheless. Please try to provide the verbose logs, as described above, and we may find a solution.

domotica-undici, 2022-07-26:
[V] 110590| ModbusMessage.cpp    [ 477] checkData: Check data #3
[D] 110590| ModbusClientRTU.cpp  [ 139] addRequestM: request for 04/03
[D] 110600| ModbusClientRTU.cpp  [ 188] addToQueue: RC=01
[D] 110600| ModbusClientRTU.cpp  [ 150] addRequestM: RC=00
[D] 110600| ModbusClientRTU.cpp  [ 206] handleConnection: Pulled request from queue
[D] Sent packet: @3FFB39A0/6:
  | 0000: 04 03 00 C9 00 01                                 |......          |
[D] 110628| ModbusClientRTU.cpp  [ 211] handleConnection: Request sent.
[D] Received packet: @3FFB392C/1:
  | 0000: EF                                                |.               |
[D] 110649| ModbusClientRTU.cpp  [ 223] handleConnection: Error response (1 bytes) received.
[V] Data: @3FFB392C/1:
  | 0000: EF                                                |.               |
[D] 110660| ModbusClientRTU.cpp  [ 244] handleConnection: Response generated.
[V] Response packet: @3FFB392C/3:
  | 0000: 04 83 EF                                          |...             |
[110680][I][main_master.cpp:56] handleError(): Error response: EF - Invalid ASCII character

domotica-undici, 2022-07-26:
a funny and strange thing is that log is the same for both device powered or unpowered and remains the same even if i swap A and B

Miq1, 2022-07-26:
Now we get further. The packet sent definitely is not Modbus ASCII, but a regular Modbus request without a CRC. This indeed is looking like a bug.
Which library version did you use? The current ``Master`` branch, the ``Tiny-server`` branch or another?

domotica-undici, 2022-07-26:
i'm using the latest version on master branch. some news: setting MB.setTimeout(100) (before was 2000) about 1 on 20 or 30 mesages get the right response. here is the log

[V] 213539| ModbusMessage.cpp    [ 477] checkData: Check data #3
[D] 213539| ModbusClientRTU.cpp  [ 139] addRequestM: request for 04/03
[D] 213549| ModbusClientRTU.cpp  [ 188] addToQueue: RC=01
[D] 213549| ModbusClientRTU.cpp  [ 150] addRequestM: RC=00
[D] 213551| ModbusClientRTU.cpp  [ 206] handleConnection: Pulled request from queue
[D] Sent packet: @3FFB39A0/6:
  | 0000: 04 03 00 C8 00 01                                 |......          |
[D] 213578| ModbusClientRTU.cpp  [ 211] handleConnection: Request sent.
[D] Received packet: @3FFB392C/1:
  | 0000: EF                                                |.               |
[D] 213599| ModbusClientRTU.cpp  [ 223] handleConnection: Error response (1 bytes) received.
[V] Data: @3FFB392C/1:
  | 0000: EF                                                |.               |
[D] 213609| ModbusClientRTU.cpp  [ 244] handleConnection: Response generated.
[V] Response packet: @3FFB392C/3:
  | 0000: 04 83 EF                                          |...             |
[213630][I][main_master.cpp:56] handleError(): Error response: EF - Invalid ASCII character
[V] 214579| ModbusMessage.cpp    [ 477] checkData: Check data #3
[D] 214579| ModbusClientRTU.cpp  [ 139] addRequestM: request for 04/03
[D] 214589| ModbusClientRTU.cpp  [ 188] addToQueue: RC=01
[D] 214589| ModbusClientRTU.cpp  [ 150] addRequestM: RC=00
[D] 214591| ModbusClientRTU.cpp  [ 206] handleConnection: Pulled request from queue
[D] Sent packet: @3FFB39A0/6:
  | 0000: 04 03 00 C8 00 01                                 |......          |
[D] 214618| ModbusClientRTU.cpp  [ 211] handleConnection: Request sent.
[D] Raw buffer received: @3FFB39A0/6:
  | 0000: 04 03 02 00 04 F3                                 |......          |
[D] Received packet: @3FFB3940/5:
  | 0000: 04 03 02 00 04                                    |.....           |
[D] 214666| ModbusClientRTU.cpp  [ 223] handleConnection: Data response (5 bytes) received.
[V] Data: @3FFB3940/5:
  | 0000: 04 03 02 00 04                                    |.....           |
[D] 214687| ModbusClientRTU.cpp  [ 244] handleConnection: Response generated.
[V] Response packet: @3FFB3940/5:
  | 0000: 04 03 02 00 04                                    |.....           |
[214708][I][main_master.cpp:38] handleData(): Response: serverID=4, FC=3, Token=000000D1, length=5, data=4

Miq1, 2022-07-26:
To see the raw data, please intentionally comment out your line with ``useModbusASCII()``. This will result in an error again, but we may get a glimpse on the raw data the server is sending.

domotica-undici, 2022-07-26:
here is the log without useASCII

[V] Data: @3FFB3944/1:
  | 0000: E0                                                |.               |
[D] 16418| ModbusClientRTU.cpp  [ 244] handleConnection: Response generated.
[V] Response packet: @3FFB3944/3:
  | 0000: 04 83 E0                                          |...             |
[ 16439][I][main_master.cpp:56] handleError(): Error response: E0 - Timeout
[V] 17267| ModbusMessage.cpp    [ 477] checkData: Check data #3
[D] 17267| ModbusClientRTU.cpp  [ 139] addRequestM: request for 04/03
[D] 17277| ModbusClientRTU.cpp  [ 188] addToQueue: RC=01
[D] 17277| ModbusClientRTU.cpp  [ 150] addRequestM: RC=00
[D] 17279| ModbusClientRTU.cpp  [ 206] handleConnection: Pulled request from queue
[D] Sent packet: @3FFB3944/6:
  | 0000: 04 03 00 C8 00 01                                 |......          |
[D] 17297| ModbusClientRTU.cpp  [ 211] handleConnection: Request sent.
[V] 17311| RTUutils.cpp         [ 289] receive: 3653us without data
[D] Raw buffer received: @3FFB39A0/2:
  | 0000: 04 00                                             |..              |
[D] Received packet: @3FFB3944/1:
  | 0000: E5                                                |.               |
[D] 17332| ModbusClientRTU.cpp  [ 223] handleConnection: Error response (1 bytes) received.

Miq1, 2022-07-26:
Please use the ``Tiny-server`` branch if you can.It has some modifications in the UART handling and will be the master branch soon.

Miq1, 2022-07-26:
D] Raw buffer received: @3FFB39A0/2:
| 0000: 04 00 |.. |

This is kind of enlightening: the response has a binary server ID, not ASCII, and the function code supposedly is a zero. Are you sure your device is doing a proper Modbus ASCII protocol?

domotica-undici, 2022-07-26:
i was querying the address with the slave id so 4 is right answer.
using
lib_deps = 
  https://github.com/eModbus/eModbus.git#Tiny-server

i get the same results: only few response over a lot of requests. here is the log

[V] 18288| ModbusMessage.cpp    [ 477] checkData: Check data #3
[D] 18288| ModbusClientRTU.cpp  [ 139] addRequestM: request for 04/03
[D] 18298| ModbusClientRTU.cpp  [ 219] addToQueue: RC=01
[D] 18298| ModbusClientRTU.cpp  [ 150] addRequestM: RC=00
[D] 18300| ModbusClientRTU.cpp  [ 237] handleConnection: Pulled request from queue
[D] Sent packet: @3FFB3944/6:
  | 0000: 04 03 00 C8 00 01                                 |......          |
[D] 18328| ModbusClientRTU.cpp  [ 242] handleConnection: Request sent.
[D] Received packet: @3FFB3944/1:
  | 0000: EF                                                |.               |
[D] 18349| ModbusClientRTU.cpp  [ 256] handleConnection: Error response (1 bytes) received.
[V] Data: @3FFB3944/1:
  | 0000: EF                                                |.               |
[D] 18359| ModbusClientRTU.cpp  [ 277] handleConnection: Response generated.
[V] Response packet: @3FFB3944/3:
  | 0000: 04 83 EF                                          |...             |
[ 18380][I][main_master.cpp:56] handleError(): Error response: EF - Invalid ASCII character
[V] 19299| ModbusMessage.cpp    [ 477] checkData: Check data #3
[D] 19299| ModbusClientRTU.cpp  [ 139] addRequestM: request for 04/03
[D] 19309| ModbusClientRTU.cpp  [ 219] addToQueue: RC=01
[D] 19309| ModbusClientRTU.cpp  [ 150] addRequestM: RC=00
[D] 19311| ModbusClientRTU.cpp  [ 237] handleConnection: Pulled request from queue
[D] Sent packet: @3FFB3944/6:
  | 0000: 04 03 00 C8 00 01                                 |......          |
[D] 19339| ModbusClientRTU.cpp  [ 242] handleConnection: Request sent.
[D] Raw buffer received: @3FFB39A0/6:
  | 0000: 04 03 02 00 04 F3                                 |......          |
[D] Received packet: @3FFB3958/5:
  | 0000: 04 03 02 00 04                                    |.....           |
[D] 19387| ModbusClientRTU.cpp  [ 256] handleConnection: Data response (5 bytes) received.
[V] Data: @3FFB3958/5:
  | 0000: 04 03 02 00 04                                    |.....           |
[D] 19408| ModbusClientRTU.cpp  [ 277] handleConnection: Response generated.
[V] Response packet: @3FFB3958/5:
  | 0000: 04 03 02 00 04                                    |.....           |
[ 19428][I][main_master.cpp:38] handleData(): Response: serverID=4, FC=3, Token=00000013, length=5, data=4

Miq1, 2022-07-26:
Do you happen to use a M5Stack device?

Miq1, 2022-07-26:
> i was querying the address with the slave id so 4 is right answer. 

Only if the server has a kind of rudimentary standard Modbus support - else the response to be expected was either nil, due to no recognizable request, or something starting ":048..."

domotica-undici, 2022-07-27:
I'm trying to communicate with a Viessmann fancoil, who has an olympia splendid main board that speaks modbus ascii.
even if is in italian, you can see datasheet specs:

Le periferiche bordo macchina dispongono di una porta di comunicazione seriale:
MODICON MODBUS, modo **ASCII, rete RS485
Baudrate = 9600, data bits = 7, stop bit = 1, parità = pari.**

i'm trying to work with different timeouts but with no luck

domotica-undici, 2022-07-27:
also: i use to use openhab platform to communicate by modbus with any kind of device and have no troubles, even with this device

Miq1, 2022-07-27:
😁 
I was not going to blame your setup at all, sorry if it sounded as if I did.

My first question was regarding the ESP32 type of device, since we had UART issues reported with the M5Stack devices so far. If yours is different and the issues were similar, we had an indication there is more to it than just one type of device malfunctioning.

Regarding the 7 bits: makes totally sense for Modbus ASCII, because all characters of the messages are below 128.

If you like, you could add a ``LOG_N("%02X ", b);`` line in RTUutils.cpp at line 372 after the "Is it a valid character?" comment. Then we will see the offending character.

domotica-undici, 2022-07-27:
haha, dont worry i've never thought you meant to blame me! :-)
i 'm using a common ESP-WROOM-32 

this is the offending char given by log

[N] 25090| RTUutils.cpp         [ 373] receive: 4C [D] Received packet: @3FFB392C/1:
  | 0000: EF

Miq1, 2022-07-27:
0x4C??? That is an uppercase 'L' - definitely not valid with Modbus ASCII.

bertmelis, 2022-07-27:
Is modicon the same as regular modbus?

domotica-undici, 2022-07-27:
about 0x4C... well: i don't know modbus so deep to understand why i get 0x4C. what happens is that sometimes i get a well formatted answer but the most are invalid. 

modicon seems to be the ancient version, i've never cared of it because using www.openhab.org i've got no problems reading and writing this device. 

i suppose problems are about device's age. other tests with different kind of modbus devices working perfectly. 
only a note: with standard rtu i always need to use MB.skipLeading0x00(); function to read properly. without this function i got a lot of errors.

i think i can definitely say that this library works good, but need "recent" devices

domotica-undici, 2022-07-27:
to complete, here is a more long log

[V] 1586281| ModbusMessage.cpp    [ 477] checkData: Check data #3
[D] 1586281| ModbusClientRTU.cpp  [ 139] addRequestM: request for 04/06
[D] 1586291| ModbusClientRTU.cpp  [ 219] addToQueue: RC=01
[D] 1586291| ModbusClientRTU.cpp  [ 150] addRequestM: RC=00
[D] 1586293| ModbusClientRTU.cpp  [ 237] handleConnection: Pulled request from queue
[D] Sent packet: @3FFB39A0/6:
  | 0000: 04 06 00 65 40 03                                 |...e@.          |
[D] 1586320| ModbusClientRTU.cpp  [ 242] handleConnection: Request sent.
[N] 1586330| RTUutils.cpp         [ 373] receive: 34 [N] 1586331| RTUutils.cpp         [ 373] receive: 00 [D] Received packet: @3FFB392C/1:
  | 0000: EF                                                |.               |
[D] 1586351| ModbusClientRTU.cpp  [ 256] handleConnection: Error response (1 bytes) received.
[V] Data: @3FFB392C/1:
  | 0000: EF                                                |.               |
[D] 1586362| ModbusClientRTU.cpp  [ 277] handleConnection: Response generated.
[V] Response packet: @3FFB392C/3:
  | 0000: 04 86 EF                                          |...             |
[1586382][I][main_master.cpp:56] handleError(): Error response: EF - Invalid ASCII character
[V] 1586421| ModbusMessage.cpp    [ 477] checkData: Check data #3
[D] 1586421| ModbusClientRTU.cpp  [ 139] addRequestM: request for 04/06
[D] 1586431| ModbusClientRTU.cpp  [ 219] addToQueue: RC=01
[D] 1586431| ModbusClientRTU.cpp  [ 150] addRequestM: RC=00
[D] 1586433| ModbusClientRTU.cpp  [ 237] handleConnection: Pulled request from queue
[D] Sent packet: @3FFB39A0/6:
  | 0000: 04 06 00 66 00 C8                                 |...f..          |
[D] 1586460| ModbusClientRTU.cpp  [ 242] handleConnection: Request sent.
[N] 1586470| RTUutils.cpp         [ 373] receive: 34 [N] 1586471| RTUutils.cpp         [ 373] receive: 00 [D] Received packet: @3FFB392C/1:
  | 0000: EF                                                |.               |
[D] 1586491| ModbusClientRTU.cpp  [ 256] handleConnection: Error response (1 bytes) received.
[V] Data: @3FFB392C/1:
  | 0000: EF                                                |.               |
[D] 1586502| ModbusClientRTU.cpp  [ 277] handleConnection: Response generated.
[V] Response packet: @3FFB392C/3:
  | 0000: 04 86 EF                                          |...             |
[1586522][I][main_master.cpp:56] handleError(): Error response: EF - Invalid ASCII character
[V] 1586561| ModbusMessage.cpp    [ 477] checkData: Check data #3
[D] 1586561| ModbusClientRTU.cpp  [ 139] addRequestM: request for 04/06
[D] 1586571| ModbusClientRTU.cpp  [ 219] addToQueue: RC=01
[D] 1586571| ModbusClientRTU.cpp  [ 150] addRequestM: RC=00
[D] 1586573| ModbusClientRTU.cpp  [ 237] handleConnection: Pulled request from queue
[D] Sent packet: @3FFB39A0/6:
  | 0000: 04 06 00 67 00 C8                                 |...g..          |
[D] 1586600| ModbusClientRTU.cpp  [ 242] handleConnection: Request sent.
[N] 1586610| RTUutils.cpp         [ 373] receive: 34 [N] 1586611| RTUutils.cpp         [ 373] receive: 00 [D] Received packet: @3FFB392C/1:
  | 0000: EF                                                |.               |
[D] 1586631| ModbusClientRTU.cpp  [ 256] handleConnection: Error response (1 bytes) received.
[V] Data: @3FFB392C/1:
  | 0000: EF                                                |.               |
[D] 1586642| ModbusClientRTU.cpp  [ 277] handleConnection: Response generated.
[V] Response packet: @3FFB392C/3:
  | 0000: 04 86 EF                                          |...             |
[1586662][I][main_master.cpp:56] handleError(): Error response: EF - Invalid ASCII character
[V] 1587601| ModbusMessage.cpp    [ 477] checkData: Check data #3
[D] 1587601| ModbusClientRTU.cpp  [ 139] addRequestM: request for 04/06
[D] 1587611| ModbusClientRTU.cpp  [ 219] addToQueue: RC=01
[D] 1587611| ModbusClientRTU.cpp  [ 150] addRequestM: RC=00
[D] 1587613| ModbusClientRTU.cpp  [ 237] handleConnection: Pulled request from queue
[D] Sent packet: @3FFB39A0/6:
  | 0000: 04 06 00 65 40 03                                 |...e@.          |
[D] 1587640| ModbusClientRTU.cpp  [ 242] handleConnection: Request sent.
[N] 1587651| RTUutils.cpp         [ 373] receive: 22 [D] Received packet: @3FFB392C/1:
  | 0000: EF                                                |.               |
[D] 1587661| ModbusClientRTU.cpp  [ 256] handleConnection: Error response (1 bytes) received.
[V] Data: @3FFB392C/1:
  | 0000: EF                                                |.               |
[D] 1587682| ModbusClientRTU.cpp  [ 277] handleConnection: Response generated.
[V] Response packet: @3FFB392C/3:
  | 0000: 04 86 EF                                          |...             |
[1587702][I][main_master.cpp:56] handleError(): Error response: EF - Invalid ASCII character
[... 349 lines cut ...]
  | 0000: EF                                                |.               |
[D] 1596901| ModbusClientRTU.cpp  [ 256] handleConnection: Error response (1 bytes) received.
[V] Data: @3FFB39A0/1:
  | 0000: EF                                                |.               |
[D] 1596922| ModbusClientRTU.cpp  [ 277] handleConnection: Response generated.
[V] Response packet: @3FFB39A0/3:
  | 0000: 04 86 EF                                          |...             |
[1596942][I][main_master.cpp:56] handleError(): Error response: EF - Invalid ASCII character
[... cut ...]
  | 0000: EF                                                |.               |
[D] 1597041| ModbusClientRTU.cpp  [ 256] handleConnection: Error response (1 bytes) received.
[V] Data: @3FFB39A0/1:
  | 0000: EF                                                |.               |
[D] 1597062| ModbusClientRTU.cpp  [ 277] handleConnection: Response generated.
[V] Response packet: @3FFB39A0/3:
  | 0000: 04 86 EF                                          |...             |
[1597082][I][main_master.cpp:56] handleError(): Error response: EF - Invalid ASCII character

bertmelis, 2022-07-27:
I suppose your hardware setup is different when using the ESP compared to OpenHAB. Could you tell us more about the setup? (connections, termination, adapters...)

Miq1, 2022-07-27:
There is a lot of strange data coming in. In the ``if`` clause immediately after the ``LOG_N()`` line you added, can you alter that to start like ``if (0 && `` to prevent illegal characters from stopping the loop? Then we will see the complete sequence coming in.

domotica-undici, 2022-07-27:
i use a common ESP32-WROOM with pin 16 and 17 used to Serial2, while pin 18 is used as rtsPin.
i'm testing with two boards: one made by myself using a MAX485 as tranceiver directly connected to esp32 pins. and using a 120 ohm resistor between a and b. 
the second board is a LOIN32, a common from amazon + this board
https://www.amazon.it/gp/product/B0785B3BS8/ref=ppx_yo_dt_b_asin_title_o01_s00?ie=UTF8&psc=1
to convert 485 to 232.
in this last board i've unsoldered pullup resistor on both sides of the transceiver.
pin used are the same on both boards

domotica-undici, 2022-07-27:
> There is a lot of strange data coming in. In the `if` clause immediately after the `LOG_N()` line you added, can you alter that to start like `if (0 && ` to prevent illegal characters from stopping the loop? Then we will see the complete sequence coming in.

you men like this? if (0 && (b & 0x80) || ASCIIread[b] == 0xFF) {

Miq1, 2022-07-27:
> you men like this? if (0 && (b & 0x80) || ASCIIread[b] == 0xFF) {

yes, just let it evaluate to false in any case.

domotica-undici, 2022-07-27:
haha! with if (0) it works perfectly!!

[V] 69849| ModbusMessage.cpp    [ 477] checkData: Check data #3
[D] 69849| ModbusClientRTU.cpp  [ 139] addRequestM: request for 04/03
[D] 69859| ModbusClientRTU.cpp  [ 219] addToQueue: RC=01
[D] 69859| ModbusClientRTU.cpp  [ 150] addRequestM: RC=00
[D] 69861| ModbusClientRTU.cpp  [ 237] handleConnection: Pulled request from queue
[D] Sent packet: @3FFB3944/6:
  | 0000: 04 03 00 C8 00 01                                 |......          |
[D] 69889| ModbusClientRTU.cpp  [ 242] handleConnection: Request sent.
[N] 69899| RTUutils.cpp         [ 373] receive: 26 [N] 69899| RTUutils.cpp         [ 373] receive: 00 [N] 69910| RTUutils.cpp         [ 373] receive: 7F [N] 69912| RTUutils.cpp         [ 373] receive: 3A [N] 69913| RTUutils.cpp         [ 373] receive: 30 [N] 69923| RTUutils.cpp         [ 373] receive: 34 [N] 69923| RTUutils.cpp         [ 373] receive: 30 [N] 69933| RTUutils.cpp         [ 373] receive: 33 [N] 69933| RTUutils.cpp         [ 373] receive: 30 [N] 69934| RTUutils.cpp         [ 373] receive: 32 [N] 69944| RTUutils.cpp         [ 373] receive: 30 [N] 69944| RTUutils.cpp         [ 373] receive: 30 [N] 69954| RTUutils.cpp         [ 373] receive: 30 [N] 69954| RTUutils.cpp         [ 373] receive: 34 [N] 69965| RTUutils.cpp         [ 373] receive: 46 [N] 69965| RTUutils.cpp         [ 373] receive: 33 [N] 69965| RTUutils.cpp         [ 373] receive: 0D [N] 69975| RTUutils.cpp         [ 373] receive: 0A [D] Raw buffer received: @3FFB39A0/6:
  | 0000: 04 03 02 00 04 F3                                 |......          |
[D] Received packet: @3FFB3958/5:
  | 0000: 04 03 02 00 04                                    |.....           |
[D] 69996| ModbusClientRTU.cpp  [ 256] handleConnection: Data response (5 bytes) received.
[V] Data: @3FFB3958/5:
  | 0000: 04 03 02 00 04                                    |.....           |
[D] 70017| ModbusClientRTU.cpp  [ 277] handleConnection: Response generated.
[V] Response packet: @3FFB3958/5:
  | 0000: 04 03 02 00 04                                    |.....           |
[ 70027][I][main_master.cpp:38] handleData(): Response: serverID=4, FC=3, Token=00000045, length=5, data=4
[V] 70860| ModbusMessage.cpp    [ 477] checkData: Check data #3
[D] 70860| ModbusClientRTU.cpp  [ 139] addRequestM: request for 04/03
[D] 70870| ModbusClientRTU.cpp  [ 219] addToQueue: RC=01
[D] 70870| ModbusClientRTU.cpp  [ 150] addRequestM: RC=00
[D] 70872| ModbusClientRTU.cpp  [ 237] handleConnection: Pulled request from queue
[D] Sent packet: @3FFB3944/6:
  | 0000: 04 03 00 C8 00 01                                 |......          |
[D] 70900| ModbusClientRTU.cpp  [ 242] handleConnection: Request sent.
[N] 70910| RTUutils.cpp         [ 373] receive: 22 [N] 70910| RTUutils.cpp         [ 373] receive: 03 [N] 70921| RTUutils.cpp         [ 373] receive: 1A [N] 70921| RTUutils.cpp         [ 373] receive: 02 [N] 70931| RTUutils.cpp         [ 373] receive: 6A [N] 70931| RTUutils.cpp         [ 373] receive: 00 [N] 70931| RTUutils.cpp         [ 373] receive: 3A [N] 70942| RTUutils.cpp         [ 373] receive: 30 [N] 70942| RTUutils.cpp         [ 373] receive: 34 [N] 70952| RTUutils.cpp         [ 373] receive: 30 [N] 70952| RTUutils.cpp         [ 373] receive: 33 [N] 70962| RTUutils.cpp         [ 373] receive: 30 [N] 70962| RTUutils.cpp         [ 373] receive: 32 [N] 70963| RTUutils.cpp         [ 373] receive: 30 [N] 70973| RTUutils.cpp         [ 373] receive: 30 [N] 70973| RTUutils.cpp         [ 373] receive: 30 [N] 70983| RTUutils.cpp         [ 373] receive: 34 [N] 70983| RTUutils.cpp         [ 373] receive: 46 [N] 70994| RTUutils.cpp         [ 373] receive: 33 [N] 70994| RTUutils.cpp         [ 373] receive: 0D [N] 70994| RTUutils.cpp         [ 373] receive: 0A [D] Raw buffer received: @3FFB39A0/6:
  | 0000: 04 03 02 00 04 F3                                 |......          |
[D] Received packet: @3FFB3958/5:
  | 0000: 04 03 02 00 04                                    |.....           |
[D] 71025| ModbusClientRTU.cpp  [ 256] handleConnection: Data response (5 bytes) received.
[V] Data: @3FFB3958/5:
  | 0000: 04 03 02 00 04                                    |.....           |
[D] 71035| ModbusClientRTU.cpp  [ 277] handleConnection: Response generated.
[V] Response packet: @3FFB3958/5:
  | 0000: 04 03 02 00 04                                    |.....           |
[ 71056][I][main_master.cpp:38] handleData(): Response: serverID=4, FC=3, Token=00000046, length=5, data=4
[V] 71871| ModbusMessage.cpp    [ 477] checkData: Check data #3
[D] 71871| ModbusClientRTU.cpp  [ 139] addRequestM: request for 04/03
[D] 71881| ModbusClientRTU.cpp  [ 219] addToQueue: RC=01
[D] 71881| ModbusClientRTU.cpp  [ 150] addRequestM: RC=00
[D] 71883| ModbusClientRTU.cpp  [ 237] handleConnection: Pulled request from queue
[D] Sent packet: @3FFB3944/6:
  | 0000: 04 03 00 C8 00 01                                 |......          |
[D] 71911| ModbusClientRTU.cpp  [ 242] handleConnection: Request sent.
[N] 71921| RTUutils.cpp         [ 373] receive: 26 [N] 71921| RTUutils.cpp         [ 373] receive: 0D [N] 71934| RTUutils.cpp         [ 373] receive: 3A [N] 71936| RTUutils.cpp         [ 373] receive: 30 [N] 71936| RTUutils.cpp         [ 373] receive: 34 [N] 71946| RTUutils.cpp         [ 373] receive: 30 [N] 71946| RTUutils.cpp         [ 373] receive: 33 [N] 71946| RTUutils.cpp         [ 373] receive: 30 [N] 71957| RTUutils.cpp         [ 373] receive: 32 [N] 71957| RTUutils.cpp         [ 37 [... cut ...]
  | 0000: 04 03 02 00 04                                    |.....           |
[D] 74041| ModbusClientRTU.cpp  [ 256] handleConnection: Data response (5 bytes) received.
[V] Data: @3FFB3958/5:
  | 0000: 04 03 02 00 04                                    |.....           |
[D] 74051| ModbusClientRTU.cpp  [ 277] handleConnection: Response generated.
[V] Response packet: @3FFB3958/5:
  | 0000: 04 03 02 00 04                                    |.....           |
[ 74072][I][main_master.cpp:38] handleData(): Response: serverID=4, FC=3, Token=00000049, length=5, data=4

domotica-undici, 2022-07-27:
i can confim: using if(0) intead of if ((b & 0x80) || ASCIIread[b] == 0xFF)  doesn't loose a shot!!!!

Miq1, 2022-07-27:
yeah,the message is there, but a lot of noise/garbage in front of it as well. You are lucky the garbage is containing nearly no valid character, so the conversion logic is filtering it, but there must be something wrong with your wiring. The line has to be quiet while no request or response is transmitted - and it is not.

bertmelis, 2022-07-27:
If your wiring is OK, you're sure you are not using the same UART port for something else?

domotica-undici, 2022-07-27:
uart is ok, i'll double checked wiring and solders and are ok. using " if ((b & 0x80) || ASCIIread[b] == 0xFF)  " i get only a few good response. using "  if(0) " works like a charm. 
using openhab modbus binding and a standard wiring + usb/485 converter i got no errors at all. i don't know if that code has something that "corrects" errors

domotica-undici, 2022-07-28:
it seems i forgot to say THANK YOU for your support!

closed 2022-07-29 by domotica-undici

Miq1, 2022-07-29:
Good that you are satisfied with the outcome. The change does only cover a symptom, though, as a random colon (':') character in the garbage upfront will render the message unusable...