eModbus/eModbus #205: Stats: error counter

zivillian, 2022-07-27:
Currently the Bridge and RTU have counters for
- number of messages
- pending requests
- message count
- active clients
- etc.

It would be great to also have counter for error responses, either in the RTU client or the bridge.

My use case is to show it on my status page:
[image]

Miq1, 2022-07-28:
That would be an easy addition. Thank you for the suggestion, I will implement that eventually.

Miq1, 2022-07-30:
Rolled out with 1.6

closed 2022-07-30 by Miq1

bertmelis, 2022-07-30:
> @bertmelis I did not add the `errorCount++` to the async code - your domain, I do not want to trample down something unwittingly 😉

I'll implement in a separate PR.