eModbus/eModbus #209: EModbus with WT32-ETH1?

ArnaudVella, 2022-08-04:
Hello,
I am trying to adapt eModbus to use it with a WT32-ETH1 board (ESP32+LAN8720).
My attempts are so far unsuccessful. Has anyone else worked on this subject?

The ETH.h library was created for the LAN8720 but does not seem to be as complete as Ethernet.h which only works with the W5*00 family.

I I get stuck  in the modification of ModbusServerEthernet.h 
`using ModbusServerEthernet = ModbusServerTCP<EthernetServer, EthernetClient>;`

Miq1, 2022-08-04:
Hi,

I never encountered that board before, so my hints can only be theoretical unfortunately. 

The complete TCP communication is done in the ``serve``, ``worker`` and ``receive`` functions in ``ModbusServerTCPtemp.h`` for the servers, which is the more complicated part. 

The clients are using methods from the common ``Client.h`` only. I suppose the client defined in your ETH.h will be derived from that as well to work at all. 

So you may need to find the matching calls for accept, available etc. that are used for the server. and set up another ``ModbusServerTCPWT32.h`` or whatever file that is built like the existing ``ModbusServerEthernet.h`` containing wrapper functions for methods different with your chip.

ArnaudVella, 2022-08-12:
Hi, 
I found a simple solution but it may not be optimal. I use the WiFi.onEvent function to make the interface between ETH and ModbusServerWiFi.
Here is an example of eModbus code modified to work with a WT32-ETH1:
`
// =================================================================================================
// eModbus: Copyright 2021 by Michael Harwerth, Bert Melis and the contributors to ModbusClient
//               MIT license - see license.md for details
// =================================================================================================

// Includes: <Arduino.h> for Serial etc.
#include <Arduino.h>
#include "ModbusServerWiFi.h"
#include <ETH.h>

// Define Ethernet-based server
ModbusServerWiFi MBserver;

#define ETH_CLK_MODE    ETH_CLOCK_GPIO0_IN  // ETH_CLOCK_GPIO17_OUT
#define ETH_POWER_PIN   16                  // The enable signal for the external crystal oscillator (-1 to disable for internal APLL source)
#define ETH_TYPE        ETH_PHY_LAN8720     // Type of the Ethernet PHY LAN8720
#define ETH_ADDR        1                   // I²C-address of Ethernet PHY
#define ETH_MDC_PIN     23                  // The I²C clock signal for the Ethernet PHY
#define ETH_MDIO_PIN    18                  // The I²C IO signal for the Ethernet PHY
#define NRST            21                  // The nRST (enable) for the Ethernet PHY
static bool eth_connected = false;          // Global variable to pull status of eth.
`

// Create server
//uint16_t memo[32]; // Test server memory: 32 words
uint16_t MBHoldingRegister[37]={0}; // Test server memory: 37 registers starting from register 0

// Server function to handle FC 0x03 and 0x04
ModbusMessage FC03(ModbusMessage request)
{

  ModbusMessage response; // The Modbus message we are going to give back
  uint16_t addr = 0;      // Start address
  uint16_t words = 0;     // # of words requested
  request.get(2, addr);   // read address from request
  request.get(4, words);  // read # of words from request

  // Address overflow?
  if ((addr + words) > 37)
  {
    // Yes - send respective error response
    response.setError(request.getServerID(), request.getFunctionCode(), ILLEGAL_DATA_ADDRESS);
  }
  // Set up response
  response.add(request.getServerID(), request.getFunctionCode(), (uint8_t)(words * 2));
  // Request for FC 0x03?
  if (request.getFunctionCode() == READ_HOLD_REGISTER)
  {
    // Yes. Complete response
    for (uint8_t i = 0; i < words; ++i)
    {
      // send increasing data values
      response.add((uint16_t)(MBHoldingRegister[addr + i]));
    }
  }
  else
  {
    // No, this is for FC 0x04. Response is random
    for (uint8_t i = 0; i < words; ++i)
    {
      // send increasing data values
      response.add((uint16_t)random(1, 65535));
    }
  }
  // Send response back
  return response;
[... 122 lines cut ...]
{
  static uint32_t lastMillis = 0;

  if (millis() - lastMillis > 1000)
  {
    lastMillis = millis();
    for(int k=10; k<37; k++){
   Serial.print(" R");
   Serial.print(k);
   Serial.print(": ");
  Serial.print(MBHoldingRegister[k]);
}
Serial.println();
lastMillis = millis();
    if (MBHoldingRegister[22]==4000){   //boucle entre 0 et 4000
      MBHoldingRegister[22]=0;
    }
    else{
      MBHoldingRegister[22]++;
    } 
  }

  
}
`

closed 2022-08-12 by ArnaudVella

Miq1, 2022-08-12:
😲 ?!?

I cannot believe this will work. The ``ModbusServerWiFi`` is definitely using the ESP32's WiFi stack for all client connections and has no overlap at all with the Ethernet stack ``ETH`` will provide. Can it be your communication is done over WiFi without you noticing?

0x3333, 2023-08-13:
Sorry to comment on an old and closed issue, but I'm getting the same issue with a WT32-ETH1.

I'll give it a shot tomorrow on this code, but, I believe it could work because despite it using Wifi functions, he included `#include <ETH.h>` which will work with Wifi functions.

I'll give it a try and get back to you guys, because it is definitely broken for this setup board.