eModbus/eModbus #218: SoftwareSerial support?

kinsi55, 2022-09-01:
**Is your feature request related to a problem? Please describe.**
I am using a ESP-32S and need its hardwareserial for another device (Communicates at 115200 baud) however, need to also talk to a Modbus RTU device at 9600. Unfortunately, this library only supports Hardwareserial.

**Describe the solution you'd like**
Support for SoftwareSerial would be awesome 👍 

**Describe alternatives you've considered**
I suppose my only real alternative is using Serial0 which is kinda icky, or trying to put the other device on Software which I dont think is gonna work well.

Miq1, 2022-09-01:
There is a [Fork of eModbus](https://github.com/serfreeman1337/eModbus) done by serfreeman1337, that seems to have added SoftwareSerial. I did not try it myself nor have any shares in serfreeman1337's developments, but that could be a solution.

kinsi55, 2022-09-01:
I ended up using ModbusMaster which works fine for my needs, thanks regardless!

Miq1, 2022-09-01:
Fine with that. Good luck!

closed 2022-09-01 by Miq1