eModbus/eModbus #227: TCP async client is lacking onResponse handling with timeout

Miq1, 2022-10-13:
If the TCP async client is running into a request timeout, only the ``onError`` handler is called, if registered, A potential ``onResponse`` handler is ignored.

Miq1, 2022-10-13:
I may have fixed it - needed it, else I had not interfered with your code 😉

Miq1, 2022-10-13:
Ooops. Messed it up, I am afraid.

bertmelis, 2022-10-13:
I haven't had time yet to look at it.

Miq1, 2022-10-13:
No worries, but how can I rollback my commit?

Miq1, 2022-10-13:
> Rewrite history? (Which I'm fine with) Or just commit again?

I do not find the former, so will do the latter.

Miq1, 2022-10-13:
Undid the changes. The bug also can be found in the ``onDisconnect`` function, where also only the ``onError`` case is considered. The ``onError`` call signature is different to that in the non-async code, so I think there may be more to it than to simply add some ``onResponse`` call like I attempted to do.

Miq1, 2022-10-13:
Uh-oh... There is something else fishy in there. I manage to get a WDT reset upon sending a request:

```
[V] 38364| ModbusMessage.cpp    [ 477] checkData: Check data #3
[V] Enqueue: @3FFF11A4/6:
  | 0000: 01 06 00 01 00 01                                 |......          |
[D] 38368| ModbusClientTCPAsync.cpp [  60] connect: connecting
[D] 38374| ModbusClientTCPAsync.cpp [ 112] addRequestM: Add TCP request result: 00

 ets Jan  8 2013,rst cause:4, boot mode:(3,7)

wdt reset
load 0x4010f000, len 3460, room 16 
tail 4
chksum 0xcc
load 0x3fff20b8, len 40, room 4
tail 4
chksum 0xc9
csum 0xc9
v00059650
~ld
```

Miq1, 2022-10-13:
> Uh-oh... There is something else fishy in there. I manage to get a WDT reset upon sending a request:

Got that. One needs to do an explicit ``connect()`` first, as the last target is not kept with the async client. If you will blindly fire another request for the same target that has disconnected in the meantime, you will end up as above.

I do not really like that - the non-async client is keeping a target internally (modifiable by ``setTarget()``) and do the connect automatically if a fresh request comes in. Was it possible to add that to the async client, too?

bertmelis, 2022-10-13:
Possible, yes of course.

Miq1, 2022-10-14:
> Possible, yes of course.

Yes, please? 🙏🏻 😉

Miq1, 2022-10-19:
@bertmelis Can you have a look at [this commit @mcuadros did](https://github.com/eModbus/eModbus/commit/020187e29c8b5819bb25f24080e81445ae447a44)?
It lacks a few bits on the ``onResponse`` corrections, but the ``onDisconnect`` stuff could be interesting for the async code as well - I simply cannot judge that.

bertmelis, 2022-10-21:
I'm back home next week.