eModbus/eModbus #228: Several problems using ModbusClientTCPasync

mcuadros, 2022-10-23:
**Describe the bug**

Working on my project using ModbusClientTCPasync with `syncRequest` I run into several problems, this problems caused memory leaks and restastarts under a heavy usage in 30mins or so.

Due to my own deadlines, I couldn't open proper issues here, and I tried to fixed by my own, now it's working with my fork after 10h, and more than 1M requests  done. 

Here is the list of problems I found and the solutions I applied, the solutions are dirty and for my only use case of eModbus, since are the works of debuging the codebase for 40h with the goal of making it work to reach my deadlines.

- When a `syncRequest` fails doesn't return until lost patience is met (1min), so I implemented an `isConnected` method to handle the connection by hand and avoid making request to a disconnected client. https://github.com/eModbus/eModbus/commit/020187e29c8b5819bb25f24080e81445ae447a44
- Disconnect doesn't clean properly the internal queues https://github.com/eModbus/eModbus/commit/020187e29c8b5819bb25f24080e81445ae447a44#diff-97e9b5de8eb31838da2befd02446dc1c5bf735f09e9ae051f205044c5aa38882R194-R205
- `waitSync` leaks `synResponses` even when lostPatience is triggered or the request is successful (I don't know how), I solved cleaning the map at the end of waitSync (I am using maxInFlight `1`) https://github.com/eModbus/eModbus/commit/d0c9155bc9dd27cb15ee3ba025155557f4f2d95f
- `ModbusClientTCPasync` gets in an invalid state when an error happens, and all the subsequent requests errors since the client can't connect again. I solved it by reading the state directly from `AsyncClient.state()` https://github.com/eModbus/eModbus/commit/02bf261665edde651d7083781f3e3ac78bbb5070

**Example code**

My code base is extensive and I don't have to make a standalone code to make it work. So here are a few snippset to understand the configuration related to the client 

```c++
void MasterBrick::asClient(TokenGenerator *generator)
{
  _client = new ModbusClientTCPasync(brick->getIP(), 502);
  _client->setMaxInflightRequests(1);
  _client->setTimeout(500);
  _client->setIdleTimeout(500); 
  _generator = generator;
}
```
```c++
template <typename... Args>
ModbusMessage MasterBrick::_doRequest(Args&&... args)
{
  _client->connect();
  int timeout = millis() + 500;
  while (!_client->isConnected()) {
    if (timeout > millis()) {
      _client->disconnect(true);
      delay(10);
      continue;
    }
    
    LOG_E("brick=%d, request error: not connected\n", brick->address());
    ModbusMessage response;
    response.setError(brick->address(), 0, TIMEOUT);

    return response;
  }
 
  LOG_D("brick=%d, making request\n", xPortGetCoreID(), brick->address());
  auto start = millis();
  auto response = _client->syncRequest(_generator->generate(), brick->address(), std::forward<Args>(args) ...);
  Error err = response.getError();
  if (err != SUCCESS) {
    _stats.error();

    ModbusError e(err);
    int elpased =  millis()-start;
    LOG_E("brick=%d, elpased=%dms, request error: %02X - %s\n", brick->address(), elpased, (int)e, (const char *)e);

    delay(10);
    return response;
  }

  _stats.success(millis() - start);
  return response;
}

```

**Additional context**
I just wanted to report this since maybe you found it helpful, I don't have much time now to contribute back, but at least I reported some possible bugs.

Miq1, 2022-10-23:
@mcuadros Thanks a lot! @bertmelis will be back this week and announced already he will look into the async TCP stuff.

bertmelis, 2022-10-23:
Thank you for the detailed writeup!
I'm back but suffer jetlag. My brain is on slow mode... Allow me a few days to get into the details.

mcuadros, 2022-10-23:
Sure, no problem. My code is running beside some weird request failures for a few seconds and back online after (I need to investigate), but at least the code is stable and has no memory leaks. 

My solutions are made for my use case of eModbus, but for sure is a starting point to solve the problems properly,

Miq1, 2022-10-27:
> * When a `syncRequest` fails doesn't return until lost patience is met (1min), so I implemented an `isConnected` method to handle the connection by hand and avoid making request to a disconnected client. [020187e](https://github.com/eModbus/eModbus/commit/020187e29c8b5819bb25f24080e81445ae447a44)

That is a laziness effect, I am afraid 😀 . The 60 seconds are required for Modbus ASCII only, but I missed to check that and rather took the longest time. Will fix that!

mcuadros, 2022-10-27:
If you make configurable better, I believe.

bertmelis, 2022-10-27:
Darn... Computer issues. Will be on it shortly

Miq1, 2022-10-27:
> If you make configurable better, I believe.

That is already done, you may set any timeout for the client. The set timeout is not considered yet, that is the issue.

Miq1, 2022-10-28:
@bertmelis I ran into the same again in quite a complicated way. The missing ``onResponse`` callback in ``onDisconnected()`` blinded me for IP connection errors happening. Only when I looked into the async client's verbose log I saw what happened.

mcuadros, 2022-10-30:
Hi, after many debug I found some problems even with my code after leaving the board running for some hours making requests every 100ms to two servers and trying to connect to some others. And usually ends crashing sooner or later.

```
E (5980317) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (5980317) task_wdt:  - async_tcp (CPU 0/1)
E (5980317) task_wdt: Tasks currently running:
E (5980317) task_wdt: CPU 0: IDLE
E (5980317) task_wdt: CPU 1: IDLE
E (5980317) task_wdt: Aborting.
```

This is one of the errors that I was getting, usually happens when one of the callbacks are blocked for a long time. My filling is something related to the locks for both queue rx and tx.

So I decided to create an implementation just for my use case sync request with 1 client, in this way, I was able to debug and provide the proper delays and avoid locks for a long time, and I got a version running for 15h and making almost 500k requests, without memory leaks:

[image]

The warnings are unsuccessful requests, and the errors are 3 warnings in a row (because I retry a request 3 times). These errors usually come from disconnects from `TCPAsync` and not being able to get a new connection from lwip, getting the error `pcb == NULL`. Having 20 clients at the same time is too much for `TCPAsync`.

How it helps.

Miq1, 2022-10-30:
I am too unexperienced with the async stuff to judge what you did differently, so @bertmelis must have a look.
But can you elaborate a bit for laymen as myself what you modified to achieve that impressive throughput?

bertmelis, 2022-11-11:
@mcuadros 
In the current implementation (and in PR #232) it is not possible to add new requests from inside a callback. Is this a wanted feature?

bertmelis, 2022-11-11:
> @mcuadros In the current implementation (and in PR #232) it is not possible to add new requests from inside a callback. Is this a wanted feature?

the PR now includes unlocked onResponse/onData/onError callbacks.