eModbus/eModbus #230: Compilation error toolchain-xtensa-esp32

Beetix, 2022-11-01:
**Describe the bug**
Compilation fails with error linked to the toolchain

**To Reproduce**
platformio.ini
```
[env:m5stack-core2]
platform = espressif32
board = m5stack-core2
framework = arduino
lib_deps = 
	ModbusClient=https://github.com/eModbus/eModbus.git
	m5stack/M5Core2@^0.1.5
```

**Example code**
```c
#include "M5Core2.h"
#include <ModbusServerWiFi.h>

ModbusServerWiFi MBserver;

const uint8_t SERVER_ID(1);
uint16_t memo[128];                 // Test server memory

// Worker function for serverID=1, function code 0x03 or 0x04
ModbusMessage FC03(ModbusMessage request) {
  uint16_t addr = 0;        // Start address to read
  uint16_t wrds = 0;        // Number of words to read
  ModbusMessage response;

  // Get addr and words from data array. Values are MSB-first, get() will convert to binary
  request.get(2, addr);
  request.get(4, wrds);
  
  // address valid?
  if (!addr || addr > 128) {
    // No. Return error response
    response.setError(request.getServerID(), request.getFunctionCode(), ILLEGAL_DATA_ADDRESS);
    return response;
  }

  // Modbus address is 1..n, memory address 0..n-1
  addr--;

  // Number of words valid?
  if (!wrds || (addr + wrds) > 127) {
    // No. Return error response
    response.setError(request.getServerID(), request.getFunctionCode(), ILLEGAL_DATA_ADDRESS);
    return response;
  }

  // Prepare response
  response.add(request.getServerID(), request.getFunctionCode(), (uint8_t)(wrds * 2));

  // Loop over all words to be sent
  for (uint16_t i = 0; i < wrds; i++) {
    // Add word MSB-first to response buffer
    response.add(memo[addr + i]);
  }

  // Return the data response
  return response;
}

void setup() {
    M5.begin();

    // Register the worker function with the Modbus server
    MBserver.registerWorker(SERVER_ID, READ_HOLD_REGISTER, &FC03);
[... 131 lines cut ...]
/home/bfreeman/.platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/8.4.0/chrono:826:7: error: non-constant condition for static assertion
       static_assert(system_clock::duration::min()
                     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
       < system_clock::duration::zero(),
       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/bfreeman/.platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/8.4.0/chrono:826:7:   in 'constexpr' expansion of 'std::chrono::operator< <long long int, std::ratio<1, 1000000000>, long long int, std::ratio<1, 1000000000> >(std::chrono::duration<long long int, std::ratio<1, 1000000000> >::min, std::chrono::duration<long long int, std::ratio<1, 1000000000> >::zero())'
/home/bfreeman/.platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/8.4.0/chrono:826:7: error: the value of 'std::chrono::duration<long long int, std::ratio<1, 1000000000> >::min' is not usable in a constant expression
/home/bfreeman/.platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/8.4.0/chrono:437:2: note: 'std::chrono::duration<long long int, std::ratio<1, 1000000000> >::min' was not declared 'constexpr'
  min() noexcept
  ^~~
Compiling .pio/build/m5stack-core2/lib3aa/FS/FS.cpp.o
Compiling .pio/build/m5stack-core2/lib3aa/FS/vfs_api.cpp.o
Compiling .pio/build/m5stack-core2/lib9dd/SPIFFS/SPIFFS.cpp.o
Compiling .pio/build/m5stack-core2/libea0/WiFiClientSecure/WiFiClientSecure.cpp.o
Compiling .pio/build/m5stack-core2/libea0/WiFiClientSecure/esp_crt_bundle.c.o
Compiling .pio/build/m5stack-core2/libea0/WiFiClientSecure/ssl_client.cpp.o
Compiling .pio/build/m5stack-core2/libb63/HTTPClient/HTTPClient.cpp.o
Compiling .pio/build/m5stack-core2/lib7c5/SD/SD.cpp.o
Compiling .pio/build/m5stack-core2/lib7c5/SD/sd_diskio.cpp.o
Compiling .pio/build/m5stack-core2/lib7c5/SD/sd_diskio_crc.c.o
Compiling .pio/build/m5stack-core2/libc72/M5Core2/AXP192.cpp.o
Compiling .pio/build/m5stack-core2/libc72/M5Core2/Fonts/Font16.c.o
*** [.pio/build/m5stack-core2/src/main.cpp.o] Error 1
============================================= [FAILED] Took 5.33 seconds =============================================
```

Miq1, 2022-11-01:
I am afraid this rather is looking like an error in the chrono implementation of the tool chain you are using, nothing related to the eModbus library. It looks like you are using a non-standard release: can you try to install another?

Beetix, 2022-11-02:
I checked. It is the latest release of the toolchain: [https://registry.platformio.org/tools/espressif/toolchain-xtensa-esp32/versions](https://registry.platformio.org/tools/espressif/toolchain-xtensa-esp32/versions)

Miq1, 2022-11-02:
Yeah, but still: the library is using a mutex - part of the standard C++ library - which in turn seems to use chrono - again a standard lib component - that finally has that error. I can not see anything we are able to change here - nothing for us to do about 🤷🏻‍♂️ 
You better should address the maintainers of the core to have that fixed.

Miq1, 2022-11-02:
I am using the PlatformIO platform version 5.2.0 that has xtensa version 8.4.0+2021r2-patch3 - just the patch level seems to be different from yours.
The error seems to be caused by a misled #include, that causes the <chrono> ``min()`` (for minutes) calls to be interpreted as calls to the math ``min(a, b)`` function. Whyever...

Beetix, 2022-11-09:
Adding `platform_packages = espressif/toolchain-xtensa-esp32@8.4.0+2021r2-patch3` to my `platformio.ini` and running 
```
$ pio run -t clean
$ pio run
```
yields the same error...

Miq1, 2022-11-10:
[This issue at M5stack](https://github.com/m5stack/M5Core2/issues/79) sounds very similar. You could try the workaround described there.

Miq1, 2022-11-22:
Closing this as inactive now.

closed 2022-11-22 by Miq1