eModbus/eModbus #233: Blocking calls

Naheel-Azawy, 2022-11-06:
#### Do not use to report bugs!

**Is your feature request related to a problem? Please describe.**
Non blocking calls are great. But for my use case, I actually need it to be blocking.

**Describe the solution you'd like**
Along with `MB.addRequest`, it would be great to have something like `MB.addRequestSync`.
Or maybe, another call like `response = MB.waitResponse()`

**Describe alternatives you've considered**
I tried storing data in a global and then keep checking if the data is there or not. But, this is too dirty.

**Additional context**
I'm not sure if this is a feature request or a question though. Maybe I'm just missing something.
I also wanted to say thanks to the devs. This is an actually good modbus library that can practically be used.

Miq1, 2022-11-06:
Please see the docs - there are ``syncRequest()`` calls since long that probably will do what you desire. 😉

closed 2022-11-06 by Miq1

Naheel-Azawy, 2022-11-07:
I did go over the documentation but I have no idea how I missed it.
Anyway, perfect answer. Thanks.

Miq1, 2022-11-07:
👍🏻 👋🏻