eModbus/eModbus #234: RTU connection: client stops with E"-CRC Check error

bastel-123, 2022-11-08:
Hi,

I started the usage of your great library yet. Currently I'm testing the RTU connection with your example code. After some minutes the client request stopps regulary with the error message "handleError: Error response: E2 - CRC check error". Nothing more happens.

What can I do to get the client resuming requests after an error?

regards
Frank

Miq1, 2022-11-08:
Hi Frank, we will need some more information before we may be able to help. Normally the examples will do their work in ``setup()`` and stop after having received a response. The E2 error points to some data error is happening.
- what is your hardware? ESP32 (which?), ESP8266?
- which server are you requesting? Device type, baud rate etc.
- please provide your sketch. Even if it is one of our examples, you may have changed something
- try to catch a verbose (``-DLOG_LEVEL=6`` as a compiler flag, ``MBUlogLvl=6;`` in your ``setup()``
-

bastel-123, 2022-11-08:
Hi Michael,

 

thank you for your very fast reaction. Do you speak German?

 

Regards

Frank

Miq1, 2022-11-08:
Ja, ich spreche Deutsch 😀

bastel-123, 2022-11-08:
Hallo Michael,

 

vielen Dank für deinen Support 😊.

 

Deutsch macht es (für mich) einfacher.

 

Zu deinen Fragen:

*	Ich nutzte einen ESP32 Wroom (38 Pins) und habe unter PlatformIO das Denky32 Board ausgewählt
*	Ich frage den Smart Meter EASTRON SDM630 per Modbus ab.
*	Der Sketch ist unverändert (Examples/RTU04example), bis auf Register, Modbus_ID vom EASTROM und dem Interval (=1000)
*	Log_Level werde ich machen und Dir zusenden

 

Viele Grüße

Frank

Miq1, 2022-11-08:
Okay, danke, das ist schon handhabbarer.

PlatformIO ist gut, da muss ich nicht erklären, wie man Compilerflags setzt 😁 
"Denky32" ist mir ja noch nie untergekommen... Unterscheidet sich das vom DevkitC-Board?
EastronSDM630 ist recht gängig und sollte keine Probleme machen. Du musst halt die Kommunikationsparameter auf beiden Seiten korrekt setzen (Baudrate, Datenbits, Parity usw.).
Wie hast Du die Verbindung zwischen beiden verdrahtet? Auf der ESP32-Seite brauchst Du ja einen RS485-Adapter, und eventuell musst Du auch den RS485-Bus terminieren (Pullup von A+ nach Vcc, Pulldown von B- nach GND, jeweils mit 4K7-Widerständen oder sogar 10K).

PS: Du benutzt einen Client für Github, der immer die Bneachrichtigungsmail zitiert - kannst Du das abstellen?

bastel-123, 2022-11-09:
Hallo Michael,

 

vielen lieben Dank für deine Hinweise und Tipps 😊.

 

Es läuft jetzt soweit. Es war wohl ein Layer-8 Fehler, nachdem ich den Example-Code nochmal übernommen habe, lief es. Keine Ahnung warum, Technik halt…

 

Ich bin jetzt dabei weiter mit deiner Library zu experimentieren. Ich habe mir schon einige Libraries (modbus8266, modbus master und weitere) angesehen, aber Dein Konzept gefällt mir am besten. 

 

Vielen lieben Dank nochmal für deine Mühe und deinen Support.

 

Ich hoffe, ich darf dich nochmal belästigen, wenn ich eine Frage habe?

 

Viele Grüße

Frank

closed 2022-11-09 by bastel-123

Miq1, 2022-11-09:
@bastel-123 Schön, dass es jetzt geklappt hat. Natürlich darfst du weiter fragen!

Die Ehre gebührt aber genauso @bertmelis, der trägt mindestens so viel zur Library bei wie ich!

bastel-123, 2022-11-10:
Hallo Michael,

 

wie kann ich dich am Besten per eMail erreichen? 

 

Ich habe eine Frage: Ich habe einen Modbus-Server (FC4) und einen Client (jeweils im Example-Code) gegeneinander verbunden (jeweils eine ESP32 mit einem RS485). Die Anfrage kommt beim Server an, er antwortet auch, aber beim Client kommt nichts an.

 

Viele Grüße

Frank

Miq1, 2022-11-10:
So ohne Details kann ich dazu nichts sagen, außer: normalerweise sollte das gehen...
Pack' mal die Logs von Client und Server hier herein, dann sieht man meist mehr. 

Wenn Du separat schreiben willst, kannst du die Adresse aus meinem Github-Profil benutzen.