eModbus/eModbus #235: Few compiler warnings to solve

bertmelis, 2022-11-09:
```
ModbusTypeDefs.h:127:10: warning: type qualifiers ignored on function return type [-Wignored-qualifiers]
   static const FCType getType(uint8_t functionCode);
          ^~~~~
ModbusTypeDefs.h:132:10: warning: type qualifiers ignored on function return type [-Wignored-qualifiers]
   static const FCType redefineType(uint8_t functionCode, const FCType type = FCUSER);
          ^~~~~
RTUutils.cpp:373:40: warning: 'b' may be used uninitialized in this function [-Wmaybe-uninitialized]
           if ((b & 0x80) || ASCIIread[b] == 0xFF) {
                             ~~~~~~~~~~~^
```

The last error is a false positive but can be easily avoided at line 240: `int b;` to be replaced by `int b = 0`;

Miq1, 2022-11-09:
Will do. Not before tomorrow I am afraid - I have been shoving heavy furniture and need some rest 😁

Miq1, 2022-11-10:
Should be okay now - trivial, I hope. I did not explicitly test it 😊

closed 2022-11-10 by Miq1

Miq1, 2022-11-10:
Argh. Missed to change the ``ModbusTypeDefs.cpp`` as well.