eModbus/eModbus #236: Strange results

georgeslegros, 2022-11-16:
#### Do not use to discuss topics! 

**Describe the bug**
I am using the ModbusTcpClient and based my code partly on the examples I found in this repo.
I'm sending  the following request 

```
Error err = MB.addRequest((uint32_t)millis(), modbusSlave, READ_HOLD_REGISTER, modbusPowerAddress, 2);
```

I have validated the modbusSlave and modbusPowerAddress.

Anyway, I am asking to read 2 words from a holding register.

The handle data returns the following results:

```
Response: serverID=3, FC=3, Token=00001283, length=7:
03 03 04 00 00 03 55 
```

I'm a bit puzzled to have a response that has 7 bytes and to be honest, I'm not too sure how to interpret the result.

**Expected behavior**
I expect to receive a 4 bytes result and if I could get a way to transform this into a decimal (int) value that'd be great.

Miq1, 2022-11-16:
Hi,

You in fact **do get** 4 data bytes, you will have to take the necessary header into consideration:
- ``03`` is the server ID
- the next ``03`` is the function code for "read holding register"
- ``04`` is the number of bytes to follow - this ends the header
- The next ``00 00`` is the first register content
- ``03 55`` is the second ``uint16_t`` value

To read these, you need to do something like
```
uint16_t value1, value2;
uint16_t offset = 3;   // skip the header
offset = response.get(offset, value1);
offset = response.get(offset, value2);
```
Assuming that your received response message is named ``response``.

closed 2022-11-16 by Miq1

georgeslegros, 2022-11-16:
Thanks a lot, that makes a lot of sense!