eModbus/eModbus #239: Add option to use software serial

I-Connect, 2022-11-28:
Hi,

We ran out of HW serials, would it be possible to use/add a constructor for software serial?

We use this library for that:
https://github.com/plerup/espsoftwareserial

Thx

Miq1, 2022-11-28:
Someone did that in a fork of eModbus already, so yes, that is possible. It is not on our roadmap yet, but I definitely will have a look at it.

I-Connect, 2022-11-29:
Hey Michael, thx. do you know which fork, there are a lot :-)

Then maybe I will use the fork until it is implemented in the original lib.

Miq1, 2022-11-29:
Sure, it was [this fork](https://github.com/serfreeman1337/eModbus) by @serfreeman1337.

Thermelgy-Repo, 2022-11-30:
> Hi,
> 
> We ran out of HW serials, would it be possible to use/add a constructor for software serial?
> 
> We use this library for that: https://github.com/plerup/espsoftwareserial
> 
> Thx

Hai @I-Connect,

I suggest to change the UART HW Serial port to the other IO Pin explicitly, Any pins other than Read only pin & Strapping Pins works. It will closely behave like a HW serial pin.

I-Connect, 2022-12-03:
the fork you mentioned also had some adaptations for the esp8265, I started with your lib and changed only the software serial part serfreeman did. It seems to work (with SW serial) although it only works on core 1 for me but that could have to do with all the other tasks I am running...

If you want I can create a pull request...?

Miq1, 2022-12-03:
> If you want I can create a pull request...?

Although this will be welcome, wait a sec, please. @bertmelis currently is trying to refactor the async TCP interface in the ``fix-async`` branch that I would like to merge first and independently. You could of course use that branch to create your PR, since I reckon the affected sources will be disjunct.

I-Connect, 2022-12-03:
There are indeed only changes in the RTU files so it should not bite but I understand you want to wait, no probs... currently have it implemented in a fork

Let me know when you are ready

Miq1, 2022-12-03:
👍🏻

Miq1, 2023-02-27:
Closing this as now it is possible to use any ``Stream``-based interface.

closed 2023-02-27 by Miq1