eModbus/eModbus #278: ModbusClient TCP how to stop a running work task

Miq1, 2023-03-10:
### Discussed in https://github.com/eModbus/eModbus/discussions/277

<div type='discussions-op-text'>

<sup>Originally posted by **dima-sz** March 10, 2023</sup>
ModbusClient TCP how to stop a running worker task started by calling begin(int coreID)?</div>

Miq1, 2023-03-26:
I added ``ModbusClientTCP::end()`` to the ``master`` branch. I would ask you to test it and report your findings here.

dima-sz, 2023-03-27:
Thanks, it works.

dima-sz, 2023-03-27:
But if you call the end() command when it is not yet executing a work task,
it will cause a crash.

E (76818) task_wdt: Task watchdog got triggered. The following tasks did
not reset the watchdog in time:
E (76818) task_wdt:  - async_tcp (CPU 1)
E (76818) task_wdt: Tasks currently running:
E (76818) task_wdt: CPU 0: IDLE
E (76818) task_wdt: CPU 1: loopTask
E (76818) task_wdt: Aborting.

abort() was called at PC 0x4013761c on core 0

Backtrace: 0x40083ae1:0x3ffbfe9c |<-CORRUPTED

Miq1, 2023-03-27:
Argh. Missed an ``if (worker) { ... }`` there. Will fix that asap.

Thank yiu for testing it!

dima-sz, 2023-03-27:
and also, if you start an already running process, it also crashes.

Miq1, 2023-03-27:
Thanks again - I fixed it a minute ago in ``master``. Could you give it another try?

dima-sz, 2023-03-27:
Thank you very much, everything works great now!

Miq1, 2023-03-27:
Thanks for verifying!

dima-sz, 2023-03-27:
There was another problem if you create two clients MB0 and MB1 and poll
devices that are not online (offline)

MB0.setTarget(IP-1, 502);
MB0.start();

MB1.setTarget(IP-2, 502);
MB1.start();

   it will crash.

Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was
unhandled.

dima-sz, 2023-03-27:
Sorry

MB0.setTarget(IP-1, 502);
MB0.begin();

MB1.setTarget(IP-2, 502);
MB1.begin();

Miq1, 2023-03-27:
Can we have a full exception dump please? 

The error hints f.i. to a use of a single memory area for both clients, where the second tries to access while the first already has freed it. This may happen more often because a connection is tried for 3s before determining it is impossible for offline servers. So the clients will stay in that state without doing anything else.

Miq1, 2023-03-27:
Do you use the same ``onData``/``onError``/``onResponse`` handlers for both clients? is there some dynamic data management involved in those functions?

dima-sz, 2023-03-27:
I want to say right away that this problem was not in the version before
the addition of the end() function, it was tested on 5-6 clients.
I use one onResponse handler.

exception dump is different, here are some options

assert failed: lwip_unlink_select_cb
IDF/components/lwip/lwip/src/api/sockets.c:1865 (select_cb->prev == NULL)

Backtrace: 0x40083a49:0x3ffd9140 0x4008bcfd:0x3ffd9160
0x40091919:0x3ffd9180 0x400fd8ef:0x3ffd92b0 0x400ff28b:0x3ffd92d0
0x400f5645:0x3ffd9370 0x400d9cc3:0x3ffd93e0 0x400d956e:0x3ffd9430
0x400e49ab:0x3ffd9460

==================================================

Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was
unhandled.

Core  0 register dump:
PC      : 0x400fd7c9  PS      : 0x00060330  A0      : 0x80111a88  A1      :
0x3ffb2f60
A2      : 0x06141150  A3      : 0x0000001d  A4      : 0x00000030  A5      :
0x00000001
A6      : 0x00010000  A7      : 0x3ffc61a4  A8      : 0x3ffc61a0  A9      :
0x0000001d
A10     : 0x00000001  A11     : 0x00000001  A12     : 0x00000001  A13     :
0x00000001
A14     : 0x00060b20  A15     : 0x00000001  SAR     : 0x00000010  EXCCAUSE:
0x0000001c
EXCVADDR: 0x0614116c  LBEG    : 0x40089660  LEND    : 0x40089676  LCOUNT  :
0xffffffff

Backtrace: 0x400fd7c6:0x3ffb2f60 0x40111a85:0x3ffb2f90
0x40103b61:0x3ffb2fb0 0x40103c6f:0x3ffb2ff0 0x40107053:0x3ffb3010
0x40107111:0x3ffb3030 0x400ff6f5:0x3ffb3050

================================================================

assert failed: lwip_unlink_select_cb
IDF/components/lwip/lwip/src/api/sockets.c:1865 (select_cb->prev == NULL)

Backtrace: 0x40083a49:0x3ffd9580 0x4008bcfd:0x3ffd95a0
0x40091919:0x3ffd95c0 0x400fd8ef:0x3ffd96f0 0x400ff28b:0x3ffd9710
0x400f5645:0x3ffd97b0 0x400d9cc3:0x3ffd9820 0x400d956e:0x3ffd9870
0x400e49ab:0x3ffd98a0

===========================================================

dima-sz, 2023-03-27:
> Do you happen to issue the end() while the connection is still hanging
> for the offline server? There have been reports that killing a TCP client
> while the polling is still going on results in a dangling pointer in lwIP,
> hence crashing on the next access to it (see here
> for an example).

Very strange, I wanted to reproduce this problem with a simple example,
without my main code, but I did not succeed, this error does not occur.
Perhaps in my code there is incompatibility of different libraries, I use
#include <Arduino.h>
#include <SimpleKalmanFilter.h>
#include "fs.h"
#include <LittleFS.h>
#include <wifi.h>
#include <AsyncTCP.h>
#include <ESPAsyncWebServer.h>
#include <WebAuthentication.h>
#include <DNSServer.h>
#include <AsyncElegantOTA.h>
#include "eprom.h"
#include <OneWire.h>
#include <DallasTemperaturemy.h>
#include "soc/soc.h"
#include "soc/rtc_cntl_reg.h"
#include "Button2.h"
#include "ModbusClientTCP.h"
#include "CoilData.h"
but, as I said earlier, this was not there before the changes were made,
now I have checked it again.

Miq1, 2023-03-29:
Any news? Else I would like to close here.

dima-sz, 2023-03-29:
Unfortunately, I'm busy with other important things right now and can't
continue checking for now, so close this one for now.

Miq1, 2023-03-29:
No worries, I will leave it open then. Whenever you find the time I will be eager to learn the outcome.

Miq1, 2023-05-12:
I am assuming nothing more will be posted here.

closed 2023-05-12 by Miq1