eModbus/eModbus #282: Compile Error v1.7.0

stif, 2023-03-26:
**Describe the bug**

i use PlatformIO and eModbus and have a compile error since v1.7.0

i still use Arduino Core 1.0.6 because of Ethernet network issues with Core 2.0.4 (another problem)
but the eModbus 1.7.0 does not compile with Core 2.0.4 either, although with a different compile error 🤔

the project compiles fine with eModbus v1.6.0 on Core 1.0.6 and Core 2.0.4

**Compile Error**

Arduino Core 1.0.6
```
- framework-arduinoespressif32 @ 3.10006.210326 (1.0.6) 
 - tool-esptoolpy @ 1.30100.210531 (3.1.0) 
 - toolchain-xtensa32 @ 2.50200.97 (5.2.0)
...
In file included from .pio/libdeps/esp32-poe/eModbus/src/ModbusClientRTU.h:14:0,
                 from lib/modbus/modbusHandler.h:6,
                 from src/main.cpp:21:
.pio/libdeps/esp32-poe/eModbus/src/RTUutils.h: In static member function 'static void RTUutils::prepareHardwareSerial(HardwareSerial&, uint16_t)':
.pio/libdeps/esp32-poe/eModbus/src/RTUutils.h:58:5: error: 'class HardwareSerial' has no member named 'setTxBufferSize'
   s.setTxBufferSize(bufferSize);
```

Arduino Core 2.0.4
```
 - framework-arduinoespressif32 @ 3.20004.0 (2.0.4) 
 - tool-esptoolpy @ 1.30300.0 (3.3.0) 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
...
.pio/libdeps/esp32-poe-latest/eModbus/src/ModbusClientRTU.cpp: In member function 'void ModbusClientRTU::begin(HardwareSerial&, int)':
.pio/libdeps/esp32-poe-latest/eModbus/src/ModbusClientRTU.cpp:66:10: error: 'class HardwareSerial' has no member named 'setRxFIFOFull'
   serial.setRxFIFOFull(1);
          ^~~~~~~~~~~~~
Compiling .pio/build/esp32-poe-latest/lib6c0/eModbus/ModbusServerRTU.cpp.o
Compiling .pio/build/esp32-poe-latest/lib6c0/eModbus/ModbusServerTCPasync.cpp.o
*** [.pio/build/esp32-poe-latest/lib6c0/eModbus/ModbusClientRTU.cpp.o] Error 1
.pio/libdeps/esp32-poe-latest/eModbus/src/ModbusServerRTU.cpp: In member function 'void ModbusServerRTU::begin(HardwareSerial&, int)':
.pio/libdeps/esp32-poe-latest/eModbus/src/ModbusServerRTU.cpp:76:10: error: 'class HardwareSerial' has no member named 'setRxFIFOFull'
   serial.setRxFIFOFull(1);
          ^~~~~~~~~~~~~
*** [.pio/build/esp32-poe-latest/lib6c0/eModbus/ModbusServerRTU.cpp.o] Error 1
```

Miq1, 2023-03-26:
The arduino-esp core version 2.0.x is required for 1.7 to be able to make use of the heavily modified UART handling in the core. Looks like the core you are using is missing these mods.

Miq1, 2023-03-26:
[Here is the definition of the function in core 2.0 ``HardwareSerial.h``](https://github.com/espressif/arduino-esp32/blob/0d84018d969309addacbcc3e3782c1fadc95fbc8/cores/esp32/HardwareSerial.h#L89)

stif, 2023-03-26:
I missed a upgrade of the Platformio espressif32 Platform. Now i am on 6.1.0 (was 5.1.0) and it compiles fine.
```
 - framework-arduinoespressif32 @ 3.20007.0 (2.0.7) 
 - tool-esptoolpy @ 1.40500.0 (4.5.0) 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
```
Really need to check the ethernet network issue with Olimex PoE Hardware and Core 2
Until then i will stick to eModbus v1.6.0

Thank you very much

closed 2023-03-26 by stif

rob040, 2023-03-28:
I had today the exact same problem. 
Thanks to this ticket I solved it easily.
Thanks for reporting the solution.