eModbus/eModbus #287: Question de débutant :-)

Thierrybelg, 2023-04-10:
Hello,
Sorry for my bad english, I'm French.
I would try your library but the example generate errors :
Any idea why ?

Thanks.
Thierry

In file included from c:\Users\ThierryBERGER\OneDrive - BUPRAFISC\Documents\Arduino\libraries\eModbus\src\ModbusClientRTU.h:14:0,
                 from c:\Users\ThierryBERGER\OneDrive - BUPRAFISC\Documents\Arduino\libraries\eModbus\src\ModbusClientRTU.cpp:5:
c:\Users\ThierryBERGER\OneDrive - BUPRAFISC\Documents\Arduino\libraries\eModbus\src\RTUutils.h: In static member function 'static void RTUutils::prepareHardwareSerial(HardwareSerial&, uint16_t)':
c:\Users\ThierryBERGER\OneDrive - BUPRAFISC\Documents\Arduino\libraries\eModbus\src\RTUutils.h:58:5: error: 'class HardwareSerial' has no member named 'setTxBufferSize'
   s.setTxBufferSize(bufferSize);
     ^
c:\Users\ThierryBERGER\OneDrive - BUPRAFISC\Documents\Arduino\libraries\eModbus\src\ModbusClientRTU.cpp: In member function 'void ModbusClientRTU::begin(HardwareSerial&, int)':
c:\Users\ThierryBERGER\OneDrive - BUPRAFISC\Documents\Arduino\libraries\eModbus\src\ModbusClientRTU.cpp:66:10: error: 'class HardwareSerial' has no member named 'setRxFIFOFull'
   serial.setRxFIFOFull(1);
          ^
In file included from c:\Users\ThierryBERGER\OneDrive - BUPRAFISC\Documents\Arduino\libraries\eModbus\src\ModbusServerRTU.h:15:0,
                 from c:\Users\ThierryBERGER\OneDrive - BUPRAFISC\Documents\Arduino\libraries\eModbus\src\ModbusServerRTU.cpp:5:
c:\Users\ThierryBERGER\OneDrive - BUPRAFISC\Documents\Arduino\libraries\eModbus\src\RTUutils.h: In static member function 'static void RTUutils::prepareHardwareSerial(HardwareSerial&, uint16_t)':
c:\Users\ThierryBERGER\OneDrive - BUPRAFISC\Documents\Arduino\libraries\eModbus\src\RTUutils.h:58:5: error: 'class HardwareSerial' has no member named 'setTxBufferSize'
   s.setTxBufferSize(bufferSize);
     ^
c:\Users\ThierryBERGER\OneDrive - BUPRAFISC\Documents\Arduino\libraries\eModbus\src\ModbusServerRTU.cpp: In member function 'void ModbusServerRTU::begin(HardwareSerial&, int)':
c:\Users\ThierryBERGER\OneDrive - BUPRAFISC\Documents\Arduino\libraries\eModbus\src\ModbusServerRTU.cpp:76:10: error: 'class HardwareSerial' has no member named 'setRxFIFOFull'
   serial.setRxFIFOFull(1);
          ^

exit status 1

Compilation error: exit status 1

Miq1, 2023-04-10:
You will need to upgrade to a arduino-esp32 core version 2.0.x and up. Only with the newer versions of the core we may use the improved UART handling.

closed 2023-04-10 by Miq1

Thierrybelg, 2023-04-10:
Thanks, I will try it.