eModbus/eModbus #294: Modbus is not Responding when the PPP communication is established

Thermelgy-Repo, 2023-04-26:
Thank you Team for your great library, we are using this library more than 1 yrs working amazingly, Recently found an issue:

We are using RS485 Modbus RTU in Serial2 UART of esp32 & GSM module is connected to GPIO26, 27 & configured as explicit UART. The communication with GSM module is established with PPP. 
Individually both are working perfectly & while integrating also when no Modbus Slave is connected both the modules are working perfectly.
But when the slave is connected, initially both PPP & eModbus are working, But after the PPP establishes the connection the eModbus polling stops & the _ModbusClientRTU.cpp_ logs are not showing. The PPP is working fine. As my preliminary assumption is that the Worker task is not executing. Is the HardwareSerial conflict caused the task to exit?

**Build Log:**

```
PLATFORM: Espressif 32 (3.5.0) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.10006.210326 (1.0.6) 
 - tool-esptoolpy @ 1.30100.210531 (3.1.0) 
 - toolchain-xtensa32 @ 2.50200.97 (5.2.0)
Dependency Graph
|-- eModbus @ 1.6.0

```

**Note**
Check the Serial Logs, After _01:11:16.731 ->_, there is no ModbusClientRTU logs.

**Serial Logs**:

```
01:11:02.117 ->  Comm : PPP[V][PPPCom_Lib.cpp:58] PPP_init(): Waiting for PPPoS task to be created...
01:11:02.117 -> [I][GSMMain_Lib.cpp:33] M66_init(): Wait for GSM to connect...
01:11:02.117 -> WiFi range: 0 dBm
01:11:02.117 -> 1970-01-01 00:00:22  E rpc.c:287:mount_fs        FS -> /spiffs: 259
01:11:02.499 -> 1970-01-01 00:00:23  I log.c:18:mg_log_set       Setting log level to 1
01:11:02.499 -> 1970-01-01 00:00:23  E sock.c:117:ll_write       0x37 tUc -1/27 118
01:11:02.499 -> 1970-01-01 00:00:23  E sock.c:117:ll_write       0x37 tUc -1/33 118
01:11:02.499 -> [V] 23069| ModbusMessage.cpp    [ 477] checkData: Check data #3
01:11:02.499 -> Client disconnected
01:11:02.499 -> [D] 23071| ModbusClientRTU.cpp  [ 139] addRequestM: request for 01/03
01:11:02.499 -> Re-attempting to connect
01:11:02.499 -> [D] 23079| ModbusClientRTU.cpp  [ 240] handleConnection: Pulled request from queue
01:11:02.499 -> [D] 23079| ModbusClientRTU.cpp  [ 222] addToQueue: RC=01
01:11:02.499 -> Connecting as: MQCL_Serv1_Dev
01:11:02.499 -> [D] Sent packet: @3FFD32E0/6:
01:11:02.499 -> [D] 23095| ModbusClientRTU.cpp  [ 150] addRequestM: RC=00
01:11:02.499 ->   | 0000: 01 03 10 04 00 08                                 |......          |
01:11:02.499 -> [D] 23110| ModbusClientRTU.cpp  [ 245] handleConnection: Request sent.
01:11:02.595 -> [V] 23157| RTUutils.cpp         [ 289] receive: 1844us without data
01:11:02.595 -> [D] Raw buffer received: @3FFDEC44/21:
01:11:02.595 ->   | 0000: 01 03 10 00 01 00 00 00  00 00 00 00 00 00 00 00  |................|
01:11:02.595 ->   | 0010: 00 00 00 25 59                                    |...%Y           |
01:11:02.595 -> [D] Received packet: @3FFD338C/19:
01:11:02.595 ->   | 0000: 01 03 10 00 01 00 00 00  00 00 00 00 00 00 00 00  |................|
01:11:02.595 ->   | 0010: 00 00 00                                          |...             |
01:11:02.595 -> [D] 23185| ModbusClientRTU.cpp  [ 259] handleConnection: Data response (19 bytes) received.
01:11:02.643 -> [V] Data: @3FFD338C/19:
01:11:02.643 ->   | 0000: 01 03 10 00 01 00 00 00  00 00 00 00 00 00 00 00  |................|
01:11:02.643 ->   | 0010: 00 00 00                                          |...             |
01:11:02.643 -> [D] 23209| ModbusClientRTU.cpp  [ 280] handleConnection: Response generated.
01:11:02.643 -> [V] Response packet: @3FFD338C/19:
01:11:02.643 ->   | 0000: 01 03 10 00 01 00 00 00  00 00 00 00 00 00 00 00  |................|
01:11:02.643 ->   | 0010: 00 00 00                                          |...             |
01:11:02.643 -> [D] Data: @3FFB382C/19:
01:11:02.643 ->   | 0000: 01 03 10 00 01 00 00 00  00 00 00 00 00 00 00 00  |................|
01:11:02.643 ->   | 0010: 00 00 00                                          |...             |
 -> [D] 25646| ModbusClientRTU.cpp  [ 280] handleConnection: Response generated.
01:11:05.085 -> [V] Response packet: @3FFD33A4/53:
01:11:05.085 ->   | 0000: 02 03 32 00 00 00 00 00  00 00 00 00 00 00 00 00  |..2.............|
01:11:05.085 ->   | 0010: 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
01:11:05.085 ->   | 0020: 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
01:11:05.085 ->   | 0030: 00 00 00 00 00                                    |.....           |
01:11:05.085 -> [D] Data: @3FFDEC44/53:
[... 476 lines cut ...]
01:11:41.613 -> 1970-01-01 00:01:02  E event.c:18:mg_error       0x3d DNS timeout
01:11:42.617 -> 1970-01-01 00:01:03    mDash.c:203:timercb       ram: 167104, 2 0x0/0x0/0x3ffd2c38
01:11:42.617 -> 1970-01-01 00:01:03    sock.c:303:mg_connect     0x3e -> wss://mdash.net/api/v2/rpc
01:11:42.617 -> 1970-01-01 00:01:03    dns.c:203:mg_resolve      0x3e resolving mdash.net, txnid 17
01:11:42.617 -> 1970-01-01 00:01:03    tls.c:59:mg_tls_init      0x3e Setting TLS, CA: /spiffs/ca.pem, cert: null, key: null
01:11:42.617 -> 1970-01-01 00:01:03    sock.c:228:close_conn     0x3d closed
01:11:42.856 -> 1970-01-01 00:01:03    mDash.c:203:timercb       ram: 125636, 2 0x3ffd1f14/0x0/0x0
01:11:42.856 -> 1970-01-01 00:01:03    sock.c:303:mg_connect     0x36 -> udp://time.google.com:123
01:11:42.856 -> 1970-01-01 00:01:03    dns.c:203:mg_resolve      0x36 resolving time.google.com, txnid 18
01:11:42.856 -> 1970-01-01 00:01:03  E dns.c:124:dns_cb          stray DNS reply
01:11:42.856 -> 
01:11:42.856 -> 
01:11:42.856 ->  Second : 4907
01:11:42.856 -> IDI : 1, 1, 1, 1, 
01:11:42.856 -> IRLY : {{1,0},{2,0},{3,0},{4,0},}Pkt1 - Fail: 1, Succ: 8, Pkt2 - Fail: 0, Succ: 8, Pkt3 - Fail: 0, Succ: 8, Pkt4 - Fail: 0, Succ: 8, Pkt5 - Fail: 8, Succ: 0, 
01:11:42.856 -> Param1: 0.00, Param2: 0.00, Param3: 0.00, Param4: 0.00, Param5: 0.00, Param6: 0.00, Param7: 0.00, Param8: 0.00, Param9: 0.00, Param10: 0.00, Param11: 0.00, Param12: 0.00, Param13: 0.00, Param14: 0.00, Param15: 0.00, 
01:11:42.856 -> Free Heap : 191264
01:11:42.856 -> MData: 
01:11:43.814 -> 1970-01-01 00:01:04    mDash.c:203:timercb       ram: 126204, 2 0x3ffd1f14/0x0/0x3ffdf918
01:11:44.820 -> 1970-01-01 00:01:05    mDash.c:203:timercb       ram: 126204, 2 0x3ffd1f14/0x0/0x3ffdf918
01:11:45.346 -> 1970-01-01 00:01:05    sock.c:272:mg_connect_res 0x3e resolved: 157.90.34.55:443
01:11:45.395 -> 1970-01-01 00:01:05    sntp.c:60:mg_sntp_send    0x36 request sent, ct 65, next at 3665
01:11:45.395 -> 1970-01-01 00:01:05    sock.c:272:mg_connect_res 0x36 resolved: 216.239.35.0:123

```

Miq1, 2023-04-27:
Looks like the worker task is killed without a trace here, but as I do not know the PPP stuff at all, it is difficult to spot a reason.

Two things caught my attention, though: you are using the older 1.6.0 version of eModbus and the last thing logged seems to be a CRC error.

The former most probably is required for the quite outdated arduino-esp core you are using. Is there a reason why you did not update yet to a more up-to-date core? The UART handling has vastly been improved with 2.0.x and the last 1.7.0 eModbus version takes advantage of that.

The latter may hint to some collision or race condition happening with the UART just before the worker stops.

Sorry I have no better ideas at the moment of what might be happening here.

Miq1, 2023-05-12:
Any news on this?

Thermelgy-Repo, 2023-06-15:
Thanks @Miq1 for your guidence, 
I have solved it by updating the esp-Arduino framework from V1.0.6 to V2.0.9 & also updated all other libraries. I have found that eModbus V1.0.7 is showing much CRC or Packet error compared to V1.0.6, So I used the same V1.0.6. 

I have one doubt that, Can I enable RTU alone & disabling TCP?. I am trying to minimize the Wifi RAM to disable IPV6, But the Async Web Library on TCP is affecting. Anyway I am using only RTU, So Any method to avoid IPV6 in Async?

Miq1, 2023-06-16:
If you are using RTU only, the TCP stuff in the library will be compiled when using the Arduino IDE, but not linked into your binary anyway.

Thermelgy-Repo, 2023-06-16:
> If you are using RTU only, the TCP stuff in the library will be compiled when using the Arduino IDE, but not linked into your binary anyway.
> 
> If you are still using eModbus 1.6, you will not benefit from the different UART handling in the core. You will have to upgrade to 1.7 to enjoy that. If 1.7 was giving you an increased number of CRC errors, the root cause is another. In the past it turned out most of the time the culprit was the wiring and/or improper termination or biasing...

Thank you for your response,

I am using a custom sdkconfig, If I am disabling the IPV6 in `LWIP CONFIG_LWIP_IPV6` , I am getting the compilation error related to AsyncTCP.cpp. This is the Compilation error :

```
Compiling .pio\build\esp32dev\lib312\eModbus\CoilData.cpp.o
Compiling .pio\build\esp32dev\lib312\eModbus\Logging.cpp.o
Compiling .pio\build\esp32dev\lib312\eModbus\ModbusClient.cpp.o
Compiling .pio\build\esp32dev\lib312\eModbus\ModbusClientRTU.cpp.o
Compiling .pio\build\esp32dev\lib312\eModbus\ModbusClientTCP.cpp.o
.pio/libdeps/esp32dev/AsyncTCP/src/AsyncTCP.cpp: In function 'void _handle_async_event(lwip_event_packet_t*)':
.pio/libdeps/esp32dev/AsyncTCP/src/AsyncTCP.cpp:180:68: error: no matching function for call to 'AsyncClient::_s_dns_found(const char*&, ip_addr_t*, void*&)'
         AsyncClient::_s_dns_found(e->dns.name, &e->dns.addr, e->arg);
                                                                    ^
In file included from .pio/libdeps/esp32dev/AsyncTCP/src/AsyncTCP.cpp:24:
.pio/libdeps/esp32dev/AsyncTCP/src/AsyncTCP.h:137:17: note: candidate: 'static void AsyncClient::_s_dns_found(const char*, ip_addr*, void*)'
     static void _s_dns_found(const char *name, struct ip_addr *ipaddr, void *arg);
                 ^~~~~~~~~~~~
.pio/libdeps/esp32dev/AsyncTCP/src/AsyncTCP.h:137:17: note:   no known conversion for argument 2 from 'ip_addr_t*' {aka 'ip4_addr*'} to 'ip_addr*'
.pio/libdeps/esp32dev/AsyncTCP/src/AsyncTCP.cpp: In function 'void _tcp_dns_found(const char*, ip_addr*, void*)':       
.pio/libdeps/esp32dev/AsyncTCP/src/AsyncTCP.cpp:319:59: error: invalid application of 'sizeof' to incomplete type 'ip_addr'
         memcpy(&e->dns.addr, ipaddr, sizeof(struct ip_addr));
                                                           ^
.pio/libdeps/esp32dev/AsyncTCP/src/AsyncTCP.cpp: In member function 'bool AsyncClient::connect(IPAddress, uint16_t)':   
.pio/libdeps/esp32dev/AsyncTCP/src/AsyncTCP.cpp:694:10: error: 'ip_addr_t' {aka 'struct ip4_addr'} has no member named 'type'
     addr.type = IPADDR_TYPE_V4;
          ^~~~
.pio/libdeps/esp32dev/AsyncTCP/src/AsyncTCP.cpp:695:10: error: 'ip_addr_t' {aka 'struct ip4_addr'} has no member named 'u_addr'; did you mean 'addr'?
     addr.u_addr.ip4.addr = ip;
          ^~~~~~
          addr
.pio/libdeps/esp32dev/AsyncTCP/src/AsyncTCP.cpp: In member function 'bool AsyncClient::connect(const char*, uint16_t)': 
.pio/libdeps/esp32dev/AsyncTCP/src/AsyncTCP.cpp:724:39: error: 'ip_addr_t' {aka 'struct ip4_addr'} has no member named 'u_addr'; did you mean 'addr'?
         return connect(IPAddress(addr.u_addr.ip4.addr), port);
                                       ^~~~~~
                                       addr
.pio/libdeps/esp32dev/AsyncTCP/src/AsyncTCP.cpp: In member function 'void AsyncClient::_dns_found(ip_addr*)':
.pio/libdeps/esp32dev/AsyncTCP/src/AsyncTCP.cpp:962:24: error: invalid use of incomplete type 'struct ip_addr'
     if(ipaddr && ipaddr->u_addr.ip4.addr){
                        ^~
In file included from .pio/libdeps/esp32dev/AsyncTCP/src/AsyncTCP.cpp:24:
.pio/libdeps/esp32dev/AsyncTCP/src/AsyncTCP.h:53:8: note: forward declaration of 'struct ip_addr'
 struct ip_addr;
        ^~~~~~~
.pio/libdeps/esp32dev/AsyncTCP/src/AsyncTCP.cpp:963:33: error: invalid use of incomplete type 'struct ip_addr'
         connect(IPAddress(ipaddr->u_addr.ip4.addr), _connect_port);
                                 ^~
In file included from .pio/libdeps/esp32dev/AsyncTCP/src/AsyncTCP.cpp:24:
.pio/libdeps/esp32dev/AsyncTCP/src/AsyncTCP.h:53:8: note: forward declaration of 'struct ip_addr'
 struct ip_addr;
        ^~~~~~~
.pio/libdeps/esp32dev/AsyncTCP/src/AsyncTCP.cpp: In member function 'uint32_t AsyncClient::getRemoteAddress()':
.pio/libdeps/esp32dev/AsyncTCP/src/AsyncTCP.cpp:1052:28: error: 'ip_addr_t' {aka 'struct ip4_addr'} has no member named 'u_addr'; did you mean 'addr'?
     return _pcb->remote_ip.u_addr.ip4.addr;
                            ^~~~~~
                            addr
.pio/libdeps/esp32dev/AsyncTCP/src/AsyncTCP.cpp: In member function 'uint32_t AsyncClient::getLocalAddress()':
.pio/libdeps/esp32dev/AsyncTCP/src/AsyncTCP.cpp:1066:27: error: 'ip_addr_t' {aka 'struct ip4_addr'} has no member named 'u_addr'; did you mean 'addr'?
     return _pcb->local_ip.u_addr.ip4.addr;
                           ^~~~~~
                           addr
.pio/libdeps/esp32dev/AsyncTCP/src/AsyncTCP.cpp: In member function 'void AsyncServer::begin()':
.pio/libdeps/esp32dev/AsyncTCP/src/AsyncTCP.cpp:1262:16: error: 'ip_addr_t' {aka 'struct ip4_addr'} has no member named 'type'
     local_addr.type = IPADDR_TYPE_V4;
                ^~~~
.pio/libdeps/esp32dev/AsyncTCP/src/AsyncTCP.cpp:1263:16: error: 'ip_addr_t' {aka 'struct ip4_addr'} has no member named 'u_addr'; did you mean 'addr'?
     local_addr.u_addr.ip4.addr = (uint32_t) _addr;
                ^~~~~~
                addr
*** [.pio\build\esp32dev\lib93c\AsyncTCP\AsyncTCP.cpp.o] Error 1
```

Modified sdkconifg:
`
CONFIG_LWIP_IPV6=`

Modified sdkconfig.h:

`#define CONFIG_LWIP_IPV6 0`

Platform.ini File:

```
platform = espressif32@6.3.1
board = esp32dev
framework = arduino
board_upload.flash_size = 16MB
board_upload.maximum_size = 16777216
build_flags =
	-Isdkconfig
```

Miq1, 2023-06-16:
That is rather an issue of AsyncTCP requiring something your config does not provide.

You could either throw out or deactivate by preprocessor ``#ifdef``s all TCP stuff in eModbus. That should get you around that. It also may be possible to prevent PlatformIO from caching all library code regardless if being used, but I do not know how to do it. You could place a question in the PlatformIO support forum to learn about that.

Miq1, 2023-07-06:
Looks like this issue has dried out...

closed 2023-07-06 by Miq1