eModbus/eModbus #308: Modbus TCP/IP server over Ethernet not compiling on Arduino environment

codev123, 2023-07-05:
HI sir, I have a application where i want to read modbus data from 9 energy meters over modbus rtu and create a buffer to store these meters parameters, and make tcp ip ethernet server to serve that data to TCP IP ethernet client. Now i am able to create modbus rtu part of application but while using ethernet server your example code is not even compiling and gives me these errors. please sir help me! thank you

In file included from C:\Users\deven\OneDrive\Documents\Arduino\libraries\eModbus-master\src/ModbusServerEthernet.h:15,
                 from C:\Users\deven\Desktop\testmodbusethernet\testmodbusethernet.ino:7:
C:\Users\deven\OneDrive\Documents\Arduino\libraries\eModbus-master\src/ModbusServerTCPtemp.h: In instantiation of 'static void ModbusServerTCP<ST, CT>::serve(ModbusServerTCP<ST, CT>*) [with ST = EthernetServer; CT = EthernetClient]':
C:\Users\deven\OneDrive\Documents\Arduino\libraries\eModbus-master\src/ModbusServerTCPtemp.h:162:45:   required from 'bool ModbusServerTCP<ST, CT>::start(uint16_t, uint8_t, uint32_t, int) [with ST = EthernetServer; CT = EthernetClient; uint16_t = short unsigned int; uint8_t = unsigned char; uint32_t = unsigned int]'
C:\Users\deven\Desktop\testmodbusethernet\testmodbusethernet.ino:118:31:   required from here
C:\Users\deven\OneDrive\Documents\Arduino\libraries\eModbus-master\src/ModbusServerTCPtemp.h:224:8: error: cannot declare variable 'mserver' to be of abstract type 'EthernetServer'
     ST mserver(myself->serverPort);
        ^~~~~~~
In file included from C:\Users\deven\OneDrive\Documents\Arduino\libraries\eModbus-master\src/ModbusServerEthernet.h:9,
                 from C:\Users\deven\Desktop\testmodbusethernet\testmodbusethernet.ino:7:
C:\Users\deven\OneDrive\Documents\Arduino\libraries\Ethernet\src/Ethernet.h:254:7: note:   because the following virtual functions are pure within 'EthernetServer':
 class EthernetServer : public Server {
       ^~~~~~~~~~~~~~
In file included from C:\Users\deven\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.9\cores\esp32/Arduino.h:182,
                 from C:\Users\deven\Desktop\testmodbusethernet\testmodbusethernet.ino:6:
C:\Users\deven\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.9\cores\esp32/Server.h:28:18: note: 	'virtual void Server::begin(uint16_t)'
     virtual void begin(uint16_t port=0) =0;

this error ocuured while compiling TCPEthernetServer example code of library.

Miq1, 2023-07-05:
Please have a look at the ``platform.ini`` file of the eModbus lib. We are using a special Ethernet lib provided by Max Gerhardt that has some deficits of the standard core lib corrected. To me it looks like you have run over one of these.

Miq1, 2023-07-06:
@codev123 Any update yet?

codev123, 2023-07-06:
Thankyou very much sir! You were right, the suggested library is working, but i can access server only through router which allot it IP address, but if i manually set ip address using ip, gateway and dns in ethernet class, i can't read it then.

Also, can we keep this issue open for some time till i complete my application code. Thankyou very much sir. Namaste!

codev123, 2023-07-06:
Also can you please suggest me a flow for my code, i want to read 9 parameters per energy meter, 10 energy meters will give me 90 parameters. Then i want to access those 90 parameters through ethernet modbus tcp ip server by ethernet client. 

Right now my approach is to read meters through rtu client, save data simultaneously to a buffer and serve it to tcp server. Will it work?

Miq1, 2023-07-06:
Of course that will work. If the requests via TCP to your device are frequent, this way to do it may save some processing time for the price of not-yet-realtime data.
If on the other hand the requests will be less frequent or even few only, a bridge would be a better solution, as it will respond with realtime data. 
With the latter method the requests must know about the different meters' server IDs of course, while with your approach all meters' data can be arranged to different addresses, but the same server ID and may be requested all in a single request.
Whatever suits you best.

codev123, 2023-07-07:
hello sir, i have another problem while running rtu client to read energy meters modbus data, the error response comes " packet length error E5". thank you sir

Miq1, 2023-07-07:
Are you using the newest 1.7 version of the lib and an arduino-esp core 2.0.x or newer? With these the UART handling was massively modified and some internal causes for the E5 error have been removed.
If you do use the current versions already my bet would be on cabling, biasing or termination issues.

codev123, 2023-07-07:
sir, i have updated emodbus library installed and esp32 arduino 2.0.9 https://espressif.github.io/arduino-esp32/package_esp32_index.json   on my system. also i have standard modbus circuit tested at various boards and running smoothly. but with esp32 i am getting packet length error E5. 

also, i tried to run rtu client using modbusmaster library, 4-20ma one, and there i had E0 response i.e. invalid slave id, when i changed library to drop first byte its working proprly then. 

what changes should i do in your library or my code so that i can do same workaround to use emodbus also. how to use skipLeading0x00(true); in my code.
 thank you sir.

Miq1, 2023-07-07:
E0 is TIMEOUT, by the way. Both errors typically are resulting from improper communication on the RS485 adapter in combination with the ESP32. That you seem to need to skip the first 0x00 byte is another evidence.

Bert has explained that in a pinned discussion post, much better than I could do it. Perhaps you will want to read it some time. My shortened version is: when switching from reading to writing mode or back (when the DE/RE line is toggled), some RS485 adapters will let the UART lines float or pulse for a short time. The ESP32 UART registers that as a 0x00 byte coming in. Depending on the time it will take for the first Modbus message byte coming in, this is either interpreted as a TIMEOUT for the singular zero byte, or as packet length error, because the zero byte is taken as part of the message.

This effect does not happen with RS485 adapters that do automatic half-duplex switching nor with some of the regular adapters, that seem to be quicker switching or have a means to prevent the line floating.

codev123, 2023-07-20:
hello Sir, the issue was resolved with MB.skipLeading0x00(true); line. and my code is working fine. thank you.  
Another Query Sir, can i use this library on rp2040 pico pi board. if yes then how can i port this library to rp2040 arduino-pico core of rp2040. thank you very much sir for your help.

Miq1, 2023-07-20:
Unfortunately I do not know anything about the rp2040, so I am out of the game here.

codev123, 2023-07-20:
ok sir. i was talking about raspberry pico pi board. thank you very much for your help.

closed 2023-07-20 by codev123